Ysurac/openmptcprouter #111: VPS Script install not complete

SoGooDFR, 2018-07-28:
## Expected Behavior
Fresh Debian install and execute : 
`wget -O - http://www.openmptcprouter.com/server/debian9-x86_64.sh | sh`
And after script we get : '/root/openmptcprouter_config.txt'

## Actual Behavior
No generate of /root/openmptcprouter_config.txt and not modify '/etc/motd'.

## Steps to Reproduce the Problem

  1. Fresh OS + OpenMPTCPRounter install on VPS.
  2. No generated files by script.

## Specifications

  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.36.

Ysurac, 2018-07-28:
I've no problem on updating but I will try on a new install

PxxBxx, 2018-07-28:
BTW when upgrading, the script fails/stops when invoking : 
```
systemctl -q disable omr-6in4
```
I removed the line and was able to do the upgrade (I didn't have this service beforehand, I upgrades 0.37 to 0.37, in order to fetch the new shorewall conf)

nicoz76, 2018-07-28:
same problem on a fresh install ( debian9 OVH ).

closed 2018-07-28 by Ysurac

nicoz76, 2018-07-28:
Good,   it's OK   after the install script now.