Ysurac/openmptcprouter #113: RPI3 system/kernel log flooded with cpufreq errors

PxxBxx, 2018-07-29:
## Expected Behavior

a system log with only relevant logs, in order to easily find what is needed (eg. dnsmasq requests when enabled)

## Actual Behavior

the system log is literally flooded with errors from cpufreq commands, multiple time per seconds
  
```
Sun Jul 29 11:27:45 2018 kern.err kernel: [ 2745.177966] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 600000 (-12)
Sun Jul 29 11:27:45 2018 kern.err kernel: [ 2745.188822] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (600000)
Sun Jul 29 11:27:45 2018 kern.err kernel: [ 2745.201939] cpufreq: __target_index: Failed to change cpu frequency: -22
Sun Jul 29 11:27:45 2018 kern.err kernel: [ 2745.237965] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 600000 (-12)
Sun Jul 29 11:27:45 2018 kern.err kernel: [ 2745.248850] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (600000)
Sun Jul 29 11:27:45 2018 kern.err kernel: [ 2745.261945] cpufreq: __target_index: Failed to change cpu frequency: -22
Sun Jul 29 11:27:45 2018 kern.err kernel: [ 2745.297963] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 600000 (-12)
Sun Jul 29 11:27:45 2018 kern.err kernel: [ 2745.308829] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (600000)
Sun Jul 29 11:27:45 2018 kern.err kernel: [ 2745.321893] cpufreq: __target_index: Failed to change cpu frequency: -22
Sun Jul 29 11:27:46 2018 kern.err kernel: [ 2745.548045] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 600000 (-12)
Sun Jul 29 11:27:46 2018 kern.err kernel: [ 2745.558987] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (600000)
Sun Jul 29 11:27:46 2018 kern.err kernel: [ 2745.572156] cpufreq: __target_index: Failed to change cpu frequency: -22
```

my architecture : 
```
root@OpenMPTCProuter:/proc# cat cpuinfo
processor	: 0
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
```

## Steps to Reproduce the Problem

  1. let a OMR run for a couple minutes on a RPI3 (first rev)
  2. launch readlog or dmesg

## Specifications

  - OpenMPTCProuter version: 0.29.1 (also in 0.29 with previous installs)
  - OpenMPTCProuter VPS version: 0.39
  - OpenMPTCProuter platform: RPI3

Ysurac, 2018-07-29:
I think you added something on USB, you can set on **System**->**OpenMPTCProuter**->**Advanced setting**, **Minimum scaling CPU frequency** to the same value as **Maximum scaling CPU frequency**, so *1400000*.

PxxBxx, 2018-07-29:
You're right, having something on USB (a powered switch and a 4g usb router), I also saw that could trigger this bug.
On your advice I set the min_freq to the same value as max_freq, but I choose 600000, I'll try and see if it affect the performance of the router.
Thanks, it was not OMR related, but RPI-related.

Ysurac, 2018-08-03:
Should be done automatically in latest version now.

closed 2018-08-03 by Ysurac

pankajmishra16, 2019-06-11:
Hi Ysurac , i am also trying to use the USB Post with eth adopter and the other end of the eth in connected to internet.

DeeJayMX, 2020-05-20:
Same on 0.54.7 and RPi 4B 4GB
No USB connected, only Power and Eth but Power is from USB-C...

rom289, 2020-05-24:
HI,
I have the same problem. No USB port used on RPi 4B 4GB (0.54.7)

Ysurac, 2020-05-24:
I still have a related issue in 0.55 beta...
Should be solved by fixing RPI speed, I'm following RPI official kernel issue and hope to fix it, else I will use this workaround for next release.

rom289, 2020-05-24:
A reboot seems to fix it. For the moment...