Ysurac/openmptcprouter #116: After a while only one Wan used - Corrected with a router reboot.

Maudissure, 2018-08-02:
## Expected Behavior
All WAN used for connections

## Actual Behavior
After some time, only one Wan is used for connections. The two Wan are correctly connected to the VPS server but all the traffic is handled only by one Wan.

Usually the master Wan is the one who lost traffic. 

## Steps to Reproduce the Problem

  1. Reboot the router
  2. Wait some hours (don't know how many)
  3. Start a download or streaming and check on OMR utility --> One WAN used only.

## Specifications

  - OpenMPTCProuter version: 0.29.1
  - OpenMPTCProuter VPS version: 0.41
  - OpenMPTCProuter platform: RPI3 B+

Ysurac, 2018-08-02:
You can try OpenMPTCProuter 0.30
What error do you have in OpenMPTCProuter status page ?

Maudissure, 2018-08-03:
Ok with version OMR 0.30/VPS 0.43.

I close.

closed 2018-08-03 by Maudissure