Ysurac/openmptcprouter #117: Add VMware Tools

fnoir, 2018-08-03:
## Expected Behavior

VMware tools enabled

## Actual Behavior

If it possible to add VMware tools by default on VMware image

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.30 r7201-4f3082583a
  - OpenMPTCProuter platform: VMware ESXi 6.7

Ysurac, 2018-08-03:
It's difficult to add them by default, because vmware image is created from the raw image.
But I can add them as available package (if I can compile it). It's https://github.com/vmware/open-vm-tools ?

fnoir, 2018-08-03:
Yes, it's [https://github.com/vmware/open-vm-tools](https://github.com/vmware/open-vm-tools).

if it isn't possible to add them by default on VMware image, add open-vm-tools on available package is a good idea !

Ysurac, 2018-08-10:
It will be available when I will have a real working package. Should be ok next week.

fnoir, 2019-03-23:
I use open-vm-tools (v10.3.5-1) and 0 problem. Great job!