Ysurac/openmptcprouter #118: VPN tunnel down

SoGooDFR, 2018-08-03:
## Actual Behavior
In Status page:
- VPN tunnel down
- Multipath blocked on the connection

System logs:
```
Fri Aug  3 11:33:09 2018 daemon.notice netifd: omrvpn (9893): udhcpc: started, v1.28.3
Fri Aug  3 11:33:09 2018 daemon.notice netifd: omrvpn (9893): udhcpc: sending discover
Fri Aug  3 11:33:10 2018 daemon.warn odhcpd[3948]: A default route is present but there is no public prefix on eth0 thus we don't announce a default route!
Fri Aug  3 11:33:10 2018 user.notice post-tracking-post-tracking: tun0 switched off
Fri Aug  3 11:33:10 2018 user.notice post-tracking-post-tracking: omrvpn down
Fri Aug  3 11:33:12 2018 daemon.notice netifd: omrvpn (9893): udhcpc: sending discover
Fri Aug  3 11:33:15 2018 daemon.notice netifd: omrvpn (9893): udhcpc: sending discover
Fri Aug  3 11:34:45 2018 daemon.err uhttpd[4646]: Failed to establish connection
Fri Aug  3 11:35:05 2018 daemon.err uhttpd[4646]: Failed to establish connection
Fri Aug  3 11:35:25 2018 daemon.err uhttpd[4646]: Failed to establish connection
Fri Aug  3 11:35:45 2018 daemon.err uhttpd[4646]: Failed to establish connection
Fri Aug  3 11:36:05 2018 daemon.err uhttpd[4646]: Failed to establish connection
Fri Aug  3 11:36:25 2018 daemon.err uhttpd[4646]: Failed to establish connection
```

## Steps to Reproduce the Problem

  1. Fresh install on both
  2. Enter base configuration

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.30.2 r7202-7d15f96eaf
  - OpenMPTCProuter VPS version: < OpenMPTCProuter VPS 0.43 >
  - OpenMPTCProuter platform: x64 Intel NUC

Ysurac, 2018-08-03:
It's not a supported release... Only release available on https://www.openmptcprouter.com/download have "support" here.
I think you use Glorytun TCP, if multipath is blocked on the connection it will not work properly.

Maudissure, 2018-08-03:
I have this issue too, a Tun restart on interface resolve it.

fnoir, 2018-08-03:
I have this issue too.

With Glorytun TCP and Multipath TCP enabled=> stability problems
With Glorytun TCP and Multipath TCP disabled => OK

Ysurac, 2018-08-03:
What stability problems ? What do you have in log ? And what OpenMPTCProuter version ?

fnoir, 2018-08-03:
> What stability problems ?

For exemple, on Debian, i can't download package

> What do you have in log ?

Logs x50 with Glorytun TCP and Multipath TCP enabled : 

```
Fri Aug  3 15:46:35 2018 user.info : using tcp fast open
Fri Aug  3 15:46:35 2018 user.info : using tcp fast open
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6339]: initializing ciphers... chacha20
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6338]: initializing ciphers... chacha20
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6338]: listening at 0.0.0.0:1100
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6339]: listening at 0.0.0.0:1100
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6339]: tcp port reuse enabled
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6338]: tcp port reuse enabled
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6339]: UDP relay enabled
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6338]: UDP relay enabled
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6339]: udp port reuse enabled
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6338]: udp port reuse enabled
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6338]: running from root user
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6339]: running from root user
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6337]: initializing ciphers... chacha20
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6337]: listening at 0.0.0.0:1100
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6337]: tcp port reuse enabled
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6337]: UDP relay enabled
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6337]: udp port reuse enabled
Fri Aug  3 15:46:35 2018 daemon.info /usr/bin/ss-redir[6337]: running from root user
Fri Aug  3 15:46:35 2018 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Fri Aug  3 15:48:51 2018 daemon.err /usr/bin/ss-redir[6338]: send: Broken pipe
Fri Aug  3 15:58:45 2018 daemon.err /usr/bin/ss-redir[6338]: server recv: Operation timed out
Fri Aug  3 16:18:00 2018 daemon.err /usr/bin/ss-redir[6338]: server recv: Operation timed out
Fri Aug  3 16:42:14 2018 daemon.err /usr/bin/ss-redir[6338]: server recv: Operation timed out
```

> And what OpenMPTCProuter version ?

openmptcprouter v0.30 r7201-4f3082583a (RPi or VMware ESXi)

Ysurac, 2018-08-03:
I don't like these *"mptcp_detect_mapping Mappings do not match!"*...
Can you try to enable in **Network**->**MPTCP**, **Multipath TCP checksum** ?

SoGooDFR, 2018-08-12:
Glorytun TCP = Work.
Glorytun UDP = Don't work.

closed 2018-09-11 by Ysurac