Ysurac/openmptcprouter #119: No Apple device can get IP thru DHCP server

Maudissure, 2018-08-03:
## Expected Behavior
Apple devices get IP address from DHCP server

## Actual Behavior
None of my Apple devices get IP from DHCP server (iPhone, iPad, Apple TV)

## Steps to Reproduce the Problem

## Specifications

  - OpenMPTCProuter version: 0.30
  - OpenMPTCProuter VPS version: 0.43
  - OpenMPTCProuter platform: RPI3B+

Maudissure, 2018-08-03:
Oh well , just found it !

It's linked to IPV6, if IPV6 is turned off in Openmptcprouter advanced configuration interface, no Apple device can get an IP address from the DHCP server.

Ysurac, 2018-08-03:
And devices other than Apple get IPv4 via DHCP ?

Maudissure, 2018-08-03:
Yes all other devices get their IP .

Ysurac, 2018-08-03:
You can try ```tcpdump -i eth0 -n port 67 or port 68```, on OpenMPTCProuter, this will monitor DHCP, and check if you see the iPad or Apple TV.

Maudissure, 2018-08-03:
Sorry, i had to revert back to my load balancing network, i removed the Openmptcp router.

For now, i 'can't make tests anymore.

closed 2018-08-03 by Ysurac

Maudissure, 2018-08-03:
I think you should leave this point open, even if i can't test it, maybe
some peoples could do it with their apple devices.

Cordialement,

Emmanuel GELINEAU.

rubaz100, 2020-03-21:
iPadOS works fine (works but not always just some connection freezes(?)
iPhone 7/iOS get a ip but dont have an internet connection. I dont know what to do.
Internet connection is not stable. Disconnected from the internet (not wifi or dhcp)

Ysurac, 2020-03-21:
If it's not in DHCP, then it's not related to this issue, open a new one.