Ysurac/openmptcprouter #120: Upload speed issue

fnoir, 2018-08-04:
## Actual Behavior

I have a lot of timeout, big latency (>10 seconds)...

## Logs

Fri Aug  3 18:15:14 2018 daemon.err /usr/bin/ss-redir[6225]: send: Connection reset by peer
Fri Aug  3 18:35:04 2018 daemon.err /usr/bin/ss-redir[6226]: server recv: Operation timed out
Fri Aug  3 19:20:16 2018 daemon.err /usr/bin/ss-redir[6225]: server recv: Operation timed out
Fri Aug  3 20:41:10 2018 daemon.err /usr/bin/ss-redir[6225]: server recv: Operation timed out
Fri Aug  3 22:02:57 2018 daemon.err /usr/bin/ss-redir[6227]: server recv: Operation timed out
Fri Aug  3 22:16:38 2018 daemon.err /usr/bin/ss-redir[6225]: server recv: Operation timed out
Fri Aug  3 22:33:05 2018 daemon.err /usr/bin/ss-redir[6225]: server recv: Operation timed out
Sat Aug  4 00:14:08 2018 daemon.notice netifd: omrvpn (3207): udhcpc: sending renew to 10.255.255.1
Sat Aug  4 00:14:08 2018 daemon.notice netifd: omrvpn (3207): udhcpc: lease of 10.255.255.2 obtained, lease time 43200
Sat Aug  4 01:33:51 2018 user.notice post-tracking-post-tracking: wan2 switched off
Sat Aug  4 01:33:51 2018 user.notice post-tracking-post-tracking: Remove server xxx.xxx.xxx.xxx route via wan2
Sat Aug  4 01:34:03 2018 user.notice post-tracking-post-tracking: Set server xxx.xxx.xxx.xxx route via wan2 metric 4
Sat Aug  4 01:34:03 2018 user.notice post-tracking-post-tracking: Multipath wan2 switched to on
Sat Aug  4 02:14:34 2018 daemon.err /usr/bin/ss-redir[6227]: server recv: Operation timed out
Sat Aug  4 02:22:38 2018 daemon.err /usr/bin/ss-redir[6227]: remote recv: Operation timed out
Sat Aug  4 05:42:57 2018 daemon.err /usr/bin/ss-redir[6226]: server recv: Operation timed out
Sat Aug  4 06:14:08 2018 daemon.notice netifd: omrvpn (3207): udhcpc: sending renew to 10.255.255.1
Sat Aug  4 06:18:09 2018 daemon.err /usr/bin/ss-redir[6225]: server recv: Operation timed out
Sat Aug  4 06:43:41 2018 daemon.err /usr/bin/ss-redir[6225]: server recv: Operation timed out
Sat Aug  4 09:14:08 2018 daemon.notice netifd: omrvpn (3207): udhcpc: sending renew to 10.255.255.1
Sat Aug  4 09:15:27 2018 daemon.err /usr/bin/ss-redir[6227]: server recv: Operation timed out
Sat Aug  4 09:29:31 2018 daemon.err /usr/bin/ss-redir[6226]: server recv: Operation timed out
Sat Aug  4 09:40:50 2018 daemon.err /usr/bin/ss-redir[6227]: remote recv: Connection refused
Sat Aug  4 10:00:48 2018 daemon.err /usr/bin/ss-redir[6225]: send: Connection reset by peer
Sat Aug  4 10:00:48 2018 daemon.err /usr/bin/ss-redir[6225]: send: Connection reset by peer
Sat Aug  4 10:02:51 2018 daemon.err /usr/bin/ss-redir[6225]: server recv: Operation timed out

## Specifications

  - OpenMPTCProuter version: OpenMPTCProuter v0.30
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS v0.43
  - OpenMPTCProuter platform: RPI3 or VMware

fnoir, 2018-08-04:
With router on VMware, latency is better (< 5 seconds)

Ysurac, 2018-08-05:
Timeout with shadowsocks is not a problem.
You can try Glorytun UDP for better latency.

Ysurac, 2018-08-14:
How do you measure latency ?

fnoir, 2018-08-14:
On Windows: curl -o NUL -s -w "Connect: %{time_connect}\nStart Transfer: %{time_starttransfer}\nTotal: %{time_total}\n"  https://...

Ysurac, 2018-08-14:
And what is the result if you run it on OpenMPTCProuter directly ?

Ysurac, 2018-08-19:
I'm able to reproduce this issue, so now I should be able to debug it :)

fnoir, 2018-08-20:
Great, good luck :D

fnoir, 2018-09-25:
Hi, any news ?

closed 2019-03-13 by Ysurac