Ysurac/openmptcprouter #121: Apple devices not detected on network

Maudissure, 2018-08-04:
## Expected Behavior

Connection of some Apple devices on network

## Actual Behavior

The DHCP server is shut down on OpenMPTCProuter, using another DHCP server on my network.

iPhone 6s : Get an IPV4, no internet connection

iPad Air 2 : Get an IPV4, no internet connection

Apple TV 4K : Ethernet not even detected on the device

## Steps to Reproduce the Problem

All other devices than Apple devices are fully functional.

All Apple devices normally work with another router.

All home devices are connected on the same switch (TPlink 24 ports) directly or by an Tplink Archer C9 Wifi router (which is used as DHCP server for testing)..

## Specifications

  - OpenMPTCProuter version: 0.30.4
  - OpenMPTCProuter VPS version: 0.43
  - OpenMPTCProuter platform: RPI3B+

Maudissure, 2018-08-05:
More testing with OMR DHCP and the iPhone  : 

root@OpenMPTCProuter:~# tcpdump -i eth0 -n port 67
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
14:54:33.077569 IP 192.168.1.2.67 > 192.168.1.129.68: BOOTP/DHCP, Reply, length 300
14:54:33.079237 IP 192.168.1.2.67 > 192.168.1.195.68: BOOTP/DHCP, Reply, length 300
14:54:33.079993 IP 192.168.1.2.67 > 192.168.1.195.68: BOOTP/DHCP, Reply, length 300
14:54:33.094913 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 04:03:d6:08:d3:93, length 300
14:54:33.096659 IP 192.168.1.2.67 > 192.168.1.129.68: BOOTP/DHCP, Reply, length 300
14:54:34.221617 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 04:03:d6:08:d3:93, length 300
14:54:38.649398 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from b4:8b:19:1b:4e:ee, length 300
14:54:38.815589 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 04:03:d6:08:d3:93, length 300
14:54:41.955496 IP 192.168.1.2.67 > 192.168.1.131.68: BOOTP/DHCP, Reply, length 300
14:54:41.969572 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 04:03:d6:08:d3:93, length 300
14:54:41.971358 IP 192.168.1.2.67 > 192.168.1.131.68: BOOTP/DHCP, Reply, length 300
14:54:43.096104 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 04:03:d6:08:d3:93, length 300
14:54:48.665717 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from b4:8b:19:1b:4e:ee, length 300
14:54:49.745073 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from b4:8b:19:1b:4e:ee, length 300
14:54:52.049860 IP 192.168.1.2.67 > 192.168.1.198.68: BOOTP/DHCP, Reply, length 300
14:54:52.051766 IP 192.168.1.2.67 > 192.168.1.198.68: BOOTP/DHCP, Reply, length 300
14:54:53.100479 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from b4:8b:19:1b:4e:ee, length 300

The iPhone get an IPV4 for 1 sec then lost connection, and again it get the next available IPV4 and lost it, and again, and again.

Ysurac, 2018-08-05:
What is the result if on **Interfaces**->**LAN**->**DHCP** you set a smaller lease time than 12h ?
And in **Advanced Setings** if you set **Force DHCP** ?

Maudissure, 2018-08-12:
Last version of OMR and VPS with fresh install.

Still same problems. 

1 - No Apple devices get an IP address from OMR DHCP server. The devices gets an IP for 1 or 2 secondes the lost it, then again gets another IP and lost again, and again, and again ...

2 - With manual IP settings, no internet connection for all Apple devices.

Ysurac, 2018-08-14:
Apple TV and iphone send DHCP request, the server answer with an IP that is declined by Apple TV and iPhone.
Try to set DHCP lease time to 1h.

Maudissure, 2018-08-14:
Lease time set to 1h, same behavior : 

Tue Aug 14 06:52:08 2018 daemon.info dnsmasq-dhcp[24841]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:52:08 2018 daemon.info dnsmasq-dhcp[24841]: DHCPOFFER(eth0) 192.168.1.135 d0:d2:b0:83:83:16
Tue Aug 14 06:52:08 2018 daemon.info dnsmasq-dhcp[24841]: DHCPREQUEST(eth0) 192.168.1.156 b4:8b:19:1b:4e:ee
Tue Aug 14 06:52:08 2018 daemon.info dnsmasq-dhcp[24841]: DHCPACK(eth0) 192.168.1.156 b4:8b:19:1b:4e:ee iPhone
Tue Aug 14 06:52:08 2018 daemon.info dnsmasq-dhcp[24841]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:52:08 2018 daemon.info dnsmasq-dhcp[24841]: DHCPOFFER(eth0) 192.168.1.135 d0:d2:b0:83:83:16
Tue Aug 14 06:52:08 2018 daemon.info dnsmasq-dhcp[24841]: DHCPREQUEST(eth0) 192.168.1.156 b4:8b:19:1b:4e:ee
Tue Aug 14 06:52:08 2018 daemon.info dnsmasq-dhcp[24841]: DHCPACK(eth0) 192.168.1.156 b4:8b:19:1b:4e:ee iPhone
Tue Aug 14 06:52:14 2018 daemon.info dnsmasq-dhcp[24841]: DHCPDECLINE(eth0) 192.168.1.156 b4:8b:19:1b:4e:ee
Tue Aug 14 06:52:28 2018 daemon.info dnsmasq-dhcp[24841]: DHCPDISCOVER(eth0) b4:8b:19:1b:4e:ee
Tue Aug 14 06:52:28 2018 daemon.info dnsmasq-dhcp[24841]: DHCPOFFER(eth0) 192.168.1.157 b4:8b:19:1b:4e:ee
Tue Aug 14 06:52:28 2018 daemon.info dnsmasq-dhcp[24841]: DHCPDISCOVER(eth0) b4:8b:19:1b:4e:ee
Tue Aug 14 06:52:28 2018 daemon.info dnsmasq-dhcp[24841]: DHCPOFFER(eth0) 192.168.1.157 b4:8b:19:1b:4e:ee
Tue Aug 14 06:52:28 2018 daemon.info dnsmasq-dhcp[24841]: DHCPDISCOVER(eth0) b4:8b:19:1b:4e:ee
Tue Aug 14 06:52:28 2018 daemon.info dnsmasq-dhcp[24841]: DHCPOFFER(eth0) 192.168.1.157 b4:8b:19:1b:4e:ee
Tue Aug 14 06:52:31 2018 daemon.info dnsmasq-dhcp[24841]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:52:31 2018 daemon.info dnsmasq-dhcp[24841]: DHCPOFFER(eth0) 192.168.1.136 d0:d2:b0:83:83:16
Tue Aug 14 06:52:31 2018 daemon.info dnsmasq-dhcp[24841]: DHCPREQUEST(eth0) 192.168.1.157 b4:8b:19:1b:4e:ee
Tue Aug 14 06:52:31 2018 daemon.info dnsmasq-dhcp[24841]: DHCPACK(eth0) 192.168.1.157 b4:8b:19:1b:4e:ee iPhone
Tue Aug 14 06:52:31 2018 daemon.info dnsmasq-dhcp[24841]: DHCPREQUEST(eth0) 192.168.1.157 b4:8b:19:1b:4e:ee
Tue Aug 14 06:52:31 2018 daemon.info dnsmasq-dhcp[24841]: DHCPACK(eth0) 192.168.1.157 b4:8b:19:1b:4e:ee iPhone
Tue Aug 14 06:52:31 2018 daemon.info dnsmasq-dhcp[24841]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:52:31 2018 daemon.info dnsmasq-dhcp[24841]: DHCPOFFER(eth0) 192.168.1.136 d0:d2:b0:83:83:16
Tue Aug 14 06:52:32 2018 user.notice post-tracking-post-tracking: Multipath wan1 switched to on
Tue Aug 14 06:52:32 2018 daemon.info dnsmasq-dhcp[24841]: DHCPREQUEST(eth0) 192.168.1.136 d0:d2:b0:83:83:16
Tue Aug 14 06:52:32 2018 daemon.info dnsmasq-dhcp[24841]: DHCPACK(eth0) 192.168.1.136 d0:d2:b0:83:83:16 Apple-TV
Tue Aug 14 06:52:37 2018 daemon.info dnsmasq-dhcp[24841]: DHCPDECLINE(eth0) 192.168.1.157 b4:8b:19:1b:4e:ee
Tue Aug 14 06:52:38 2018 daemon.info dnsmasq-dhcp[24841]: DHCPDECLINE(eth0) 192.168.1.136 d0:d2:b0:83:83:16

Maudissure, 2018-08-21:
OMR V0.32.3

I have some news, i forced the IPV4 address in OMR DHCP config and now get this behavior : 

Tue Aug 21 11:43:12 2018 daemon.info dnsmasq-dhcp[3820]: DHCPACK(eth0) 192.168.1.113 00:04:4b:73:3e:b3 Shield_TV
Tue Aug 21 11:43:15 2018 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Tue Aug 21 11:43:15 2018 user.notice post-tracking-post-tracking: Set server 51.254.127.83 route via wan1 metric 1
Tue Aug 21 11:43:57 2018 daemon.info dnsmasq-dhcp[3820]: DHCPREQUEST(eth0) 192.168.1.54 d0:d2:b0:83:83:16
Tue Aug 21 11:43:57 2018 daemon.info dnsmasq-dhcp[3820]: DHCPACK(eth0) 192.168.1.54 d0:d2:b0:83:83:16 Apple_TV
Tue Aug 21 11:43:58 2018 daemon.info dnsmasq-dhcp[3820]: DHCPDECLINE(eth0) 192.168.1.54 d0:d2:b0:83:83:16
Tue Aug 21 11:43:58 2018 daemon.warn dnsmasq-dhcp[3820]: disabling DHCP static address 192.168.1.54 for 10m
Tue Aug 21 11:44:08 2018 daemon.warn dnsmasq-dhcp[3820]: not using configured address 192.168.1.54 because it was previously declined
Tue Aug 21 11:44:11 2018 daemon.info dnsmasq-dhcp[3820]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 21 11:44:11 2018 daemon.info dnsmasq-dhcp[3820]: DHCPOFFER(eth0) 192.168.1.208 d0:d2:b0:83:83:16
Tue Aug 21 11:44:11 2018 daemon.warn dnsmasq-dhcp[3820]: not using configured address 192.168.1.54 because it was previously declined
Tue Aug 21 11:44:11 2018 daemon.info dnsmasq-dhcp[3820]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 21 11:44:11 2018 daemon.info dnsmasq-dhcp[3820]: DHCPOFFER(eth0) 192.168.1.208 d0:d2:b0:83:83:16
Tue Aug 21 11:44:12 2018 daemon.info dnsmasq-dhcp[3820]: DHCPREQUEST(eth0) 192.168.1.208 d0:d2:b0:83:83:16
Tue Aug 21 11:44:12 2018 daemon.info dnsmasq-dhcp[3820]: DHCPACK(eth0) 192.168.1.208 d0:d2:b0:83:83:16 Apple_TV
Tue Aug 21 11:44:12 2018 daemon.warn dnsmasq-dhcp[3820]: not giving name Apple_TV.lan to the DHCP lease of 192.168.1.208 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.54
Tue Aug 21 11:44:14 2018 daemon.info dnsmasq-dhcp[3820]: DHCPDECLINE(eth0) 192.168.1.208 d0:d2:b0:83:83:16
Tue Aug 21 11:45:07 2018 daemon.warn dnsmasq-dhcp[3820]: not using configured address 192.168.1.54 because it was previously declined
Tue Aug 21 11:45:11 2018 daemon.info dnsmasq-dhcp[3820]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 21 11:45:11 2018 daemon.info dnsmasq-dhcp[3820]: DHCPOFFER(eth0) 192.168.1.209 d0:d2:b0:83:83:16
Tue Aug 21 11:45:11 2018 daemon.warn dnsmasq-dhcp[3820]: not using configured address 192.168.1.54 because it was previously declined
Tue Aug 21 11:45:11 2018 daemon.info dnsmasq-dhcp[3820]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 21 11:45:11 2018 daemon.info dnsmasq-dhcp[3820]: DHCPOFFER(eth0) 192.168.1.209 d0:d2:b0:83:83:16
Tue Aug 21 11:45:12 2018 daemon.info dnsmasq-dhcp[3820]: DHCPREQUEST(eth0) 192.168.1.209 d0:d2:b0:83:83:16
Tue Aug 21 11:45:12 2018 daemon.info dnsmasq-dhcp[3820]: DHCPACK(eth0) 192.168.1.209 d0:d2:b0:83:83:16 Apple_TV
Tue Aug 21 11:45:12 2018 daemon.warn dnsmasq-dhcp[3820]: not giving name Apple_TV.lan to the DHCP lease of 192.168.1.209 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.54
Tue Aug 21 11:45:13 2018 daemon.info dnsmasq-dhcp[3820]: DHCPDECLINE(eth0) 192.168.1.209 d0:d2:b0:83:83:16
Tue Aug 21 11:45:19 2018 daemon.err /usr/bin/ss-redir[7166]: remote recv: Operation timed out
Tue Aug 21 11:45:23 2018 daemon.warn dnsmasq-dhcp[3820]: not using configured address 192.168.1.54 because it was previously declined
Tue Aug 21 11:45:27 2018 daemon.info dnsmasq-dhcp[3820]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 21 11:45:27 2018 daemon.info dnsmasq-dhcp[3820]: DHCPOFFER(eth0) 192.168.1.210 d0:d2:b0:83:83:16
Tue Aug 21 11:45:27 2018 daemon.warn dnsmasq-dhcp[3820]: not using configured address 192.168.1.54 because it was previously declined
Tue Aug 21 11:45:27 2018 daemon.info dnsmasq-dhcp[3820]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 21 11:45:27 2018 daemon.info dnsmasq-dhcp[3820]: DHCPOFFER(eth0) 192.168.1.210 d0:d2:b0:83:83:16
Tue Aug 21 11:45:28 2018 daemon.info dnsmasq-dhcp[3820]: DHCPREQUEST(eth0) 192.168.1.210 d0:d2:b0:83:83:16
Tue Aug 21 11:45:28 2018 daemon.info dnsmasq-dhcp[3820]: DHCPACK(eth0) 192.168.1.210 d0:d2:b0:83:83:16 Apple_TV
Tue Aug 21 11:45:28 2018 daemon.warn dnsmasq-dhcp[3820]: not giving name Apple_TV.lan to the DHCP lease of 192.168.1.210 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.54
Tue Aug 21 11:46:17 2018 daemon.warn dnsmasq-dhcp[3820]: not using configured address 192.168.1.54 because it was previously declined
Tue Aug 21 11:46:17 2018 daemon.info dnsmasq-dhcp[3820]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 21 11:46:17 2018 daemon.info dnsmasq-dhcp[3820]: DHCPOFFER(eth0) 192.168.1.210 d0:d2:b0:83:83:16
Tue Aug 21 11:46:18 2018 daemon.info dnsmasq-dhcp[3820]: DHCPREQUEST(eth0) 192.168.1.210 d0:d2:b0:83:83:16
Tue Aug 21 11:46:18 2018 daemon.info dnsmasq-dhcp[3820]: DHCPACK(eth0) 192.168.1.210 d0:d2:b0:83:83:16 Apple_TV
Tue Aug 21 11:46:18 2018 daemon.warn dnsmasq-dhcp[3820]: not giving name Apple_TV.lan to the DHCP lease of 192.168.1.210 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.54
Tue Aug 21 11:46:24 2018 daemon.info dnsmasq-dhcp[3820]: DHCPDECLINE(eth0) 192.168.1.210 d0:d2:b0:83:83:16
Tue Aug 21 11:46:34 2018 daemon.warn dnsmasq-dhcp[3820]: not using configured address 192.168.1.54 because it was previously declined
[... cut ...]
Tue Aug 21 11:47:48 2018 daemon.info dnsmasq-dhcp[3820]: DHCPOFFER(eth0) 192.168.1.212 d0:d2:b0:83:83:16
Tue Aug 21 11:47:49 2018 daemon.info dnsmasq-dhcp[3820]: DHCPREQUEST(eth0) 192.168.1.212 d0:d2:b0:83:83:16
Tue Aug 21 11:47:49 2018 daemon.info dnsmasq-dhcp[3820]: DHCPACK(eth0) 192.168.1.212 d0:d2:b0:83:83:16 Apple_TV
Tue Aug 21 11:47:49 2018 daemon.warn dnsmasq-dhcp[3820]: not giving name Apple_TV.lan to the DHCP lease of 192.168.1.212 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.54

Ysurac, 2018-08-21:
Not really, DHCP server offer IP 192.168.1.54 to Apple TV but Apple TV decline this address, so the address is banned for 10 minutes. Then the problem is that 192.168.1.54 is already affected to Apple TV that request an IP and can't be affected, so the same error for 10 minutes.

The first problem is why Apple TV refuse 192.168.1.54 ? If you ping the IP there is no answer ?

Maudissure, 2018-08-21:
I make the test as soon as i return at home.

Ysurac, 2018-08-26:
Same question:  "The first problem is why Apple TV refuse 192.168.1.54 ? If you ping the IP there is no answer ?"

Maudissure, 2018-08-26:
No, no answer.

Maudissure, 2019-06-09:
Nope, nothing new. 

I tried with OMR V0.47 and VPS 0.994.

Same results with apple devices. DHCP not working and : 

   - Iphone Xr : works only with fixed IP on OMR
   - iPad Air 2 : Works only with fixed IP on one of the modem.
   - Apple TV : Works only with fixed IP on one of the modem (So, no local lan connection ...)

And company computer lost internet connection after some time even with OMR bypass on ADSL modem.

So still a no go for me.

I'm sure this is a DNS problem but nothing on logs, so i will stay on my multi Wan solution. Too bad.

Maudissure, 2019-07-21:
Ok, Tried last version (V0.51 with VPS V0.9962

Sun Jul 21 17:29:12 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Sun Jul 21 17:29:12 2019 daemon.info dnsmasq-dhcp[24229]: DHCPOFFER(eth0) 192.168.1.170 d0:d2:b0:83:83:16
Sun Jul 21 17:29:12 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Sun Jul 21 17:29:12 2019 daemon.info dnsmasq-dhcp[24229]: DHCPOFFER(eth0) 192.168.1.170 d0:d2:b0:83:83:16
Sun Jul 21 17:29:13 2019 daemon.info dnsmasq-dhcp[24229]: DHCPREQUEST(eth0) 192.168.1.170 d0:d2:b0:83:83:16
Sun Jul 21 17:29:13 2019 daemon.info dnsmasq-dhcp[24229]: DHCPACK(eth0) 192.168.1.170 d0:d2:b0:83:83:16 Apple-TV-2
Sun Jul 21 17:29:18 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDECLINE(eth0) 192.168.1.170 d0:d2:b0:83:83:16
Sun Jul 21 17:29:24 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDISCOVER(eth0) f4:06:16:89:8b:cc
Sun Jul 21 17:29:24 2019 daemon.info dnsmasq-dhcp[24229]: DHCPOFFER(eth0) 192.168.1.172 f4:06:16:89:8b:cc
Sun Jul 21 17:29:24 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDISCOVER(eth0) f4:06:16:89:8b:cc
Sun Jul 21 17:29:24 2019 daemon.info dnsmasq-dhcp[24229]: DHCPOFFER(eth0) 192.168.1.172 f4:06:16:89:8b:cc
Sun Jul 21 17:29:25 2019 daemon.info dnsmasq-dhcp[24229]: DHCPREQUEST(eth0) 192.168.1.172 f4:06:16:89:8b:cc
Sun Jul 21 17:29:25 2019 daemon.info dnsmasq-dhcp[24229]: DHCPACK(eth0) 192.168.1.172 f4:06:16:89:8b:cc iPhoneauMaudit
Sun Jul 21 17:29:29 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDECLINE(eth0) 192.168.1.172 f4:06:16:89:8b:cc
Sun Jul 21 17:29:31 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Sun Jul 21 17:29:31 2019 daemon.info dnsmasq-dhcp[24229]: DHCPOFFER(eth0) 192.168.1.172 d0:d2:b0:83:83:16
Sun Jul 21 17:29:32 2019 daemon.info dnsmasq-dhcp[24229]: DHCPREQUEST(eth0) 192.168.1.172 d0:d2:b0:83:83:16
Sun Jul 21 17:29:32 2019 daemon.info dnsmasq-dhcp[24229]: DHCPACK(eth0) 192.168.1.172 d0:d2:b0:83:83:16 Apple-TV-2
Sun Jul 21 17:29:36 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDECLINE(eth0) 192.168.1.172 d0:d2:b0:83:83:16
Sun Jul 21 17:29:57 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Sun Jul 21 17:29:57 2019 daemon.info dnsmasq-dhcp[24229]: DHCPOFFER(eth0) 192.168.1.173 d0:d2:b0:83:83:16
Sun Jul 21 17:29:57 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Sun Jul 21 17:29:57 2019 daemon.info dnsmasq-dhcp[24229]: DHCPOFFER(eth0) 192.168.1.173 d0:d2:b0:83:83:16
Sun Jul 21 17:29:58 2019 daemon.info dnsmasq-dhcp[24229]: DHCPREQUEST(eth0) 192.168.1.173 d0:d2:b0:83:83:16
Sun Jul 21 17:29:58 2019 daemon.info dnsmasq-dhcp[24229]: DHCPACK(eth0) 192.168.1.173 d0:d2:b0:83:83:16 Apple-TV-2
Sun Jul 21 17:30:04 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDECLINE(eth0) 192.168.1.173 d0:d2:b0:83:83:16
Sun Jul 21 17:30:18 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Sun Jul 21 17:30:18 2019 daemon.info dnsmasq-dhcp[24229]: DHCPOFFER(eth0) 192.168.1.174 d0:d2:b0:83:83:16
Sun Jul 21 17:30:18 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Sun Jul 21 17:30:18 2019 daemon.info dnsmasq-dhcp[24229]: DHCPOFFER(eth0) 192.168.1.174 d0:d2:b0:83:83:16
Sun Jul 21 17:30:19 2019 daemon.info dnsmasq-dhcp[24229]: DHCPREQUEST(eth0) 192.168.1.174 d0:d2:b0:83:83:16
Sun Jul 21 17:30:19 2019 daemon.info dnsmasq-dhcp[24229]: DHCPACK(eth0) 192.168.1.174 d0:d2:b0:83:83:16 Apple-TV-2
Sun Jul 21 17:30:28 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDISCOVER(eth0) f4:06:16:89:8b:cc
Sun Jul 21 17:30:28 2019 daemon.info dnsmasq-dhcp[24229]: DHCPOFFER(eth0) 192.168.1.175 f4:06:16:89:8b:cc
Sun Jul 21 17:30:28 2019 daemon.info dnsmasq-dhcp[24229]: DHCPDISCOVER(eth0) f4:06:16:89:8b:cc
Sun Jul 21 17:30:28 2019 daemon.info dnsmasq-dhcp[24229]: DHCPOFFER(eth0) 192.168.1.175 f4:06:16:89:8b:cc

Ysurac, 2019-07-21:
Nothing I can do if Apple TV decline IP offered ;)
It decline .172 and .173. Seems to accept .174 ? If these IP are not already used then Apple TV have a problem somewhere.

Maudissure, 2019-07-28:
So why all my Apple devices works on my TP-link TL-480T+ without a problem  ?

Ysurac, 2019-07-28:
Because it doesn't use dnsmasq or a RFC conformant DHCP server ?
Some Apple device are know to no be RFC conformant. If you find a config that work with dnsmasq I can apply it, else I can do nothing.

Maudissure, 2019-09-05:
Is it possible to turn off OMR DHCP and use a separate DHCP server with same subnet settings and still get MPTCP working ?

Ysurac, 2019-09-05:
yes

closed 2019-12-20 by Ysurac

Neustradamus, 2019-12-20:
@Ysurac: It is solved?

Maudissure, 2020-01-30:
No , it's not.

This solution will never work with apple devices.

Ysurac, 2020-01-30:
The issue is related to iOS and dnsmasq that is used in OpenMPTCProuter, OpenWrt and many routers.
I don't have any iOS devices so I can't make any tests