Ysurac/openmptcprouter #122: shadowsocks not working

lapinus57, 2018-08-04:
I installed openrouter on my intel nuc (NUC7i5BNK) and openrouter vps on my VPS 1and1, and in the aprèçu of the networks it is noted shadows not working.

## Specifications

  - OpenMPTCProuter version: (actual version)
  - OpenMPTCProuter VPS version: (actual version)
  - OpenMPTCProuter platform: (x86_64)

lapinus57, 2018-08-05:
OpenMPTCProuter version v0.30.4 r7204-8be3af93b4
OpenMPTCProuter VPS version 0.43

sudo systemctl status shadowsocks-libev
shadowsocks-libev.service - shadowsocks-libev default server servvice
 Loaded : loaded (/lib/systemd/shadowsocks-libev.service; disabled; vendor preset: enabled)
 Active : failed (result : exit-code) since .... 3s ago
   Docs : man:shadowsocks-libdev(8)
  Process: 3504 ExecStart= /usr/bin/ss-server - c $CONFFILE $DEMON_ARGS (code=exited, status=255)
  Main PID 3504 (code=exited, status=255)

- localhost systemd[1] : started shadowsocks-libev Default Server Service.
- localhost ss-sserver[3504] usuing tcp fast open
- localhost systemd shadowsocks-libev.service : main process exited, code=exited, status=255/n/a
- localhost systemd shadowsockeds-libev.service : Unit entred failed state
- localhost systemd shadowsockeds-libev.service : failed with result 'exit-code'

for install vps i have used this link : http://www.openmptcprouter.com/server/debian9-x86_64.sh

Ysurac, 2018-08-05:
Set verbose to 2 in ```/etc/shadowsocks-libev/config.json``` and try to run it manually: ```ss-server -c /etc/shadowsocks-libev/config.json```.

lapinus57, 2018-08-05:
[image]

i have try and no change

lapinus57, 2018-08-05:
I have change /etc/shadowsocks-libev/config.json
  "server_port":65101,   to   "server_port":65111,
add "local_adress":"127.0.0.1"
add "nameserver":"8.8.8.8"

root@localhost:~# systemctl status shadowsocks-libev
● shadowsocks-libev.service - Shadowsocks-libev Default Server Service
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-08-05 13:11:13 UTC; 8s ago
     Docs: man:shadowsocks-libev(8)
 Main PID: 12416 (ss-server)
   CGroup: /system.slice/shadowsocks-libev.service
           └─12416 /usr/bin/ss-server -c /etc/shadowsocks-libev/config.json -u --fast-open

Aug 05 13:11:13 localhost ss-server[12416]:  2018-08-05 13:11:13 INFO: initializing ciphers... chacha20
Aug 05 13:11:13 localhost ss-server[12416]:  2018-08-05 13:11:13 INFO: using nameserver: 8.8.8.8
Aug 05 13:11:13 localhost ss-server[12416]:  2018-08-05 13:11:13 INFO: tcp server listening at [::0]:65111
Aug 05 13:11:13 localhost ss-server[12416]:  2018-08-05 13:11:13 INFO: tcp port reuse enabled
Aug 05 13:11:13 localhost ss-server[12416]:  2018-08-05 13:11:13 INFO: tcp server listening at 0.0.0.0:65111
Aug 05 13:11:13 localhost ss-server[12416]:  2018-08-05 13:11:13 INFO: tcp port reuse enabled
Aug 05 13:11:13 localhost ss-server[12416]:  2018-08-05 13:11:13 INFO: udp server listening at [::0]:65111
Aug 05 13:11:13 localhost ss-server[12416]:  2018-08-05 13:11:13 INFO: udp port reuse enabled
Aug 05 13:11:13 localhost ss-server[12416]:  2018-08-05 13:11:13 INFO: udp server listening at 0.0.0.0:65111
Aug 05 13:11:13 localhost ss-server[12416]:  2018-08-05 13:11:13 INFO: udp port reuse enabled

Ysurac, 2018-08-05:
Shadowsocks must be on 65101 to work with default OpenMPTCProuter config.
Can you run a ```netstat -lapute``` to check if something else is not running on port 65101.
To run shadowsocks, it's not shadowsocks-libev service, it's ```shadowsocks-libev-server@config```

lapinus57, 2018-08-05:
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       User       Inode      PID/Program name
tcp        0      0 0.0.0.0:65101           0.0.0.0:*               LISTEN      root       6753       590/ss-server
tcp        0      0 [::]:65101              [::]:*                  LISTEN      root       6751       590/ss-server
udp        0      0 0.0.0.0:65101           0.0.0.0:*                           root       6757       590/ss-server
udp        0      0 0.0.0.0:65111           0.0.0.0:*                           nobody     58945      12416/ss-server
udp6       0      0 [::]:65101              [::]:*                              root       6755       590/ss-server
udp6       0      0 [::]:65111              [::]:*                              nobody     58943      12416/ss-server

● shadowsocks-libev-server@config.service - Shadowsocks-Libev Custom Server Service for config
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-server@.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-08-05 13:43:26 UTC; 20s ago
     Docs: man:ss-server(1)
  Process: 18171 ExecStart=/usr/bin/ss-server -c /etc/shadowsocks-libev/config.json (code=exited, status=255)
 Main PID: 18171 (code=exited, status=255)

Aug 05 13:43:26 localhost ss-server[18171]:  2018-08-05 13:43:26 INFO: UDP relay enabled
Aug 05 13:43:26 localhost ss-server[18171]:  2018-08-05 13:43:26 INFO: initializing ciphers... chacha20
Aug 05 13:43:26 localhost ss-server[18171]:  2018-08-05 13:43:26 INFO: using nameserver: 8.8.8.8
Aug 05 13:43:26 localhost ss-server[18171]:  2018-08-05 13:43:26 INFO: tcp server listening at [::0]:65111
Aug 05 13:43:26 localhost ss-server[18171]:  2018-08-05 13:43:26 INFO: tcp port reuse enabled
Aug 05 13:43:26 localhost ss-server[18171]:  2018-08-05 13:43:26 ERROR: bind: Address already in use
Aug 05 13:43:26 localhost ss-server[18171]:  2018-08-05 13:43:26 ERROR: bind() error

Ysurac, 2018-08-05:
Check key on both side, check if chacha20 is set on both side, if it's ok you can try to desactivate fastopen.

lapinus57, 2018-08-06:
no change shadowsocks not working

lapinus57, 2018-08-07:
on my VPS
{
    "server":["[::0]", "0.0.0.0"],
    "server_port":65101,
    "local_adress":"127.0.0.1",
    "local_port":1081,
    "nameserver":"8.8.8.8",
    "mode":"tcp_and_udp",
    "key":"*****************************",
    "timeout":600,
    "method":"chacha20",
    "verbose":2,
    "prefer_ipv6": true,
    "fast_open":false,
    "reuse_port": true,
    "mptcp": true
}

Ysurac, 2018-08-07:
No idea, but you should try a fresh install after all this not working changes.

lapinus57, 2018-08-07:
after trying a fresh install on both devices and reset shadowsocks key,
i dont get shadowsocks is not working :) , now i get no ipv6 access.
One problem are solved, new problem poping.
In my old key i had "_" and after looking in my local router "_" are change with "/".
To get a new key, I first try to change "_" to "/" in my old key on my vps, and I have an error. With the restart of systemctl and the status I saw this error and view the new key and I use it.

Now For ipv6 (in french Grrrrrrrrrrrrrrrrrrrrrr!!!!!!)

Ysurac, 2018-08-07:
Key in ```/root/openmptcprouter_config.txt``` should be in correct format for OpenMPTCProuter. But in ShadowSocks is not exactly the same. I will check if I didn't make an error somewhere.

The error mean that IPv6 tunnel is working, but can't go to the Internet at VPS end. IPv6 is configured on the VPS ?

(English is good and at least this can help everybody who have same problem)

lapinus57, 2018-08-07:
après avoir comparé la clef dans le message de fin d'installation et celle dans /etc/shadowsocks-libev/config.json, "-" a était remplacé par "/".
Dans openmptcprouter en remplacant ":" par "_" la clef devient invalide

lapinus57, 2018-08-07:
after comparing the key in the end-of-installation message with the one in /etc/shadowsocks-libev/config.json, "-" a was replaced with "/".
In openmptcprouter replacing ":" with "_" the key becomes invalid

lapinus57, 2018-08-07:
in my management website i'nt have option with ipv6  i can get just an other ipv4 adress

Ysurac, 2018-08-07:
What is the provider ? There is really VPS without IPv6 today ?
You can disable IPv6 in **System**->**OpenMPTCProuter**->**Advanced settings** or only disable *Router Advertisement-Service* in LAN interface DHCP.

lapinus57, 2018-08-08:
I tried to disable IPv6 and the message does not disappear

Ysurac, 2018-08-08:
True, I will remove this warning when IPv6 is disabled in next release.

closed 2018-08-08 by Ysurac