Ysurac/openmptcprouter #123: Router stability issues

Maudissure, 2018-08-06:
## Expected Behavior

router's 24/24 working.

## Actual Behavior

Router crash after some hours.

No more internet connection, no more access to interface, no more SSH access.

Reboot the router solve problem.

## Steps to Reproduce the Problem

  1. Boot the router
  2. Using network (Browsing, Torrent download, Netflix, Molotov).
  3. Network crash
  4. Reboot

## Specifications

  - Three WAN used (1 ADSL, 2 LTE)
  - OpenMPTCProuter version: 0.30.4
  - OpenMPTCProuter VPS version: 0.43
  - OpenMPTCProuter platform: RPI3B+

Maudissure, 2018-08-06:
No USB linked on the RPI3B+ .
Actual Temp. (after about 2h of use) : 54°C

Ysurac, 2018-08-06:
You should monitor temperature and check that your power supply can deliver at least 2.5A.
For me it's an hardware problem, not related to OpenMPTCProuter.

The only to check why it is crashing is using a hdmi monitor or a console cable.

Ysurac, 2018-08-13:
OpenMPTCProuter v0.31.1 should fix problems related to RPI ethernet port.

Maudissure, 2018-08-13:
Great ! I test that as soon as possible.

Maudissure, 2018-08-13:
Internet instability still there but seems more stable with Router V0.31.1.

After 3 hours i have a RPI3b+ temp à 50.5°C.

Network connection lost only on IPV4, IPV6 seems stable.

Reboot the router fixed the problem, until next IPV4 crash.

Kinda bad handled memory ?

Ysurac, 2018-08-13:
Without log info I can't say anything ;)
What do you call an IPv4 crash ?

Maudissure, 2018-08-13:
I leaved network infos opened on two PC during these 3 hours of test.

After each network crash i noticed that was the IPV4 connectivity that was down. IPV6 seemed to be up.

Sorry for the lack of technicals informations i'm very not familiar with linux systems.

Vinc32du44, 2018-11-26:
Hello,

Same for me ... After few hours ( i think about lease time of DHCP bail) RPi crash ???
Restart Rpi and all seemed to be OK, all the devices recover an IP address but after a long time it is not immediate.
Why is DHCP not stable and never crash ?
Thank you for your answers

Ysurac, 2018-11-26:
You checked power supply and CPU temperature ? 
When you have a problem, how do you check if RPI is crashed ? Can you ping it using static IPs ? 
What is your client IP/gateway when you have a problem ?

Without any log this will be difficult to check where is your problem.

closed 2019-03-13 by Ysurac