Ysurac/openmptcprouter #125: Wireless/WiFi is not detected

afriza, 2018-08-08:
## Expected Behavior
  - Network > Wireless is there
  - `uci show wireless` shows something
  - WiFi is able to scan and connect
  - under overview, the following is displayed
[image]

## Actual Behavior
  - Network > Wireless is missing on a cleanly formatted & flashed OpenMPTCProuter
  - `uci show wireless` shows `uci: Entry not found`
  -  when flashing v0.30.4 over 0.27 (last known with working WiFi), WiFi is not able to scan nor connect
    - the following is displayed under overview
[image]

## Steps to Reproduce the Problem

  1. Flash OpenMPTCProuter version 0.29, 0.29.1, 0.30, 0.30.3 or 0.30.4
  2. try to connect to WiFi or do anything related to wireless

## Specifications

  - OpenMPTCProuter version: (0.29, 0.29.1, 0.30, 0.30.3, 0.30.4)
  - OpenMPTCProuter VPS version: (-)
  - OpenMPTCProuter platform: (RPI3B+)

Ysurac, 2018-08-10:
Fixed in OpenMPTCProuter v0.31

closed 2018-08-10 by Ysurac

FunkeDope, 2019-01-23:
Wifi doesnt work at all on the latest build again. Rpi2 x86.