Ysurac/openmptcprouter #127: ESXi vitualisation problem

Ajowan84, 2018-08-09:
Hi,

I'm testing OpenMPTCProuter on VmWare workstation and on ESXi.
For the test, I used only a 4G modem.
On Workstation OpenMPTCProuter works perfectly but on ESXi in modem overview I have the "Gateway Down" error for my 4G modem.
Is there any extra configuration to do on ESXi?

Thank you in advance

  - OpenMPTCProuter version:  v0.29.1 r7171-962e86d9af
  - OpenMPTCProuter VPS version: 0.39
  - OpenMPTCProuter platform: vmware

Ysurac, 2018-08-09:
I think you need to set interfaces security as promiscuous if you keep default configuration.

Ajowan84, 2018-08-09:
Thank you for your prompt response.
Where is interfaces security ?

Ysurac, 2018-08-09:
No idea. I don't use vmware only Virtualbox. You can maybe use info from here: https://github.com/Ysurac/openmptcprouter/issues/87

Ajowan84, 2018-08-09:
I found it meanwhile.
In vSwitch0 parameters, security tab

#87 is the good solution !!!

Thank you again, i like your work.