Ysurac/openmptcprouter #130: UI bug ? Or not ?

Maudissure, 2018-08-13:
## Expected Behavior

  Same informations on status page.

## Actual Behavior

**Status page after a first loading :** 

[image]

**And after refresh :** 

[image]

## Steps to Reproduce the Problem

  1. First load status page
  2. Next refresh (or so)
  3. Status Ok after one or two refresh
  4. Then go again

## Specifications

  - OpenMPTCProuter version: 0.31.1
  - OpenMPTCProuter VPS version: 0.44
  - OpenMPTCProuter platform: RPI3B+

Ysurac, 2018-08-13:
No error here. You have a problem. Check log.

Maudissure, 2018-08-13:
How do i do that ?

Ysurac, 2018-08-13:
In **status**->**System log**

Maudissure, 2018-08-13:
Strange, nothing after 16h37:51 : 

Mon Aug 13 16:37:21 2018 kern.warn kernel: [ 2483.815681] [<ffffff8008083fe0>] ret_from_fork+0x10/0x18
Mon Aug 13 16:37:23 2018 daemon.info dnsmasq-dhcp[5219]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Mon Aug 13 16:37:23 2018 daemon.info dnsmasq-dhcp[5219]: DHCPOFFER(eth0) 192.168.1.171 d0:d2:b0:83:83:16
Mon Aug 13 16:37:23 2018 daemon.info dnsmasq-dhcp[5219]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Mon Aug 13 16:37:23 2018 daemon.info dnsmasq-dhcp[5219]: DHCPOFFER(eth0) 192.168.1.171 d0:d2:b0:83:83:16
Mon Aug 13 16:37:24 2018 daemon.info dnsmasq-dhcp[5219]: DHCPREQUEST(eth0) 192.168.1.171 d0:d2:b0:83:83:16
Mon Aug 13 16:37:24 2018 daemon.info dnsmasq-dhcp[5219]: DHCPACK(eth0) 192.168.1.171 d0:d2:b0:83:83:16 Apple-TV
Mon Aug 13 16:37:51 2018 user.notice post-tracking-post-tracking: Multipath wan1 switched to on

Ysurac, 2018-08-13:
The *kern.warn* is strange. Can I have what is before that ?

Maudissure, 2018-08-13:
yes of course : 

Mon Aug 13 16:36:08 2018 kern.warn kernel: [ 2410.085218] 3f60: ffffff8008803f70 ffffff8008084718
Mon Aug 13 16:36:08 2018 kern.warn kernel: [ 2410.092523] [<ffffff8008082870>] el1_irq+0xb0/0x140
Mon Aug 13 16:36:08 2018 kern.warn kernel: [ 2410.099824] [<ffffff8008084718>] arch_cpu_idle+0x10/0x18
Mon Aug 13 16:36:08 2018 kern.warn kernel: [ 2410.107508] [<ffffff80080c6364>] do_idle+0xec/0x1b8
Mon Aug 13 16:36:08 2018 kern.warn kernel: [ 2410.114684] [<ffffff80080c6570>] cpu_startup_entry+0x20/0x28
Mon Aug 13 16:36:08 2018 kern.warn kernel: [ 2410.122598] [<ffffff800808cbcc>] secondary_start_kernel+0x17c/0x188
Mon Aug 13 16:36:11 2018 kern.err kernel: [ 2413.308466] eth0: hw csum failure
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.313980] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.62 #0
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.322153] Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.330556] Call trace:
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.335000] [<ffffff8008086bf0>] dump_backtrace+0x0/0x3b0
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.342437] [<ffffff8008086fb4>] show_stack+0x14/0x20
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.349453] [<ffffff8008546db8>] dump_stack+0x9c/0xbc
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.356389] [<ffffff800842a228>] netdev_rx_csum_fault+0x38/0x48
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.364165] [<ffffff800841dc14>] __skb_checksum_complete+0x6c/0xc0
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.372158] [<ffffff8008502f80>] tcp_v6_rcv+0x1d8/0x10e0
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.379207] [<ffffff80084daa40>] ip6_input_finish+0x268/0x368
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.386642] [<ffffff80084dab70>] ip6_input+0x30/0x80
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.393196] [<ffffff80084da7b4>] ip6_rcv_finish+0x84/0xa8
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.400128] [<ffffff80084dae88>] ipv6_rcv+0x2c8/0x388
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.406630] [<ffffff8008428b38>] __netif_receive_skb_core+0x968/0xad8
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.414554] [<ffffff800842ac3c>] __netif_receive_skb+0x6c/0x78
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.421830] [<ffffff800842acf0>] process_backlog+0xa8/0x158
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.428846] [<ffffff800842e7b8>] net_rx_action+0x110/0x298
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.435771] [<ffffff8008081038>] __do_softirq+0x100/0x1fc
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.442605] [<ffffff80080984f4>] irq_exit+0x6c/0xc8
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.448894] [<ffffff800808d05c>] handle_IPI+0xa4/0x118
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.455407] [<ffffff8008080c90>] bcm2836_arm_irqchip_handle_irq+0x80/0xb0
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.463612] Exception stack(0xffffff8008803e30 to 0xffffff8008803f70)
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.471467] 3e20:                                   0000000000000020 0000000000000000
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.482000] 3e40: 0000000000000001 0000000000000000 ffffff8008803f50 0000004031cb8000
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.492741] 3e60: 0000000ac87920a0 ffffffc03a3c7c60 ffffffc038ca8760 ffffff8008803ef0
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.503726] 3e80: 0000000000000700 0000000000000000 0000000000000000 0000000000000001
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.514921] 3ea0: 0000000000000038 0000000000000000 ffffff8008152c80 0000007fb9bb7ca4
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.526394] 3ec0: 0000000000000000 ffffff8008728828 ffffff8008707008 ffffff8008728000
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.538125] 3ee0: 0000000000000000 0000000000000000 ffffffc038ca8000 ffffff8008018000
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.550111] 3f00: ffffff800801c000 73f20007df00222d ffffffc038ca8000 ffffff8008803f70
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.562337] 3f20: ffffff8008084714 ffffff8008803f70 ffffff8008084718 0000000000000145
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.574822] 3f40: ffffff8008018000 ffffff800801c000 ffffffffffffffff ffffff80080df97c
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.587478] 3f60: ffffff8008803f70 ffffff8008084718
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.594785] [<ffffff8008082870>] el1_irq+0xb0/0x140
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.602090] [<ffffff8008084718>] arch_cpu_idle+0x10/0x18
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.609776] [<ffffff80080c6364>] do_idle+0xec/0x1b8
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.616954] [<ffffff80080c6570>] cpu_startup_entry+0x20/0x28
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.624872] [<ffffff800808cbcc>] secondary_start_kernel+0x17c/0x188
Mon Aug 13 16:36:11 2018 kern.err kernel: [ 2413.633858] eth0: hw csum failure
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.639364] CPU: 3 PID: 24 Comm: ksoftirqd/3 Not tainted 4.14.62 #0
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.647288] Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.654542] Call trace:
Mon Aug 13 16:36:11 2018 kern.warn kernel: [ 2413.657872] [<ffffff8008086bf0>] dump_backtrace+0x0/0x3b0
[... cut ...]
Mon Aug 13 16:37:23 2018 daemon.info dnsmasq-dhcp[5219]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Mon Aug 13 16:37:23 2018 daemon.info dnsmasq-dhcp[5219]: DHCPOFFER(eth0) 192.168.1.171 d0:d2:b0:83:83:16
Mon Aug 13 16:37:24 2018 daemon.info dnsmasq-dhcp[5219]: DHCPREQUEST(eth0) 192.168.1.171 d0:d2:b0:83:83:16
Mon Aug 13 16:37:24 2018 daemon.info dnsmasq-dhcp[5219]: DHCPACK(eth0) 192.168.1.171 d0:d2:b0:83:83:16 Apple-TV
Mon Aug 13 16:37:51 2018 user.notice post-tracking-post-tracking: Multipath wan1 switched to on

Ysurac, 2018-08-13:
That is not good at all. But should have be fixed in 0.31.1 ...
I will check if I see something about that in raspberry kernel.

Maudissure, 2018-08-13:
ok thanks.

its27, 2018-08-13:
I also had this issue with 0.31.1 on a Raspberry PI 3 B+. In fact the PI actually hung during use.

Ysurac, 2018-08-13:
With exactly same errors in log?

What is the result if you disable IPv6 (in **System**->**OpenMPTCProuter**->**Advanced settings**) ?

Ysurac, 2018-08-13:
Issue will be "fixed" (in fact a workaround will be used again for now) in OpenMPTCProuter v0.31.2 that will be available in a few hours.
This seems to be again a bug in lan78xx driver, solved in IPv4 but not for IPv6.

its27, 2018-08-13:
Thanks, was this introduced in 0.31 or has it been present all along?

Ysurac, 2018-08-13:
In 0.31.1 only and only for Raspberry PI.
A patch was added that fixed the issue I used a workaround for, but this was only for IPv4. So workaround will be back in 0.31.2 when IPv6 is enabled.

Ysurac, 2018-08-14:
OpenMPTCProuter v0.31.2 with the workaround is available now.

Maudissure, 2018-08-14:
It's better but still getting WAN1 Down/Up : 

**Tue Aug 14 06:34:55 2018 user.notice post-tracking-post-tracking: wan1 switched off**
Tue Aug 14 06:35:03 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:35:03 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.183 d0:d2:b0:83:83:16
Tue Aug 14 06:35:03 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:35:03 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.183 d0:d2:b0:83:83:16
Tue Aug 14 06:35:04 2018 daemon.info dnsmasq-dhcp[5295]: DHCPREQUEST(eth0) 192.168.1.183 d0:d2:b0:83:83:16
Tue Aug 14 06:35:04 2018 daemon.info dnsmasq-dhcp[5295]: DHCPACK(eth0) 192.168.1.183 d0:d2:b0:83:83:16 Apple-TV
Tue Aug 14 06:35:08 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDECLINE(eth0) 192.168.1.183 d0:d2:b0:83:83:16
Tue Aug 14 06:35:22 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:35:22 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.184 d0:d2:b0:83:83:16
Tue Aug 14 06:35:22 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:35:22 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.184 d0:d2:b0:83:83:16
**Tue Aug 14 06:35:43 2018 user.notice post-tracking-post-tracking: Multipath wan1 switched to on**
Tue Aug 14 06:36:12 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:36:12 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.184 d0:d2:b0:83:83:16
Tue Aug 14 06:36:12 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:36:12 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.184 d0:d2:b0:83:83:16
Tue Aug 14 06:36:13 2018 daemon.info dnsmasq-dhcp[5295]: DHCPREQUEST(eth0) 192.168.1.184 d0:d2:b0:83:83:16
Tue Aug 14 06:36:13 2018 daemon.info dnsmasq-dhcp[5295]: DHCPACK(eth0) 192.168.1.184 d0:d2:b0:83:83:16 Apple-TV
Tue Aug 14 06:36:18 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDECLINE(eth0) 192.168.1.184 d0:d2:b0:83:83:16
Tue Aug 14 06:36:31 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:36:31 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.185 d0:d2:b0:83:83:16
Tue Aug 14 06:36:31 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:36:31 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.185 d0:d2:b0:83:83:16
Tue Aug 14 06:36:41 2018 daemon.info dnsmasq-dhcp[5295]: DHCPINFORM(eth0) 192.168.1.235 68:f7:28:92:ea:2c
Tue Aug 14 06:36:41 2018 daemon.info dnsmasq-dhcp[5295]: DHCPACK(eth0) 192.168.1.235 68:f7:28:92:ea:2c PO469089
Tue Aug 14 06:37:22 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:37:22 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.185 d0:d2:b0:83:83:16
Tue Aug 14 06:37:22 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:37:22 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.185 d0:d2:b0:83:83:16
Tue Aug 14 06:37:23 2018 daemon.info dnsmasq-dhcp[5295]: DHCPREQUEST(eth0) 192.168.1.185 d0:d2:b0:83:83:16
Tue Aug 14 06:37:23 2018 daemon.info dnsmasq-dhcp[5295]: DHCPACK(eth0) 192.168.1.185 d0:d2:b0:83:83:16 Apple-TV
**Tue Aug 14 06:37:26 2018 user.notice post-tracking-post-tracking: wan1 switched off**
Tue Aug 14 06:37:29 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDECLINE(eth0) 192.168.1.185 d0:d2:b0:83:83:16
Tue Aug 14 06:37:42 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:37:42 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.186 d0:d2:b0:83:83:16
Tue Aug 14 06:37:42 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:37:42 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.186 d0:d2:b0:83:83:16
Tue Aug 14 06:38:09 2018 daemon.info dnsmasq-dhcp[5295]: DHCPREQUEST(eth0) 192.168.1.100 44:8a:5b:cb:ea:8e
Tue Aug 14 06:38:09 2018 daemon.info dnsmasq-dhcp[5295]: DHCPACK(eth0) 192.168.1.100 44:8a:5b:cb:ea:8e PC_Bureau
**Tue Aug 14 06:38:14 2018 user.notice post-tracking-post-tracking: Multipath wan1 switched to on**
Tue Aug 14 06:38:33 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:38:33 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.186 d0:d2:b0:83:83:16
Tue Aug 14 06:38:33 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:38:33 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.186 d0:d2:b0:83:83:16
Tue Aug 14 06:38:33 2018 daemon.info dnsmasq-dhcp[5295]: DHCPINFORM(eth0) 192.168.1.235 68:f7:28:92:ea:2c
Tue Aug 14 06:38:33 2018 daemon.info dnsmasq-dhcp[5295]: DHCPACK(eth0) 192.168.1.235 68:f7:28:92:ea:2c PO469089
Tue Aug 14 06:38:34 2018 daemon.info dnsmasq-dhcp[5295]: DHCPREQUEST(eth0) 192.168.1.186 d0:d2:b0:83:83:16
Tue Aug 14 06:38:34 2018 daemon.info dnsmasq-dhcp[5295]: DHCPACK(eth0) 192.168.1.186 d0:d2:b0:83:83:16 Apple-TV
Tue Aug 14 06:38:39 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDECLINE(eth0) 192.168.1.186 d0:d2:b0:83:83:16
Tue Aug 14 06:38:53 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:38:53 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.187 d0:d2:b0:83:83:16
Tue Aug 14 06:38:53 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
[... cut ...]
Tue Aug 14 06:44:18 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 14 06:44:18 2018 daemon.info dnsmasq-dhcp[5295]: DHCPOFFER(eth0) 192.168.1.191 d0:d2:b0:83:83:16
Tue Aug 14 06:44:19 2018 daemon.info dnsmasq-dhcp[5295]: DHCPREQUEST(eth0) 192.168.1.191 d0:d2:b0:83:83:16
Tue Aug 14 06:44:19 2018 daemon.info dnsmasq-dhcp[5295]: DHCPACK(eth0) 192.168.1.191 d0:d2:b0:83:83:16 Apple-TV
Tue Aug 14 06:44:25 2018 daemon.info dnsmasq-dhcp[5295]: DHCPDECLINE(eth0) 192.168.1.191 d0:d2:b0:83:83:16

Ysurac, 2018-08-14:
It's not the same problem.
omr-tracker detect that wan1 is not responding. So the ip defined as gateway doesn't answer to ping request or can't ping ip in tracker list.
I will add a message that tell exactly what is not responding (gateway or a host).

Ysurac, 2018-08-14:
OpenMPTCProuter v0.31.3 will give you why an interface is "switched off".

Ysurac, 2018-08-14:
OpenMPTCProuter can't ping wan1 gateway sometimes.
You can try from OpenMPTCProuter to ping your gateway and see if you get packet loss or slow answer.

Maudissure, 2018-08-14:
yes, can't ping the freebox : 

PING 192.168.10.1 (192.168.10.1): 56 data bytes

--- 192.168.10.1 ping statistics ---
5 packets transmitted, 0 packets received, 100% packet loss

Ysurac, 2018-08-14:
The Freebox must be configured to answer to ping request for now.
I will make some change, because some stupid modem seems to block ping by default...

Maudissure, 2018-08-14:
It is configured to answer ping request ...

Trying to put freebox active and wan2 (LTE1) master and reboot the freebox.

Ysurac, 2018-08-14:
You can try OpenMPTCProuter v0.31.4, this should work even if gateway doesn't answer to ping.

Maudissure, 2018-08-14:
Already on 0.31.4 version.

closed 2018-08-16 by Ysurac