Ysurac/openmptcprouter #131: Internet connection stability

fnoir, 2018-08-14:
## Actual Behavior

Stability problems since 2 days

## Specifications

  - OpenMPTCProuter version: v0.31.1 r7253-5b1c6570ca
  - OpenMPTCProuter VPS version: v0.44 updated with kernel 4.14.61
  - OpenMPTCProuter platform: VMware

## Logs

```
Tue Aug 14 12:14:21 2018 kern.err kernel: [45465.205134] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:14:21 2018 kern.err kernel: [45465.205295] mptcp_detect_mapping dseq 581697040 mdseq 581697040, sseq 1760596844 msseq 1760593988 dlen 3456 mdlen 3455 dfin 1 mdfin 1
Tue Aug 14 12:14:41 2018 kern.err kernel: [45485.382701] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:14:41 2018 kern.err kernel: [45485.382766] mptcp_detect_mapping dseq 1835239616 mdseq 1835239616, sseq 2431739545 msseq 2431736689 dlen 3729 mdlen 3728 dfin 1 mdfin 1
Tue Aug 14 12:15:02 2018 kern.err kernel: [45506.601712] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:15:02 2018 kern.err kernel: [45506.601835] mptcp_detect_mapping dseq 3667737336 mdseq 3667737336, sseq 887683500 msseq 887680644 dlen 11004 mdlen 11003 dfin 1 mdfin 1
Tue Aug 14 12:15:19 2018 kern.err kernel: [45522.950416] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:15:19 2018 kern.err kernel: [45522.950481] mptcp_detect_mapping dseq 222928382 mdseq 222928382, sseq 1701788054 msseq 1701785198 dlen 5566 mdlen 5565 dfin 1 mdfin 1
Tue Aug 14 12:15:22 2018 kern.err kernel: [45526.231272] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:15:22 2018 kern.err kernel: [45526.231338] mptcp_detect_mapping dseq 909499136 mdseq 909499136, sseq 3521021053 msseq 3521018197 dlen 3547 mdlen 3546 dfin 1 mdfin 1
Tue Aug 14 12:15:23 2018 kern.err kernel: [45527.032382] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:15:23 2018 kern.err kernel: [45527.032463] mptcp_detect_mapping dseq 3436164763 mdseq 3436164763, sseq 2357972871 msseq 2357970015 dlen 4716 mdlen 4715 dfin 1 mdfin 1
Tue Aug 14 12:15:47 2018 kern.err kernel: [45551.064414] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:15:47 2018 kern.err kernel: [45551.064488] mptcp_detect_mapping dseq 262533492 mdseq 262533492, sseq 2702141311 msseq 2702138455 dlen 4252 mdlen 4251 dfin 1 mdfin 1
Tue Aug 14 12:15:50 2018 kern.err kernel: [45554.070263] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:15:50 2018 kern.err kernel: [45554.070330] mptcp_detect_mapping dseq 3670688414 mdseq 3670688414, sseq 1248294547 msseq 1248291691 dlen 4877 mdlen 4876 dfin 1 mdfin 1
Tue Aug 14 12:15:51 2018 kern.err kernel: [45555.351574] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:15:51 2018 kern.err kernel: [45555.351636] mptcp_detect_mapping dseq 1486629519 mdseq 1486629519, sseq 726687141 msseq 726684285 dlen 4938 mdlen 4937 dfin 1 mdfin 1
Tue Aug 14 12:16:11 2018 kern.err kernel: [45575.547099] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:16:11 2018 kern.err kernel: [45575.547190] mptcp_detect_mapping dseq 4074102701 mdseq 4074102701, sseq 1353801215 msseq 1353798359 dlen 4407 mdlen 4406 dfin 1 mdfin 1
Tue Aug 14 12:16:14 2018 kern.err kernel: [45578.172861] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:16:14 2018 kern.err kernel: [45578.172989] mptcp_detect_mapping dseq 3183212403 mdseq 3183212403, sseq 4207847277 msseq 4207844421 dlen 4393 mdlen 4392 dfin 1 mdfin 1
Tue Aug 14 12:16:33 2018 kern.err kernel: [45597.324911] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:16:33 2018 kern.err kernel: [45597.324974] mptcp_detect_mapping dseq 4244858597 mdseq 4244858597, sseq 2329799044 msseq 2329796188 dlen 2954 mdlen 2953 dfin 1 mdfin 1
Tue Aug 14 12:16:33 2018 kern.err kernel: [45597.408115] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:16:33 2018 kern.err kernel: [45597.408233] mptcp_detect_mapping dseq 615537540 mdseq 615537540, sseq 1484301808 msseq 1484298952 dlen 3896 mdlen 3895 dfin 1 mdfin 1
Tue Aug 14 12:16:35 2018 kern.err kernel: [45599.224359] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:16:35 2018 kern.err kernel: [45599.224421] mptcp_detect_mapping dseq 2078629383 mdseq 2078629383, sseq 1236479626 msseq 1236476770 dlen 3729 mdlen 3728 dfin 1 mdfin 1
Tue Aug 14 12:16:53 2018 kern.err kernel: [45617.424073] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:16:53 2018 kern.err kernel: [45617.424149] mptcp_detect_mapping dseq 58241653 mdseq 58241653, sseq 906528998 msseq 906526142 dlen 3008 mdlen 3007 dfin 1 mdfin 1
Tue Aug 14 12:16:56 2018 kern.err kernel: [45620.287066] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:16:56 2018 kern.err kernel: [45620.287129] mptcp_detect_mapping dseq 2456706786 mdseq 2456706786, sseq 2256422262 msseq 2256419406 dlen 4578 mdlen 4577 dfin 1 mdfin 1
Tue Aug 14 12:16:56 2018 kern.err kernel: [45620.698595] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:16:56 2018 kern.err kernel: [45620.698723] mptcp_detect_mapping dseq 1920900339 mdseq 1920900339, sseq 675683262 msseq 675680406 dlen 4483 mdlen 4482 dfin 1 mdfin 1
Tue Aug 14 12:17:13 2018 kern.err kernel: [45637.607368] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:17:13 2018 kern.err kernel: [45637.607434] mptcp_detect_mapping dseq 881698345 mdseq 881698345, sseq 1223739203 msseq 1223736347 dlen 4160 mdlen 4159 dfin 1 mdfin 1
Tue Aug 14 12:17:18 2018 kern.err kernel: [45641.964989] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:17:18 2018 kern.err kernel: [45641.965120] mptcp_detect_mapping dseq 3677638066 mdseq 3677638066, sseq 284088602 msseq 284085746 dlen 5568 mdlen 5567 dfin 1 mdfin 1
Tue Aug 14 12:17:18 2018 kern.err kernel: [45642.470185] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:17:18 2018 kern.err kernel: [45642.470249] mptcp_detect_mapping dseq 2999802736 mdseq 2999802736, sseq 3627967828 msseq 3627964972 dlen 3766 mdlen 3765 dfin 1 mdfin 1
Tue Aug 14 12:17:34 2018 kern.err kernel: [45658.598671] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:17:34 2018 kern.err kernel: [45658.598796] mptcp_detect_mapping dseq 1202011887 mdseq 1202011887, sseq 71212171 msseq 71209315 dlen 3018 mdlen 3017 dfin 1 mdfin 1
Tue Aug 14 12:17:38 2018 kern.err kernel: [45662.860468] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:17:38 2018 kern.err kernel: [45662.860596] mptcp_detect_mapping dseq 4256546504 mdseq 4256546504, sseq 928616034 msseq 928613178 dlen 4251 mdlen 4250 dfin 1 mdfin 1
Tue Aug 14 12:17:40 2018 kern.err kernel: [45664.924073] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:17:40 2018 kern.err kernel: [45664.924137] mptcp_detect_mapping dseq 1204818887 mdseq 1204818887, sseq 2134446237 msseq 2134443381 dlen 5183 mdlen 5182 dfin 1 mdfin 1
Tue Aug 14 12:18:15 2018 kern.err kernel: [45699.715275] mptcp_detect_mapping Mappings do not match!
Tue Aug 14 12:18:15 2018 kern.err kernel: [45699.715381] mptcp_detect_mapping dseq 1441488084 mdseq 1441488084, sseq 3060827937 msseq 3060825081 dlen 4541 mdlen 4540 dfin 1 mdfin 1
```

```
Tue Aug 14 13:35:59 2018 user.notice post-tracking-post-tracking: wan2 switched off
Tue Aug 14 13:35:59 2018 user.notice post-tracking-post-tracking: Remove server xxx.xxx.xxx.xxx route via wan2
Tue Aug 14 13:36:12 2018 user.notice post-tracking-post-tracking: Set server xxx.xxx.xxx.xxx route via wan2 metric 4
Tue Aug 14 13:36:12 2018 user.notice post-tracking-post-tracking: Multipath wan2 switched to on
Tue Aug 14 15:54:18 2018 user.notice post-tracking-post-tracking: wan2 switched off
Tue Aug 14 15:54:18 2018 user.notice post-tracking-post-tracking: Remove server xxx.xxx.xxx.xxx route via wan2
Tue Aug 14 15:58:04 2018 user.notice post-tracking-post-tracking: Set server xxx.xxx.xxx.xxx route via wan2 metric 4
Tue Aug 14 15:58:04 2018 user.notice post-tracking-post-tracking: Multipath wan2 switched to on
```

```
Tue Aug 14 16:17:30 2018 user.notice post-tracking-post-tracking: wan2 switched off
Tue Aug 14 16:17:30 2018 user.notice post-tracking-post-tracking: Remove server xxx.xxx.xxx.xxx route via wan2
Tue Aug 14 16:20:29 2018 user.notice post-tracking-post-tracking: Set server xxx.xxx.xxx.xxx route via wan2 metric 4
Tue Aug 14 16:20:29 2018 user.notice post-tracking-post-tracking: Multipath wan2 switched to on
Tue Aug 14 16:22:49 2018 user.notice post-tracking-post-tracking: wan2 switched off
Tue Aug 14 16:22:49 2018 user.notice post-tracking-post-tracking: Remove server xxx.xxx.xxx.xxx route via wan2
Tue Aug 14 16:24:07 2018 user.notice post-tracking-post-tracking: Set server xxx.xxx.xxx.xxx route via wan2 metric 4
Tue Aug 14 16:24:07 2018 user.notice post-tracking-post-tracking: Multipath wan2 switched to on
```

Ysurac, 2018-08-14:
OpenMPTCProuter v0.31.4 that will be out in about 2 hours ([CircleCI](https://circleci.com/gh/Ysurac/openmptcprouter-feeds) is making images), will display more info when an interface is switched off.

Maudissure, 2018-08-14:
Wich is your modem on Wan2 ?

fnoir, 2018-08-14:
Huawei B593 with latest official update

Ysurac, 2018-08-14:
OpenMPTCProuter v0.31.4 is available, you should at least know why the interface is detected as down.

Maudissure, 2018-08-14:
Ok there : 

Tue Aug 14 17:36:57 2018 daemon.err uhttpd[2815]: luci: accepted login on / for root from 192.168.1.100
Tue Aug 14 17:37:41 2018 user.notice post-tracking-post-tracking: wan1 switched off because ping from 192.168.10.2 error (8.8.4.4,9.9.9.9,1.1.1.1,1.0.0.1,80.67.169.12,8.8.4.4,9.9.9.9,1.1.1.1,1.0.0.1,80.67.169.12,4.2.2.1,8.8.8.8,9.9.9.9,1.1.1.1,1.0.0.1,80.67.169.12,80.67.169.40)
Tue Aug 14 17:37:41 2018 user.notice post-tracking-post-tracking: Remove server 51.254.127.83 route via wan1
Tue Aug 14 17:38:10 2018 daemon.info dnsmasq-dhcp[5066]: DHCPREQUEST(eth0) 192.168.1.247 30:07:4d:98:17:59
Tue Aug 14 17:38:10 2018 daemon.info dnsmasq-dhcp[5066]: DHCPACK(eth0) 192.168.1.247 30:07:4d:98:17:59 Galaxy-S8
Tue Aug 14 17:38:23 2018 user.notice post-tracking-post-tracking: Set server 51.254.127.83 route via wan1 metric 3
Tue Aug 14 17:38:23 2018 user.notice post-tracking-post-tracking: Multipath wan1 switched to on
Tue Aug 14 17:40:47 2018 user.notice post-tracking-post-tracking: wan1 switched off because ping from 192.168.10.2 error (8.8.4.4,9.9.9.9,1.1.1.1,1.0.0.1,80.67.169.12,8.8.4.4,9.9.9.9,1.1.1.1,1.0.0.1,80.67.169.12,4.2.2.1,8.8.8.8,9.9.9.9,1.1.1.1,1.0.0.1,80.67.169.12,80.67.169.40,8.8.8.8,8.8.4.4,9.9.9.9,1.1.1.1,1.0.0.1,80.67.169.40,8.8.8.8,80.67.169.40,4.2.2.1,8.8.8.8,8.8.4.4,9.9.9.9)
Tue Aug 14 17:40:47 2018 user.notice post-tracking-post-tracking: Remove server 51.254.127.83 route via wan1

and my parameters : 
[image]

Ysurac, 2018-08-14:
No, the error "wan1 switched off because ping from 192.168.10.2 error (8.8.4.4,9.9.9.9,1.1.1.1,1.0.0.1,80.67.169.12,8.8.4.4,9.9.9.9,1.1.1.1,1.0.0.1,80.67.169.12,4.2.2.1,8.8.8.8,9.9.9.9,1.1.1.1,1.0.0.1,80.67.169.12,80.67.169.40,8.8.8.8,8.8.4.4,9.9.9.9,1.1.1.1,1.0.0.1,80.67.169.40,8.8.8.8,80.67.169.40,4.2.2.1,8.8.8.8,8.8.4.4,9.9.9.9)" mean that wan1 was switched of because it was not able to ping any of the IPs in the list from this interface. So there is a network problem somewhere. Are you sure no IP are used multiples times on the network ?

Maudissure, 2018-08-14:
Meanings ? The DNS IP's ?

Ysurac, 2018-08-14:
It's DNS servers, but they are, by default, only used for ping test. If they answer to ping than interface is up, else down.

The errors are about wifi brcmfmac module. I will check if there is a fix/workaround for that...

Ysurac, 2018-08-14:
I make a mistake, this will be fixed in OpenMPTCProuter v0.31.5, [CircleCI](https://circleci.com/gh/Ysurac/openmptcprouter-feeds) compile image, this will be available in a few hours.

fnoir, 2018-08-15:
Now, i have:

[image]

```
Wed Aug 15 09:21:42 2018 user.notice post-tracking-post-tracking: omrvpn down
Wed Aug 15 09:21:42 2018 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.5.1 dev wan1
Wed Aug 15 09:21:42 2018 daemon.info unbound: [6837:0] info: service stopped (unbound 1.7.3).
Wed Aug 15 09:21:42 2018 daemon.info unbound: [6837:0] info: server stats for thread 0: 8 queries, 0 answers from cache, 8 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Wed Aug 15 09:21:42 2018 daemon.info unbound: [6837:0] info: server stats for thread 0: requestlist max 7 avg 2.75 exceeded 0 jostled 0
Wed Aug 15 09:21:42 2018 daemon.info unbound: [6837:0] info: average recursion processing time 0.197948 sec
Wed Aug 15 09:21:42 2018 daemon.info unbound: [6837:0] info: histogram of recursion processing times
Wed Aug 15 09:21:42 2018 daemon.info unbound: [6837:0] info: [25%]=0.098304 median[50%]=0.196608 [75%]=0.349525
Wed Aug 15 09:21:42 2018 daemon.info unbound: [6837:0] info: lower(secs) upper(secs) recursions
Wed Aug 15 09:21:42 2018 daemon.info unbound: [6837:0] info:    0.016384    0.032768 1
Wed Aug 15 09:21:42 2018 daemon.info unbound: [6837:0] info:    0.065536    0.131072 2
Wed Aug 15 09:21:42 2018 daemon.info unbound: [6837:0] info:    0.131072    0.262144 2
Wed Aug 15 09:21:42 2018 daemon.info unbound: [6837:0] info:    0.262144    0.524288 3
Wed Aug 15 09:21:42 2018 daemon.notice unbound: [7127:0] notice: init module 0: validator
Wed Aug 15 09:21:42 2018 daemon.notice unbound: [7127:0] notice: init module 1: iterator
Wed Aug 15 09:21:42 2018 daemon.info unbound: [7127:0] info: start of service (unbound 1.7.3).
Wed Aug 15 09:21:50 2018 daemon.info unbound: [7127:0] info: generate keytag query _ta-4a5c-4f66. NULL IN
Wed Aug 15 09:21:53 2018 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Wed Aug 15 09:21:56 2018 user.notice post-tracking-post-tracking: Set server xxx.xxx.xxx.xxx route via wan1 metric 1
Wed Aug 15 09:24:00 2018 daemon.err /usr/bin/ss-redir[6287]: remote recv: Operation timed out
Wed Aug 15 09:24:00 2018 daemon.err /usr/bin/ss-redir[6287]: remote recv: Operation timed out
```

I can't play video games for exemple because i have a latency that varies from 40ms to 2500ms.

Ysurac, 2018-08-15:
Status page is ok.
ss-redir operation timed out are normal.
You seems to have a problem with the tunnel. It's used for ICMP and UDP. I will check tomorrow how I can test it and fix when there is problems.

fnoir, 2018-08-15:
I think i achieved the limit of consumption with LTE card (1.6TB)... :D It isn't unlimited...

Ysurac, 2018-08-15:
You're not using Free mobile 4G with a Freebox ? It's what I have and it's really unlimited.

Ysurac, 2018-08-15:
Maybe many users, I had the same problem in July.

Maudissure, 2018-08-15:
After some hours, network seems stable. Internet connection too. No routrer's crash also.

You're on the good way, keep going ! And thanks for all works you do.

Maudissure, 2018-08-15:
talked to fast, i got instable internet connection.

fnoir, 2018-08-18:
With openmptcprouter v0.31.7, same problems : 

```
Sat Aug 18 16:07:02 2018 daemon.err glorytun[5367]: read: Operation timed out
Sat Aug 18 16:07:02 2018 daemon.info glorytun[5367]: STOPPED tun0
Sat Aug 18 16:07:10 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: connected
Sat Aug 18 16:07:10 2018 daemon.info glorytun[5367]: STARTED tun0
Sat Aug 18 16:08:07 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:07 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:07 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:07 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:21 2018 daemon.err glorytun[5367]: read: Operation timed out
Sat Aug 18 16:08:21 2018 daemon.info glorytun[5367]: STOPPED tun0
Sat Aug 18 16:08:38 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:38 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:38 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:38 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:38 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: connected
Sat Aug 18 16:08:43 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:43 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:48 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:48 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:48 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:08:48 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:09:18 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: key exchange failed
Sat Aug 18 16:09:18 2018 daemon.info glorytun[5367]: STOPPED tun0
Sat Aug 18 16:09:19 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: connected
Sat Aug 18 16:09:59 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: key exchange failed
Sat Aug 18 16:09:59 2018 daemon.info glorytun[5367]: STOPPED tun0
Sat Aug 18 16:10:00 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: connected
Sat Aug 18 16:10:27 2018 daemon.notice netifd: omrvpn (5439): udhcpc: sending renew to 10.255.255.1
Sat Aug 18 16:10:40 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: key exchange failed
Sat Aug 18 16:10:40 2018 daemon.info glorytun[5367]: STOPPED tun0
Sat Aug 18 16:10:41 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: connected
Sat Aug 18 16:11:19 2018 daemon.info glorytun[5367]: STARTED tun0
Sat Aug 18 16:27:12 2018 daemon.err glorytun[5367]: read: Operation timed out
Sat Aug 18 16:27:12 2018 daemon.info glorytun[5367]: STOPPED tun0
Sat Aug 18 16:27:13 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: connected
Sat Aug 18 16:27:44 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Connection reset by peer
Sat Aug 18 16:27:53 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: key exchange failed
Sat Aug 18 16:27:53 2018 daemon.info glorytun[5367]: STOPPED tun0
Sat Aug 18 16:27:54 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: connected
Sat Aug 18 16:28:34 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: key exchange failed
Sat Aug 18 16:28:34 2018 daemon.info glorytun[5367]: STOPPED tun0
Sat Aug 18 16:28:35 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: connected
Sat Aug 18 16:28:46 2018 daemon.err /usr/bin/ss-local[6357]: getpeername: Socket not connected
Sat Aug 18 16:28:51 2018 daemon.err /usr/bin/ss-redir[6359]: remote recv: Operation timed out
Sat Aug 18 16:29:15 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: key exchange failed
Sat Aug 18 16:29:15 2018 daemon.info glorytun[5367]: STOPPED tun0
Sat Aug 18 16:29:16 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: connected
Sat Aug 18 16:29:56 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: key exchange failed
Sat Aug 18 16:29:56 2018 daemon.info glorytun[5367]: STOPPED tun0
Sat Aug 18 16:29:57 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: connected
Sat Aug 18 16:30:37 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: key exchange failed
Sat Aug 18 16:30:37 2018 daemon.info glorytun[5367]: STOPPED tun0
Sat Aug 18 16:30:38 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: connected
Sat Aug 18 16:31:18 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: key exchange failed
Sat Aug 18 16:31:18 2018 daemon.info glorytun[5367]: STOPPED tun0
Sat Aug 18 16:31:35 2018 daemon.err glorytun[5367]: xxx.xxx.xxx.xxx.65001: connected
Sat Aug 18 16:32:02 2018 daemon.info glorytun[5367]: STARTED tun0
```

Ysurac, 2018-08-18:
The problem seems to be a very bad connection between master wan and server.
What is the result of ```ip route``` on OpenMPTCProuter ?
You can also try to restart glorytun TCP on the server, I had some problem with it...

fnoir, 2018-08-18:
I restart server & router and same problems.

Sorry, i need stable connection this week, i turn off OpenMPTCP for moment.

Maudissure, 2018-08-18:
i'm getting this message regulary, is this important ? : 

Sat Aug 18 17:05:23 2018 daemon.warn odhcpd[1529]: A default route is present but there is no public prefix on eth0 thus we don't announce a default route!

Maudissure, 2018-08-19:
Ok, Now on OMR V0.32.1, i have some kernel issues after a night on : 

Sun Aug 19 07:14:55 2018 daemon.err uhttpd[4402]: luci: accepted login on /admin/network/mptcp/bandwidth for root from 192.168.1.100
Sun Aug 19 07:14:56 2018 kern.err kernel: [34502.335268] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:14:56 2018 kern.err kernel: [34502.346197] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:14:59 2018 kern.err kernel: [34505.333839] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:14:59 2018 kern.err kernel: [34505.344691] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:02 2018 kern.err kernel: [34508.334035] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:02 2018 kern.err kernel: [34508.344863] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:05 2018 kern.err kernel: [34511.319257] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:05 2018 kern.err kernel: [34511.330214] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:08 2018 kern.err kernel: [34514.332499] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:08 2018 kern.err kernel: [34514.343433] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:11 2018 kern.err kernel: [34517.335868] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:11 2018 kern.err kernel: [34517.346700] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:14 2018 kern.err kernel: [34520.335297] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:14 2018 kern.err kernel: [34520.346282] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:17 2018 kern.err kernel: [34523.326301] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:17 2018 kern.err kernel: [34523.337056] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:20 2018 kern.err kernel: [34526.321494] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:20 2018 kern.err kernel: [34526.332409] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:23 2018 kern.err kernel: [34529.325266] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:23 2018 kern.err kernel: [34529.336287] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:26 2018 kern.err kernel: [34532.327346] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:26 2018 kern.err kernel: [34532.338599] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:29 2018 kern.err kernel: [34535.337808] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:29 2018 kern.err kernel: [34535.348603] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:32 2018 kern.err kernel: [34538.324533] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:32 2018 kern.err kernel: [34538.335594] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:35 2018 kern.err kernel: [34541.327853] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:35 2018 kern.err kernel: [34541.338608] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:15:38 2018 kern.err kernel: [34544.333092] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:15:38 2018 kern.err kernel: [34544.344014] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:16:02 2018 kern.err kernel: [34568.321931] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:16:02 2018 kern.err kernel: [34568.332777] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:16:17 2018 kern.err kernel: [34583.324501] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:16:17 2018 kern.err kernel: [34583.335414] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:16:29 2018 kern.err kernel: [34595.318480] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:16:29 2018 kern.err kernel: [34595.329422] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:16:41 2018 kern.err kernel: [34607.318791] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:16:41 2018 kern.err kernel: [34607.329717] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[... cut ...]
Sun Aug 19 07:24:15 2018 kern.err kernel: [35060.743550] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:24:15 2018 kern.err kernel: [35060.754863] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:24:21 2018 kern.err kernel: [35066.742021] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun Aug 19 07:24:21 2018 kern.err kernel: [35066.752976] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun Aug 19 07:24:56 2018 daemon.err uhttpd[4402]: Host whois.ripe.net not found.

Ysurac, 2018-08-19:
It's not kernel issues. It's an "error" in the wifi driver.
I will add a patch that remove this warning.

Maudissure, 2018-08-19:
Sorry just see that i had the same issue before with answer from you.

nileio, 2018-08-21:
please try restarting dnsmasq
/etc/init.d/dnsmasq restart

now the warning message "A default route is present but there is no public prefix on eth0 thus we don't announce a default route!" is a possible bug https://github.com/openwrt/odhcpd/issues/79
it has to do with ipv6 but yet to find out more

Maudissure, 2018-08-21:
restarted dnsmasq, not better : 

Tue Aug 21 18:49:10 2018 daemon.info dnsmasq-dhcp[19182]: DHCPOFFER(eth0) 192.168.1.208 b4:8b:19:1b:4e:ee
Tue Aug 21 18:49:11 2018 daemon.info dnsmasq-dhcp[19182]: DHCPREQUEST(eth0) 192.168.1.208 b4:8b:19:1b:4e:ee
Tue Aug 21 18:49:11 2018 daemon.info dnsmasq-dhcp[19182]: DHCPACK(eth0) 192.168.1.208 b4:8b:19:1b:4e:ee iPhone
Tue Aug 21 18:49:14 2018 daemon.warn dnsmasq-dhcp[19182]: not using configured address 192.168.1.54 because it was previously declined
Tue Aug 21 18:49:16 2018 daemon.err /usr/bin/ss-redir[7167]: remote recv: Connection reset by peer
Tue Aug 21 18:49:18 2018 daemon.info dnsmasq-dhcp[19182]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 21 18:49:18 2018 daemon.info dnsmasq-dhcp[19182]: DHCPOFFER(eth0) 192.168.1.212 d0:d2:b0:83:83:16
Tue Aug 21 18:49:18 2018 daemon.warn dnsmasq-dhcp[19182]: not using configured address 192.168.1.54 because it was previously declined
Tue Aug 21 18:49:18 2018 daemon.info dnsmasq-dhcp[19182]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Tue Aug 21 18:49:18 2018 daemon.info dnsmasq-dhcp[19182]: DHCPOFFER(eth0) 192.168.1.212 d0:d2:b0:83:83:16
Tue Aug 21 18:49:18 2018 daemon.info dnsmasq-dhcp[19182]: DHCPDECLINE(eth0) 192.168.1.208 b4:8b:19:1b:4e:ee
Tue Aug 21 18:49:19 2018 daemon.info dnsmasq-dhcp[19182]: DHCPREQUEST(eth0) 192.168.1.212 d0:d2:b0:83:83:16
Tue Aug 21 18:49:19 2018 daemon.info dnsmasq-dhcp[19182]: DHCPACK(eth0) 192.168.1.212 d0:d2:b0:83:83:16 Apple_TV
Tue Aug 21 18:49:19 2018 daemon.warn dnsmasq-dhcp[19182]: not giving name Apple_TV.lan to the DHCP lease of 192.168.1.212 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.54
Tue Aug 21 18:49:24 2018 daemon.info dnsmasq-dhcp[19182]: DHCPDECLINE(eth0) 192.168.1.212 d0:d2:b0:83:83:16
Tue Aug 21 18:49:30 2018 daemon.info dnsmasq-dhcp[19182]: DHCPDISCOVER(eth0) b4:8b:19:1b:4e:ee
Tue Aug 21 18:49:30 2018 daemon.info dnsmasq-dhcp[19182]: DHCPOFFER(eth0) 192.168.1.210 b4:8b:19:1b:4e:ee
Tue Aug 21 18:49:30 2018 daemon.info dnsmasq-dhcp[19182]: DHCPDISCOVER(eth0) b4:8b:19:1b:4e:ee
Tue Aug 21 18:49:30 2018 daemon.info dnsmasq-dhcp[19182]: DHCPOFFER(eth0) 192.168.1.210 b4:8b:19:1b:4e:ee
Tue Aug 21 18:49:31 2018 daemon.info dnsmasq-dhcp[19182]: DHCPREQUEST(eth0) 192.168.1.210 b4:8b:19:1b:4e:ee
Tue Aug 21 18:49:31 2018 daemon.info dnsmasq-dhcp[19182]: DHCPACK(eth0) 192.168.1.210 b4:8b:19:1b:4e:ee iPhone

fnoir, 2018-09-25:
New problem with openmptcprouter v0.37 r7320-8a006c2480 : 

Tue Sep 25 17:17:55 2018 daemon.err glorytun[22301]: read: Operation timed out
Tue Sep 25 17:17:55 2018 daemon.info glorytun[22301]: STOPPED tun0
Tue Sep 25 17:18:00 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: connected
Tue Sep 25 17:18:04 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: key exchange failed
Tue Sep 25 17:18:04 2018 daemon.info glorytun[22301]: STOPPED tun0
Tue Sep 25 17:18:12 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: connected
Tue Sep 25 17:18:16 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: key exchange failed
Tue Sep 25 17:18:16 2018 daemon.info glorytun[22301]: STOPPED tun0
Tue Sep 25 17:18:20 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: connected
Tue Sep 25 17:18:24 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: key exchange failed
Tue Sep 25 17:18:24 2018 daemon.info glorytun[22301]: STOPPED tun0
Tue Sep 25 17:18:24 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: connected
Tue Sep 25 17:18:28 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: key exchange failed
Tue Sep 25 17:18:28 2018 daemon.info glorytun[22301]: STOPPED tun0
Tue Sep 25 17:18:28 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: connected
Tue Sep 25 17:18:32 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: key exchange failed
Tue Sep 25 17:18:32 2018 daemon.info glorytun[22301]: STOPPED tun0
Tue Sep 25 17:18:32 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: connected
Tue Sep 25 17:18:36 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: key exchange failed
Tue Sep 25 17:18:36 2018 daemon.info glorytun[22301]: STOPPED tun0
Tue Sep 25 17:18:36 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: connected
Tue Sep 25 17:18:40 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: key exchange failed
Tue Sep 25 17:18:40 2018 daemon.info glorytun[22301]: STOPPED tun0
Tue Sep 25 17:18:40 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: connected
Tue Sep 25 17:18:44 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: key exchange failed
Tue Sep 25 17:18:44 2018 daemon.info glorytun[22301]: STOPPED tun0
Tue Sep 25 17:18:44 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: connected
Tue Sep 25 17:18:48 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: key exchange failed
Tue Sep 25 17:18:48 2018 daemon.info glorytun[22301]: STOPPED tun0
Tue Sep 25 17:18:52 2018 daemon.err glorytun[22301]: xxx.xxx.xxx.xxx.65001: connected
Tue Sep 25 17:18:53 2018 daemon.info glorytun[22301]: STARTED tun0

Ysurac, 2018-09-27:
For glorytun there is a "read: Operation timed out", then it try to reconnect and this seems to take 60s after a failure.
For ss-redir, it receive a "Connection refused" from remote host.

If it's both the same time this seems to be a connection problem between OpenMPTCProuter and the VPS.
I will increase timeout in next release.

fnoir, 2018-09-27:
I reinstall the VPS. I monitor it to see if any anomalies are still occurring

fnoir, 2018-10-06:
After many tests:

- VMware: unstable, disconnections (2 vCPU, 2GB of memory)
- RPi: no problem since 5 days

closed 2019-02-20 by Ysurac