Ysurac/openmptcprouter #132: No VPS IP address, No WAN IP address, Gateway DOWN although modem is connected

afriza, 2018-08-15:
## Expected Behavior
I am using Huawei E1552 HSDPA USB Stick as WAN.
OpenMPTCProuter works on USB sticks modem.
## Actual Behavior

[image]

[image]

[image]

  - ping and ssh from OpenMPTCProuter to Server work.
  - https://whatismyipaddress.com/ip-lookup shows my server IP.
  - how do I check that OpenMPTCProuter works peroperly?

## Steps to Reproduce the Problem

  1. flash OpenMPTCProuter v0.31.2 and set password
  2. add modem connection `tri` and configure APN, username, password until connected.
  3. set `tri` into WAN firewall group.
  4. set `tri` as master multipath and disable multipath in `WAN1` and `WAN2`
  5. go to wizard and set Server IP, ShadowSocks key, and Glorytun key

## Specifications

  - OpenMPTCProuter version: (0.31.2)
  - OpenMPTCProuter VPS version: (0.44)
  - OpenMPTCProuter platform: (RPI3)

## Configurations
```
root@OpenMPTCProuter:~# cat /etc/config/network 

config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'
	option multipath 'off'

config globals 'globals'
	option ula_prefix 'fd95:ad3a:821b::/48'
	option multipath 'enable'
	option mptcp_path_manager 'fullmesh'
	option mptcp_scheduler 'default'
	option congestion 'bbr'
	option mptcp_checksum '0'
	option mptcp_syn_retries '5'
	option mptcp_fullmesh_num_subflows '1'
	option mptcp_fullmesh_create_on_err '1'
	option mptcp_ndiffports_num_subflows '1'

config interface 'lan'
	option ifname 'eth0'
	option proto 'static'
	option ipaddr '192.168.100.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option metric '2048'
	option multipath 'off'
	option ip4table 'lan'

config rule 'lan_rule'
	option lookup 'lan'
	option priority '100'

config interface 'wan1'
	option proto 'static'
	option ip4table 'wan'
	option defaultroute '0'
	option type 'macvlan'
	option ifname 'wan1'
	option multipath 'off'
	option netmask '255.255.255.0'

config interface 'wan2'
	option proto 'static'
	option ip4table 'wan'
	option defaultroute '0'
	option type 'macvlan'
	option ifname 'wan2'
	option multipath 'off'
	option netmask '255.255.255.0'

config device 'wan1_dev'
	option name 'wan1'
	option type 'macvlan'
	option ifname 'eth0'
	option macaddr 'auto1534142183'

config device 'wan2_dev'
	option name 'wan2'
	option type 'macvlan'
	option ifname 'eth0'
	option macaddr 'auto1534142183'

config route6
	option interface 'lan'
	option target '::/0'

config interface 'omrvpn'
	option ifname 'tun0'
	option proto 'dhcp'
	option ip4table 'vpn'
	option multipath 'off'
	option leasetime '12h'
	option mtu '1400'

config interface 'omr6in4'
	option proto '6in4'
	option ip4table 'vpn'
	option multipath 'off'
	option ipaddr '10.255.255.2'
	option peeraddr '10.255.255.1'
	option gateway 'fe80::a00:1'
	option ip6addr 'fe80::a00:2'

config interface 'tri'
	option proto '3g'
	option device '/dev/ttyUSB0'
	option service 'umts_only'
	option apn '3gprs'
	option username '3gprs'
	option password '3gprs'
	option ipv6 'auto'
	option multipath 'master'
```

```
root@OpenMPTCProuter:~# cat /etc/config/firewall 

config defaults
	option syn_flood '1'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'

config zone
	option name 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option network 'lan'

config zone
	option name 'wan'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'
	option network 'wan wan6 wan1 wan2 tri'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option src_ip 'fc00::/6'
	option dest_ip 'fc00::/6'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config include
	option path '/etc/firewall.user'

config rule
	option enabled '1'
	option target 'ACCEPT'
	option name 'Allow-All-LAN-to-VPN'
	option dest 'vpn'
	option src 'lan'

config zone 'zone_vpn'
	option name 'vpn'
	option masq '1'
	option input 'REJECT'
	option forward 'ACCEPT'
	option output 'ACCEPT'
	option network 'glorytun omrvpn omr6in4'

config rule
	option enabled '1'
	option target 'ACCEPT'
	option name 'Allow-All-Ping'
	option proto 'icmp'
	option dest '*'
	option src '*'
	option icmp_type 'echo-request'

config rule
	option enabled '1'
	option target 'ACCEPT'
	option name 'Allow-VPN-ICMP'
	option proto 'icmp'
	option src 'vpn'

config rule
	option enabled '1'
	option target 'ACCEPT'
	option name 'Allow-Lan-to-Wan'
	option dest 'wan'
	option src 'lan'

config rule
	option enabled '1'
	option target 'ACCEPT'
	option name 'ICMPv6-Lan-to-OMR'
	option src 'lan'
	option family 'ipv6'
	option proto 'icmp'
	option limit '1000/sec'
	option icmp_type 'echo-reply destination-unreachable echo-request router-advertisement router-solicitation time-exceeded'

config rule 'allow_dhcp_request_vpn'
	option name 'Allow-DHCP-Request-VPN'
	option src 'vpn'
	option proto 'udp'
	option dest_port '67'
	option target 'ACCEPT'
	option family 'ipv4'

config include 'ss_rules'
	option path '/etc/firewall.ss-rules'
	option reload '1'

config include 'ss_rules6'
	option path '/etc/firewall.ss-rules6'
	option reload '1'
```

Ysurac, 2018-08-15:
All your configuration seems ok.
What is the result, from OpenMPTCProuter, of a ping to your server IP ?
Can you also ping each gateway and tell me the max ping time ?

I think it's the problem, it's slow to answer to requests done by status page, so status page put them in error.

To check if it's working without problems, look at system log if there is no errors. You can type also ```multipath``` if it's desactivated for interfaces tri or/and wwan, then there is a problem. You can increase timeout of omr-tracker if it's a slow ping time problem.

afriza, 2018-08-15:
Ping from OpenMPTCProuter to server
```
root@OpenMPTCProuter:~# ping 115.124.72.84
PING 115.124.72.84 (115.124.72.84): 56 data bytes
64 bytes from 115.124.72.84: seq=0 ttl=53 time=461.983 ms
64 bytes from 115.124.72.84: seq=1 ttl=53 time=513.686 ms
64 bytes from 115.124.72.84: seq=2 ttl=53 time=63.634 ms
64 bytes from 115.124.72.84: seq=3 ttl=53 time=86.681 ms
64 bytes from 115.124.72.84: seq=4 ttl=53 time=80.933 ms
64 bytes from 115.124.72.84: seq=5 ttl=53 time=70.721 ms
64 bytes from 115.124.72.84: seq=6 ttl=53 time=70.594 ms
64 bytes from 115.124.72.84: seq=7 ttl=53 time=70.372 ms
64 bytes from 115.124.72.84: seq=8 ttl=53 time=80.240 ms
64 bytes from 115.124.72.84: seq=9 ttl=53 time=69.827 ms
64 bytes from 115.124.72.84: seq=10 ttl=53 time=79.536 ms
64 bytes from 115.124.72.84: seq=11 ttl=53 time=219.375 ms
64 bytes from 115.124.72.84: seq=12 ttl=53 time=109.221 ms
64 bytes from 115.124.72.84: seq=13 ttl=53 time=98.914 ms
64 bytes from 115.124.72.84: seq=14 ttl=53 time=98.444 ms
64 bytes from 115.124.72.84: seq=15 ttl=53 time=98.257 ms
64 bytes from 115.124.72.84: seq=16 ttl=53 time=97.947 ms
64 bytes from 115.124.72.84: seq=17 ttl=53 time=157.866 ms
64 bytes from 115.124.72.84: seq=18 ttl=53 time=97.679 ms
64 bytes from 115.124.72.84: seq=19 ttl=53 time=157.246 ms
64 bytes from 115.124.72.84: seq=20 ttl=53 time=137.290 ms
64 bytes from 115.124.72.84: seq=21 ttl=53 time=96.972 ms
^C
--- 115.124.72.84 ping statistics ---
22 packets transmitted, 22 packets received, 0% packet loss
round-trip min/avg/max = 63.634/137.155/513.686 ms
```

Ping to Gateway `tri` (`10.64.64.64`) is not responding.. I don't know why internet works via this modem but there is this message during interface setting up: `Mon Aug 13 13:20:23 2018 daemon.warn pppd[24466]: Could not determine remote IP address: defaulting to 10.64.64.64`
```
root@OpenMPTCProuter:~# ping 10.64.64.64
PING 10.64.64.64 (10.64.64.64): 56 data bytes
^C
--- 10.64.64.64 ping statistics ---
234 packets transmitted, 0 packets received, 100% packet loss
```

```
Mon Aug 13 13:20:13 2018 daemon.notice netifd: Interface 'tri' is setting up now
Mon Aug 13 13:20:14 2018 user.notice post-tracking-post-tracking: omrvpn down
Mon Aug 13 13:20:15 2018 user.notice post-tracking-post-tracking: wan2 switched off
Mon Aug 13 13:20:16 2018 daemon.notice netifd: tri (23897): Trying to set mode
Mon Aug 13 13:20:16 2018 daemon.err omr-tracker[23736]: Command failed: Not found
Mon Aug 13 13:20:16 2018 daemon.notice pppd[24466]: pppd 2.4.7 started by root, uid 0
Mon Aug 13 13:20:16 2018 user.notice post-tracking-post-tracking: wan1 switched off
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: abort on (BUSY)
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: abort on (NO CARRIER)
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: abort on (ERROR)
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: report (CONNECT)
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: timeout set to 10 seconds
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: send (AT&F^M)
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: expect (OK)
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: AT&F^M^M
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: OK
Mon Aug 13 13:20:17 2018 local2.info chat[24701]:  -- got it
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: send (ATE1^M)
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: expect (OK)
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: ^M
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: ATE1^M^M
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: OK
Mon Aug 13 13:20:17 2018 local2.info chat[24701]:  -- got it
Mon Aug 13 13:20:17 2018 local2.info chat[24701]: send (AT+CGDCONT=1,"IP","3gprs"^M)
Mon Aug 13 13:20:18 2018 local2.info chat[24701]: timeout set to 30 seconds
Mon Aug 13 13:20:18 2018 local2.info chat[24701]: expect (OK)
Mon Aug 13 13:20:18 2018 local2.info chat[24701]: ^M
Mon Aug 13 13:20:18 2018 local2.info chat[24701]: AT+CGDCONT=1,"IP","3gprs"^M^M
[... 37 lines cut ...]
167 packets transmitted, 6 packets received, 96% packet loss
round-trip min/avg/max = 2.064/2.422/2.619 ms
```
And last, one... `multipath` reports `wlan0 is deactivated` but after reboot, it shows:
```
root@OpenMPTCProuter:~# multipath 
3g-tri is in default mode
6in4-omr6in4 is deactivated
eth0 is deactivated
ifb0 is in default mode
ifb1 is in default mode
lo is deactivated
sit0 is in default mode
tun0 is deactivated
wan1 is deactivated
wan2 is deactivated
wlan0 is in default mode
```

  1. The `No VPS IP address` seems misleading since I have set the IP. Maybe this can be fixed? After reboot it seems to be ok though.
[image]

  2. Why was `No WAN IP address` there? (it's no longer there, but i'm just curious why it appeared).
  3. My `WWAN` uses DHCP, should I change it to static IP [as described](https://github.com/Ysurac/openmptcprouter/wiki/OpenMPTCProuter-configuration)? Why?
  4. Can I delete `WAN1` and `WAN2` ?

Ysurac, 2018-08-15:
1. "No VPS IP address" is a test to get an IP address using internet. If it's slow then I display that.
2. "No WAN IP address" is the same test but using wan to check if I can get IP address.
3. Static IP is not required. It's only easier in most case :)
4. Yes.

afriza, 2018-08-16:
Thanks for the answers!

closed 2018-08-16 by afriza

neb007, 2018-12-01:
Hello,

neb007, 2018-12-01:
Hello, 
After setup the VPS , i have this erros message : no vps ip address " 
[image]

Any idea ? 

Thank you.

Ysurac, 2018-12-01:
Really ? Why do you put your message on not related issues ? You will never get any answer if you don't open a real issue.