Ysurac/openmptcprouter #133: crash? when IPv6 disabled

afriza, 2018-08-16:
## Expected Behavior
Internet connection stable when running http://www.speedtest.net
## Actual Behavior
speed test doesn't complete
crash?
```
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.173090] [<ffffff8008086fb4>] show_stack+0x14/0x20
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.180091] [<ffffff8008546db8>] dump_stack+0x9c/0xbc
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.187010] [<ffffff800842a228>] netdev_rx_csum_fault+0x38/0x48
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.194772] [<ffffff800841dc14>] __skb_checksum_complete+0x6c/0xc0
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.202752] [<ffffff8008502f80>] tcp_v6_rcv+0x1d8/0x10e0
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.209786] [<ffffff80084daa40>] ip6_input_finish+0x268/0x368
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.217205] [<ffffff80084dab70>] ip6_input+0x30/0x80
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.223748] [<ffffff80084da7b4>] ip6_rcv_finish+0x84/0xa8
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.230660] [<ffffff80084dae88>] ipv6_rcv+0x2c8/0x388
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.237147] [<ffffff8008428b38>] __netif_receive_skb_core+0x968/0xad8
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.245056] [<ffffff800842ac3c>] __netif_receive_skb+0x6c/0x78
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.252316] [<ffffff800842acf0>] process_backlog+0xa8/0x158
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.259314] [<ffffff800842e7b8>] net_rx_action+0x110/0x298
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.266223] [<ffffff8008081038>] __do_softirq+0x100/0x1fc
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.273043] [<ffffff80080984f4>] irq_exit+0x6c/0xc8
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.279311] [<ffffff800808d05c>] handle_IPI+0xa4/0x118
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.285811] [<ffffff8008080c90>] bcm2836_arm_irqchip_handle_irq+0x80/0xb0
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.294003] Exception stack(0xffffff8008803e30 to 0xffffff8008803f70)
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.301851] 3e20:                                   0000000000000020 0000000000000000
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.312363] 3e40: 0000000000000001 0000000000000000 ffffff8008803f50 0000004031cb8000
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.323089] 3e60: 000000015a5a417c 0000000000000000 0000000000000002 ffffff8008803ef0
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.334052] 3e80: 0000000000000700 0000000000000003 0000000000000078 ffffff80085664a0
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.345233] 3ea0: 0000000000000001 ffffff80085664c8 0000000000000000 ffffff80085664c8
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.356692] 3ec0: 0000000000000000 ffffff8008728828 ffffff8008707008 ffffff8008728000
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.368403] 3ee0: 0000000000000000 0000000000000000 ffffffc038ca8000 ffffff8008018000
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.380378] 3f00: ffffff800801c000 4e6f4716f925b9b9 ffffffc038ca8000 ffffff8008803f70
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.392606] 3f20: ffffff8008084714 ffffff8008803f70 ffffff8008084718 0000000000000145
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.405086] 3f40: ffffff8008018000 ffffff800801c000 ffffffffffffffff ffffff80080df97c
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.417729] 3f60: ffffff8008803f70 ffffff8008084718
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.425038] [<ffffff8008082870>] el1_irq+0xb0/0x140
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.432337] [<ffffff8008084718>] arch_cpu_idle+0x10/0x18
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.440025] [<ffffff80080c6364>] do_idle+0xec/0x1b8
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.447196] [<ffffff80080c6570>] cpu_startup_entry+0x20/0x28
Thu Aug 16 06:10:05 2018 kern.warn kernel: [  303.455116] [<ffffff800808cbcc>] secondary_start_kernel+0x17c/0x188
Thu Aug 16 06:10:21 2018 kern.err kernel: [  318.813440] eth0: hw csum failure
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.818956] CPU: 1 PID: 11978 Comm: pgrep Not tainted 4.14.62 #0
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.827124] Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.835529] Call trace:
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.839975] [<ffffff8008086bf0>] dump_backtrace+0x0/0x3b0
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.847403] [<ffffff8008086fb4>] show_stack+0x14/0x20
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.853354] [<ffffff8008546db8>] dump_stack+0x9c/0xbc
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.859273] [<ffffff800842a228>] netdev_rx_csum_fault+0x38/0x48
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.866054] [<ffffff800841dc14>] __skb_checksum_complete+0x6c/0xc0
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.873078] [<ffffff8008502f80>] tcp_v6_rcv+0x1d8/0x10e0
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.879188] [<ffffff80084daa40>] ip6_input_finish+0x268/0x368
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.885718] [<ffffff80084dab70>] ip6_input+0x30/0x80
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.891415] [<ffffff80084da7b4>] ip6_rcv_finish+0x84/0xa8
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.897521] [<ffffff80084dae88>] ipv6_rcv+0x2c8/0x388
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.903248] [<ffffff8008428b38>] __netif_receive_skb_core+0x968/0xad8
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.910389] [<ffffff800842ac3c>] __netif_receive_skb+0x6c/0x78
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.916900] [<ffffff800842acf0>] process_backlog+0xa8/0x158
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.923146] [<ffffff800842e7b8>] net_rx_action+0x110/0x298
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.929301] [<ffffff8008081038>] __do_softirq+0x100/0x1fc
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.935374] [<ffffff80080984f4>] irq_exit+0x6c/0xc8
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.940911] [<ffffff80080d1270>] __handle_domain_irq+0x88/0xb0
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.947401] [<ffffff8008080cb4>] bcm2836_arm_irqchip_handle_irq+0xa4/0xb0
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.954867] Exception stack(0xffffff800bf63a40 to 0xffffff800bf63b80)
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.961978] 3a40: ffffffc0342dc025 ffffff8008668f71 0000000000000001 0000000000000020
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.971091] 3a60: ffffff800bf63b60 8080808080808000 ffffffc0342dc026 fefefefefefeff1f
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.980303] 3a80: 7f7f7f7f7f7f7f7f 7174726e6b626d44 7f7f7f7f7f7f7f7f 0101010101010101
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.989630] 3aa0: 0000000000000008 0000000000000001 0000000000000038 0000000000000000
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  318.999034] 3ac0: ffffff8008152c80 0000007f9ee74ca4 0000000000000000 ffffffc0354f4080
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  319.008576] 3ae0: 0000000000001000 0000000000000025 0000000000000001 ffffff8008668f70
Thu Aug 16 06:10:21 2018 kern.warn kernel: [  319.018230] 3b00: 0000000000000000 0000000000000000 ffffff8008668000 00000000ffffffec
[... 566 lines cut ...]
Thu Aug 16 06:10:42 2018 kern.warn kernel: [  340.881215] 3f40: ffffff8008008000 ffffff800800c000 ffffffffffffffff ffffff80080df97c
Thu Aug 16 06:10:42 2018 kern.warn kernel: [  340.891220] 3f60: ffffff8008073f70 ffffff8008084718
Thu Aug 16 06:10:42 2018 kern.warn kernel: [  340.897197] [<ffffff8008082870>] el1_irq+0xb0/0x140
Thu Aug 16 06:10:42 2018 kern.warn kernel: [  340.903172] [<ffffff8008084718>] arch_cpu_idle+0x10/0x18
Thu Aug 16 06:10:42 2018 kern.warn kernel: [  340.909571] [<ffffff80080c6364>] do_idle+0xec/0x1b8
Thu Aug 16 06:10:42 2018 kern.warn kernel: [  340.915485] [<ffffff80080c6570>] cpu_startup_entry+0x20/0x28
Thu Aug 16 06:10:42 2018 kern.warn kernel: [  340.922178] [<ffffff800808cbcc>] secondary_start_kernel+0x17c/0x188
Thu Aug 16 06:10:47 2018 daemon.err uhttpd[2572]: Host whois.apnic.net not found.
```

## Steps to Reproduce the Problem

  1. create 3G connection and add to WAN firewall. set it as master multipath.
  2. configure server via wizard. delete WAN1 & WAN2.
  3. System > OpenMPTCProuter > Advanced Settings > tick `Disable IPv6` > Save & Apply
  4. Run speedtest.net in browser (via LAN)

## Specifications

  - OpenMPTCProuter version: (0.31.6)
  - OpenMPTCProuter VPS version: (0.44)
  - OpenMPTCProuter platform: (RPI3)

## Extra question
  - Is it safe to flash OpenMPTCProuter over previous version (I'm using Etcher)? The 'data' doesn't seem to be wiped that way, so I decided to flash the sdcard twice to wipe the data. 1x using other big image and 1x using OpenMPTCProuter. I'm using mac, so there is no `conv=fsync` when using `dd`.

Ysurac, 2018-08-16:
This should be fixed in OpenMPTCProuter v0.31.7

No idea about mac, I only use Linux.

afriza, 2018-08-20:
Hi @Ysurac , the `Enable IPv6` is ticked after Save & Apply even after being unticked.. OMR v0.32.1

Ysurac, 2018-08-21:
Fixed in OpenMPTCProuter 0.32.4

closed 2018-08-21 by Ysurac