Ysurac/openmptcprouter #134: Netflix does not work

thomasforte, 2018-08-16:
## Expected Behavior

Playing Netflix contents.

## Actual Behavior

Netflix detect I use a proxy or unblocker. So i can't play any contents.

[image]

## Steps to Reproduce the Problem

  1. Vps is installed in an OVH Vps with OBFS=yes option
  2. "Enable ShadowSocks OBFS" is selected in the the configuration avancé of openmptcprouter.

Try playing any movies on netflix with firefox browser.

In previous versions of openmptcprouter, "Enable ShadowSocks OBFS" option doesn't exist. I think (I'm not sure) that Netflix playing has work well before. 

## Specifications

  - OpenMPTCProuter version: v0.31.6
  - OpenMPTCProuter VPS version: v0.44
  - OpenMPTCProuter platform: x86_64 on virtualbox

Ysurac, 2018-08-16:
It's not related to OpenMPTCProuter.
Netflix ban many VPS provider, you can try to find one that is not banned or use omr-bypass to use one of you WAN to access it directly (like issue https://github.com/Ysurac/openmptcprouter/issues/112).

OBFS is not related to the issue, it's more to bypass China Great Firewall.

thomasforte, 2018-08-20:
Ok, i understand. I'ts work well for me by setting up OMR-Bypass.

closed 2018-08-21 by Ysurac