Ysurac/openmptcprouter #135: "multipath blocked on the connection" message

nileio, 2018-08-20:
## Expected Behavior

status page displays no errors

## Actual Behavior
wan 2 includes the following message
"multipath blocked on the connection"

what does this message mean ? It does not seem to affect the overall performance / aggregated speeds though
## Steps to Reproduce the Problem

  1. rpi image flash
  2. using wizard to configure wan1 and wan2 only
  3.

## Specifications

  - OpenMPTCProuter version: 32.1
  - OpenMPTCProuter VPS version: 0.44
  - OpenMPTCProuter platform: RPI2

Ysurac, 2018-08-20:
This mean that multipath flags seems to be removed by a middle box on this connection.

You can test if it's really the case by using the tab *MPTCP Support Check* in **System**->**OpenMPTCProuter** an test the wan2 interface.

nileio, 2018-08-20:
i tried the test - it returns Bad Gateway for wan2.
wan2 is 4G
some observations: the gateway IP reported on MPTCP is wrong. It is not my public gateway IP which is reported on my modem.
i am doing some more tests, but not really sure why the scripts return that MPTCP is not enabled for wan2 interface..
any ideas ?

nileio, 2018-08-20:
Here is the exact result from the test for wan2
"Bad Gateway
The process did not produce any response"

Ysurac, 2018-08-20:
It take too much time to answer, it's why there is the *multipath blocked on the connection*.

Which gateway IP ? Only gateways displayed are wan gateway that you set.
If you mean public IP in wan* section, it's the public IP used on the internet.

nileio, 2018-08-20:
yes, sorry I meant my public IP reported in the WAN section..

the results of this test used to work and return results similar to those returned for wan1.
so is this a timeout issue ?

SoGooDFR, 2018-08-23:
I've the same issue.

Ysurac, 2018-08-23:
Which issue ?

Ysurac, 2018-08-24:
You can try to update to OpenMPTCProuter v0.33
You can check if it's a real error or not by using the tab **MPTCP Support Check** in **System**->**OpenMPTCProuter** an test the wan1 interface.

ThibautBascoulergue, 2018-08-24:
Same with version v0.33.

Ysurac, 2018-08-24:
And the result of the MPTCP support Check ?

ThibautBascoulergue, 2018-08-24:
1: 192.168.1.1 
2: * 
3: 10.4.2.10  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
4: 172.16.80.252  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
5: 77.154.230.105  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
6: 77.136.10.154  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
7: * 
8: * 
9: * 
10: * 
11: * 
12: * 
13: * 
14: * 
15: * 
16: * 
17: * 
18: * 
19: * 
20: * 
21: * 
22: * 
23: * 
24: * 
25: * 
26: * 
27: * 
28: * 
29: * 
30: * 
31: * 
32: * 
33: * 
34: * 
35: * 
36: * 
37: * 
38: * 
39: * 
40: * 
41: * 
42: * 
43: * 
44: * 
45: * 
46: * 
47: * 
48: * 
49: * 
50: * 
51: * 
52: * 
53: * 
54: * 
55: * 
56: * 
57: * 
58: * 
59: * 
60: * 
61: * 
62: * 
63: * 
64: * 

Result:
nil

Ysurac, 2018-08-24:
ok, not many info here... A last test: on OpenMPTCProuter via SSH, run ```omr-mptcp-intf wan1```.

ThibautBascoulergue, 2018-08-24:
root@OpenMPTCProuter:~# omr-mptcp-intf wan1
Nay, Nay, Nay, your have an old computer that does not speak MPTCP. Shame on you!

ThibautBascoulergue, 2018-08-24:
you think its my os from gateway wan 1 ?

Ysurac, 2018-08-24:
MPTCP flag seems to be removed somewhere by a middle box on wan1.
Do you have a real aggregation or not ?

ThibautBascoulergue, 2018-08-24:
router 192.168.1.1 

root@OpenWrt:~# uname -a
Linux OpenWrt 4.9.77 #0 Wed Jan 31 20:18:16 2018 mips GNU/Linux

ThibautBascoulergue, 2018-08-24:
Yes I think aggregation is ok 

[image]

Ysurac, 2018-08-24:
ok, I will put this as a warning only in next release. No idea why MPTCP flag can be removed and working... Maybe something strange on SFR network.

ThibautBascoulergue, 2018-08-24:
Ok :), So SFR...suck betwen FREE :) in my country ^^

Ysurac, 2018-08-24:
I only make my test with Free and Free mobile for now.

ThibautBascoulergue, 2018-08-24:
I have sysupgrade my 4G rooter 'wan1 SFR' with the latest snapshot of openwrt, and the problem is same :/
I think the problem comes from the sft network.
With the overthebox, I did not have this alert message, is it the problem was the same..I think so but it was not visible on the interface: /

Ysurac, 2018-08-24:
If aggregation really work, then we don't care. For OVH, all french FAI work with MPTCP. This may be a strange false alert.

OpenMPTCProuter is not a fork of OverTheBox even if I use ShadowSocks, Glorytun and the same look for the status page :)

ThibautBascoulergue, 2018-08-24:
Hehe, the aggregation works fine so it's cool. Anyway thank you for the project because it looks more stable than overthebox and faster. Before you go on your solution,  I had big flow problem on my OTB while nothing else.  Sometimes at the same time I had 30 ~ 40mbits in aggregation with OVH and now 👍 
[image]

So thanks to you for the job

Ysurac, 2018-08-29:
MPTCP check will be fixed in next release.

closed 2018-08-29 by Ysurac

neb007, 2018-11-30:
Hello,

I have the same problem. I have two wan ; SFR and Bouygues Telecom , i test it with OverTheBox it's work well. But with OpenMptcpRouter i have message "multipath blocked on the connection" . 
Version of OpenMPTCP : openmptcprouter v0.42.5 r7405-06a20afb34

Ysurac, 2018-11-30:
If you read the status page, you can see that you have a problem: the VPS server seems to be not reachable.
You should open a new issue, it's not related to this case.

neb007, 2018-12-01:
Hello Ysurac,

After installing the VPS , it's work but still have message in vps : no vps ip address  #132 

Thx

Ysurac, 2018-12-01:
Open an issue !