Ysurac/openmptcprouter #136: intermittent dnsmasq only returns ipv6

nileio, 2018-08-20:
## Expected Behavior 
ipv4 returned for dhcp requesting clients (not sure if it is only relevant to apple devices but it seems to exist across the board)

## Actual Behavior
only ipv6 is returned, no ipv4.. for those clients which have ipv6 enabled.
the behavior may be relevant to the way devices are booted / race condition with the modem.. No clue

## Steps to Reproduce the Problem

  1. using the internal switch of the modem
  2.  and fresh image  
  3. reboot

## Possible Solution

what worked with me is to restart dnsmasq manually
that is
/etc/init.d/dnsmasq restart

though there is an error returned udhcpc: no lease, failing. the above seems to fix the problem
yet to find out if it is a bug in dnsmasq or is it some config problem..

## Specifications

  - OpenMPTCProuter version: v0.32.1
  - OpenMPTCProuter VPS version: 0.33
  - OpenMPTCProuter platform: RPI2

Ysurac, 2018-08-20:
If it's only about Apple device, then this may be the same as #121

nileio, 2018-08-20:
@Ysurac yes it maybe related issue .. solution I suggested here works for me to fix dnsmasq dhcp issue for now

Ysurac, 2019-02-03:
A stable beta is a fun idea...
You should disable IPv6 or disable Router Advertisement-Service and DHCPv6-Service on LAN interface.

Maudissure, 2019-02-03:
Hi, 

Please, can you explain how to disable all IPV6 connections on OMR and VPS ?

Thanks.

Ysurac, 2019-02-03:
Checkbox on the OpenMPTCProuter wizard or setting page. This will not disable IPv6 on the VPS.

Maudissure, 2019-02-03:
Ok but IPV6 will be disable on LAN, correct ?

Ysurac, 2019-02-03:
True

Ysurac, 2019-02-03:
it's not related to IPv6. If the Apple Device decline IP the DHCP offer, I can't do anything for that. The bug is on Apple devices.

Maudissure, 2019-02-03:
Tout d'abord, pardon pour la réponse en français,

Je ne comprends pas ta réponse. Tous les appareils Apple que je possède fonctionnent tous parfaitement sur deux routeurs différents autres que l'OMR. Tu m'affirmes que s'il ne fonctionnent pas sur OMR, c'est qu'ils sont bugués. Je trouve que c'est un peu la réponse du, m'embêtes pas, c'est pas ma faute.

C'est un peu facile non ? 

J'accepterai beaucoup mieux que tu me dises que l'architecture OMR ne peut pas fonctionner avec les devices Apple, auquel cas je cesserai immédiatement de t'embêter avec et passerai à autre chose. Ce qui serait fort dommage puisque je trouve le projet vraiment intéressant et serait prêt à mettre la main à la poche si je savait qu'il serait possible une fois terminé, que je puisse l'utiliser chez moi, et ce, pour tous les appareils que je possède.

N'es-tu pas d'accord que si un routeur est incapable de tourner avec l'ensemble des appareils potentiellement présents chez les utilisateurs, celui-ci ne sert pour ainsi dire que de proof of concept mais ne pourra jamais être diffusé au grand public ?
Je ne sais pas quel est ton but pour ce projet, est-ce simplement un hobby ? Ou bien une manière de se faire reconnaitre auprès de la communauté ? Ou enfin, un projet réellement professionnel à but lucratif ?

Ne prends pas mal ces remarques, ce sont de réelles questions que je me pose.

Enfin, si je peux me permettre un humble conseil. Je pense, que plutôt que d'ajouter régulièrement des fonctionnalités au système, tu devrais te concentrer sur la réduction du backlog de bugs et proposer une version stabilisée en beta, quitte à ce que toutes les options ne soient pas disponibles. Mais qui fonctionne sur un maximum d'appareils. Pour ensuite ajouter petit à petit des fonctionnalités ou améliorations.
Ne te trompes pas, je t'admire réellement pour le temps passé à mener seul ce genre de projet Et je n'attend qu'une chose, c'est que celui-ci soit suffisamment stable pour en parler autour de moi. Car je pense sincèrement que tu gagnerais à être connu.

Ysurac, 2019-02-03:
Tout ce que je peux dire d'après les logs que tu me donnes c'est que les machines Apple refusent l'adresse que le serveur DHCP lui propose. OpenMPTCProuter s'appuie sur OpenWRT, et la partie DHCP utilise dnsmasq. Si tu trouves dans les issues de dnsmasq ou d'OpenWRT une explication et surtout une manière de corriger le soucis je l'appliquerais sans soucis. Par contre n'ayant aucune machine Apple je ne peux réaliser aucun tests, n'ayant pas de soucis sous Linux, Windows ou Android. Apple n'offre pas, à ma connaissance, de machine virtuelle permettant de tester.

Sinon OpenMPTCProuter n'est pas un project à but lucratif. Sinon il ne serait pas sur Github et je vendrais quelque chose ;)

La version 0.46 de OpenMPTCProuter qui sera disponibles d'ici peu de temps ajoute uniquement le support du plugin v2ray pour Shadowsocks, le reste est uniquement des corrections et amélioration de la stabilité, ce qui devrait permettre de fermer pas mal de bugs.

closed 2019-02-20 by Ysurac