Ysurac/openmptcprouter #137: OMR tracker can't establish connection - Status page not loading

Maudissure, 2018-08-26:
## Expected Behavior

Load of status page

## Actual Behavior

Status page not loading

got this message in system log : 

Sun Aug 26 10:22:49 2018 daemon.err omr-tracker-ss[3604]: Failed to establish connection
Sun Aug 26 10:23:03 2018 daemon.err omr-tracker-ss[3604]: Failed to establish connection
Sun Aug 26 10:23:16 2018 daemon.err omr-tracker-ss[3604]: Failed to establish connection
Sun Aug 26 10:23:30 2018 daemon.err omr-tracker-ss[3604]: Failed to establish connection
Sun Aug 26 10:23:44 2018 daemon.err omr-tracker-ss[3604]: Failed to establish connection
Sun Aug 26 10:23:57 2018 daemon.err omr-tracker-ss[3604]: Failed to establish connection
Sun Aug 26 10:24:11 2018 daemon.err omr-tracker-ss[3604]: Failed to establish connection
Sun Aug 26 10:24:24 2018 daemon.err omr-tracker-ss[3604]: Failed to establish connection

## Steps to Reproduce the Problem

  1.Select status page

## Specifications

  - OpenMPTCProuter version: V0.33.1
  - OpenMPTCProuter VPS version: V0.46
  - OpenMPTCProuter platform: RPI3B+

Ysurac, 2018-08-26:
ShadowSocks seems to be not working correctly. Check username, password and that it's running on VPS part.

Ysurac, 2018-08-26:
For the status page, I would need the result of ```http://192.168.100.1/cgi-bin/luci/admin/system/openmptcprouter/interfaces_status```

Maudissure, 2018-08-26:
Shadowsocks key is ok.

Result for this address check : 

{"wans":[{"ifname":"wan1","ipaddr":"192.168.10.2","label":"wan1","wanip":"82.237.195.51","whois":"PROXAD","ipv6_discover":"NONE","status":"OK","link":"\/cgi-bin\/luci\/admin\/network\/network\/wan1","multipath":"on","multipath_available":"OK","gw_ping":"UP","name":"wan1","gateway":"192.168.10.1","mtu":"1500","server_ping":"UP","latency":"13"},{"ifname":"wan2","ipaddr":"192.168.11.2","label":"wan2","wanip":"37.169.88.162","whois":"FREEM","ipv6_discover":"NONE","status":"OK","link":"\/cgi-bin\/luci\/admin\/network\/network\/wan2","multipath":"master","multipath_available":"OK","gw_ping":"UP","name":"wan2","gateway":"192.168.11.1","mtu":"1476","server_ping":"UP","latency":"34"},{"ifname":"wan3","ipaddr":"192.168.12.2","label":"wan3","wanip":"37.171.22.66","whois":"FREEM","ipv6_discover":"NONE","status":"OK","link":"\/cgi-bin\/luci\/admin\/network\/network\/wan3","multipath":"on","multipath_available":"OK","gw_ping":"UP","name":"wan3","gateway":"192.168.12.1","mtu":"1475","server_ping":"UP","latency":"175"}],"tunnels":[],"openmptcprouter":{"local_addr":"192.168.1.2","dhcpd":{"lan":{"range_end":"192.168.1.249","dns":"192.168.1.2","router":"192.168.1.2","netmask":"255.255.255.0","range_start":"192.168.1.100","leasetime":"12h","interface":"lan"}},"remote_from_lease":false,"tun_state":"UP","remote_addr":"192.168.1.100","tun_service":true,"server_mptcp":"","tun6_state":"UP","service_addr":"51.254.127.83","ipv6":"enabled","dns":true,"socks_service_enabled":true,"socks_service":true,"ss_addr":"51.254.127.83","version":"\"v0.33.1\"","wan_addr6":"","uptime":"5368.79","tun_ip":"10.255.255.2","wan_addr":"51.254.127.83","loadavg":"0.09 0.13 0.16"}}

Ysurac, 2018-08-26:
Status page will be fixed in 0.33.2 in a few hours (while CircleCI compile all images)
All seems to be ok in status page info.

Maudissure, 2018-08-27:
Ok for me, working now.

closed 2018-08-27 by Ysurac