Ysurac/openmptcprouter #138: OpenMPTCTP Status loop

ThibautBascoulergue, 2018-08-26:
## Expected Behavior

In the last version of router, 33.1. Status of MPTCTP connexion runs in loop and nothing is done.

## Actual Behavior

## Steps to Reproduce the Problem

[image]

## Specifications

  - OpenMPTCProuter version: (33.1)
  - OpenMPTCProuter VPS version: (OpenMPTCProuter VPS 0.46)
  - OpenMPTCProuter platform: x86_64 on ESXi

Ysurac, 2018-08-26:
What is the result of ```http://192.168.1.18/cgi-bin/luci/admin/system/openmptcprouter/interfaces_status``` ? and what do you have in logs ?

ThibautBascoulergue, 2018-08-26:
{"wans":[{"ifname":"wan1","ipaddr":"192.168.1.10","label":"wan1","wanip":"77.136.17.2","whois":"LDCOMNET","ipv6_discover":"NONE","status":"WARNING","link":"\/cgi-bin\/luci\/admin\/network\/network\/wan1","multipath":"master","multipath_available":"ERROR","gw_ping":"UP","name":"wan1","gateway":"192.168.1.1","mtu":"","server_ping":"UP","latency":"41"},{"ifname":"wan2","ipaddr":"192.168.3.10","label":"wan2","wanip":"37.173.219.192","whois":"FREEM","ipv6_discover":"NONE","status":"OK","link":"\/cgi-bin\/luci\/admin\/network\/network\/wan2","multipath":"on","multipath_available":"OK","gw_ping":"UP","name":"wan2","gateway":"192.168.3.1","mtu":"1476","server_ping":"UP","latency":"59"}],"tunnels":[],"openmptcprouter":{"local_addr":"192.168.1.254","dhcpd":{"lan":{"range_end":"192.168.1.249","dns":"192.168.1.254","router":"192.168.1.254","netmask":"255.255.255.0","range_start":"192.168.1.100","leasetime":"12h","interface":"lan"}},"remote_from_lease":false,"tun_state":"UP","remote_addr":"192.168.1.227","tun_service":true,"server_mptcp":"","tun6_state":"","service_addr":"86.105.212.170","ipv6":"disabled","dns":true,"socks_service_enabled":true,"socks_service":true,"ss_addr":"86.105.212.170","version":"\"v0.33.1\"","wan_addr6":"","uptime":"65.66","tun_ip":"10.255.255.2","wan_addr":"86.105.212.170","loadavg":"0.37 0.12 0.04"}}

ThibautBascoulergue, 2018-08-26:
nothing in log.

Ysurac, 2018-08-26:
This will be fixed in 0.33.2, in a few hours (while CircleCI compile all images).

closed 2018-08-26 by Ysurac

ThibautBascoulergue, 2018-08-26:
ok fine, thank's :)