Ysurac/openmptcprouter #141: mptcp_bbr module not available in VPS. MPTCP not enabled on the server in state openmptcprouter

Cmoi20, 2018-08-28:
## Expected Behavior
VPS with mptcp_bbr module available

## Actual Behavior
in system -> openmptcprouter -> Etat , openmptcprouter show message "mptcp is not enabled on the server".
in VPS, lsmod |grep mptcp show module mptcp_olia runing.
in list of available module, mptcp_bbr is not available

## Steps to Reproduce the Problem

  1. use last version of openmptcprouter and last version of VPS
  2. show state of openmptcprouter
  3. issue was showing

## Specifications

  - OpenMPTCProuter version: 0.33.3
  - OpenMPTCProuter VPS version: 46
  - OpenMPTCProuter platform: VMWARE

Ysurac, 2018-08-28:
There is no mptcp_bbr module, it's tcp_bbr.
What is the result of **MPTCP Support Check** in **System**->**OpenMPTCProuter** for all interfaces ?

Ysurac, 2018-08-29:
MPTCP check will be fixed in next release.

closed 2018-08-29 by Ysurac