Ysurac/openmptcprouter #142: Failed to use 2 additional USB Ethernet

afriza, 2018-08-29:
## Expected Behavior

 WAN1 & WAN2 can be used with `eth1` and `eth2`.

## Actual Behavior
When adding **second** EDIMAX USB 2.0 Fast Ethernet Adapter, the following error occurs
```
ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
```
`WAN2` gets NO IP address and the whole system hangs after a while.

Full log
```
Wed Aug 29 04:02:39 2018 kern.info kernel: [  221.545033] usb 1-1.3: new high-speed USB device number 6 using dwc_otg
Wed Aug 29 04:02:39 2018 kern.info kernel: [  221.697956] usb 1-1.3: New USB device found, idVendor=0b95, idProduct=772b
Wed Aug 29 04:02:39 2018 kern.info kernel: [  221.706831] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Wed Aug 29 04:02:39 2018 kern.info kernel: [  221.716103] usb 1-1.3: Product: AX88772B
Wed Aug 29 04:02:39 2018 kern.info kernel: [  221.721858] usb 1-1.3: Manufacturer: ASIX Elec. Corp.
Wed Aug 29 04:02:39 2018 kern.info kernel: [  221.728744] usb 1-1.3: SerialNumber: 15426E
Wed Aug 29 04:02:39 2018 kern.debug kernel: [  221.735095] usb 1-1.3: no of_node; not parsing pinctrl DT
Wed Aug 29 04:02:39 2018 kern.debug kernel: [  221.736541] asix 1-1.3:1.0: no of_node; not parsing pinctrl DT
Wed Aug 29 04:02:39 2018 daemon.notice netifd: Interface 'wan2' is enabled
Wed Aug 29 04:02:39 2018 kern.info kernel: [  222.190437] asix 1-1.3:1.0 eth2: register 'asix' at usb-3f980000.usb-1.3, ASIX AX88772B USB 2.0 Ethernet, 00:50:b6:15:42:6e
Wed Aug 29 04:02:41 2018 daemon.notice netifd: Network device 'eth2' link is up
Wed Aug 29 04:02:41 2018 daemon.notice netifd: Interface 'wan2' has link connectivity
Wed Aug 29 04:02:41 2018 daemon.notice netifd: Interface 'wan2' is setting up now
Wed Aug 29 04:02:41 2018 kern.info kernel: [  223.750492] asix 1-1.3:1.0 eth2: link up, 100Mbps, full-duplex, lpa 0x41E1
Wed Aug 29 04:02:41 2018 daemon.notice netifd: wan2 (4924): udhcpc: started, v1.28.3
Wed Aug 29 04:02:41 2018 daemon.notice netifd: wan2 (4924): udhcpc: sending discover
Wed Aug 29 04:02:41 2018 kern.warn kernel: [  223.975061] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
Wed Aug 29 04:02:41 2018 kern.warn kernel: [  223.975061]
Wed Aug 29 04:02:42 2018 kern.err kernel: [  224.624981] raspberrypi-firmware soc:firmware: Get Throttled mailbox call failed
Wed Aug 29 04:02:44 2018 daemon.notice netifd: wan2 (4924): udhcpc: sending discover
Wed Aug 29 04:02:47 2018 daemon.notice netifd: wan2 (4924): udhcpc: sending discover
```

## Steps to Reproduce the Problem

  1. Flash OMR v0.33.3
  2. Set `Maximum scaling CPU frequency` to `600000` to silence [cpufreq errors](https://github.com/Ysurac/openmptcprouter/issues/113)
  3. Plug in EDIMAX USB 2.0 Fast Ethernet Adapter. Set `WAN1` to use `eth1` with DHCP.
  4. Plug in another EDIMAX USB 2.0 Fast Ethernet Adapter. Set `WAN2` to use `eth2` with DHCP.

## Specifications

  - OpenMPTCProuter version: (v0.33.3)
  - OpenMPTCProuter VPS version: (v0.46)
  - OpenMPTCProuter platform: (RPI3)

Ysurac, 2018-08-29:
I would need kernel log too.
Why to you add USB Ethernet adapter ? It's often not a good idea on a RPI...

Ysurac, 2018-08-29:
This looks like a power problem... But I would need a few lines before: the beginning of the kernel dump.

afriza, 2018-08-29:
[image]

I used 2 additional Ethernet Adapters because I don't have a hub/switch at hand.. I assume the [guide](https://github.com/Ysurac/openmptcprouter/wiki/OpenMPTCProuter-configuration) uses a hub/switch to set up?

Ysurac, 2018-08-29:
I think it's a power supply problem.

The guide use a switch (there is often switch on modems now).

afriza, 2018-09-03:
I confirmed that using the additional two EDIMAX USB 2.0 Fast Ethernet Adapter (via ASIX driver) works on Raspbian Stretch 9.4 Desktop version. Thus, this is not a problem with the power supply.
The Linux kernel version used is:
```
pi@raspberrypi:~$ uname -a
Linux raspberrypi 4.14.62-v7+ #1134 SMP Tue Aug 14 17:10:10 BST 2018 armv7l GNU/Linux
pi@raspberrypi:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.4 (stretch)
Release:        9.4
Codename:       stretch
```

afriza, 2018-09-04:
Unfortunately this bug also appear when 2 cdc_ether devices are connected. So it is not a problem with the ASIX and probably not a problem with the cdc_ether either but something else.
```
Tue Sep  4 02:19:45 2018 kern.info kernel: [  300.400799] usb 1-1.3: new high-speed USB device number 7 using dwc_otg
Tue Sep  4 02:19:45 2018 kern.info kernel: [  300.542451] usb 1-1.3: New USB device found, idVendor=12d1, idProduct=1f01
Tue Sep  4 02:19:45 2018 kern.info kernel: [  300.553385] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Tue Sep  4 02:19:45 2018 kern.info kernel: [  300.564800] usb 1-1.3: Product: HUAWEI_MOBILE
Tue Sep  4 02:19:45 2018 kern.info kernel: [  300.573241] usb 1-1.3: Manufacturer: HUAWEI_MOBILE
Tue Sep  4 02:19:45 2018 kern.info kernel: [  300.582146] usb 1-1.3: SerialNumber: 0123456789ABCDEF
Tue Sep  4 02:19:45 2018 kern.debug kernel: [  300.591658] usb 1-1.3: no of_node; not parsing pinctrl DT
Tue Sep  4 02:19:45 2018 kern.debug kernel: [  300.654099] uas 1-1.3:1.0: no of_node; not parsing pinctrl DT
Tue Sep  4 02:19:45 2018 kern.debug kernel: [  300.654228] usb-storage 1-1.3:1.0: no of_node; not parsing pinctrl DT
Tue Sep  4 02:19:45 2018 kern.info kernel: [  300.654275] usb-storage 1-1.3:1.0: USB Mass Storage device detected
Tue Sep  4 02:19:45 2018 kern.info kernel: [  300.665408] scsi host1: usb-storage 1-1.3:1.0
Tue Sep  4 02:19:45 2018 kern.info kernel: [  300.875755] usb 1-1.3: USB disconnect, device number 7
Tue Sep  4 02:19:46 2018 kern.info kernel: [  301.680798] usb 1-1.3: new high-speed USB device number 8 using dwc_otg
Tue Sep  4 02:19:46 2018 kern.info kernel: [  301.822311] usb 1-1.3: New USB device found, idVendor=12d1, idProduct=14dc
Tue Sep  4 02:19:46 2018 kern.info kernel: [  301.833332] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Tue Sep  4 02:19:46 2018 kern.info kernel: [  301.844696] usb 1-1.3: Product: HUAWEI_MOBILE
Tue Sep  4 02:19:46 2018 kern.info kernel: [  301.852999] usb 1-1.3: Manufacturer: HUAWEI_MOBILE
Tue Sep  4 02:19:46 2018 kern.debug kernel: [  301.862082] usb 1-1.3: no of_node; not parsing pinctrl DT
Tue Sep  4 02:19:47 2018 kern.debug kernel: [  302.000435] cdc_ether 1-1.3:1.0: no of_node; not parsing pinctrl DT
Tue Sep  4 02:19:47 2018 daemon.notice netifd: Interface 'wan2' is enabled
Tue Sep  4 02:19:47 2018 daemon.notice netifd: Network device 'eth2' link is up
Tue Sep  4 02:19:47 2018 daemon.notice netifd: Interface 'wan2' has link connectivity
Tue Sep  4 02:19:47 2018 daemon.notice netifd: Interface 'wan2' is setting up now
Tue Sep  4 02:19:47 2018 kern.info kernel: [  302.052562] cdc_ether 1-1.3:1.0 eth2: register 'cdc_ether' at usb-3f980000.usb-1.3, CDC Ethernet Device, 0c:5b:8f:27:9a:64
Tue Sep  4 02:19:47 2018 kern.err kernel: [  302.053917] cdc_ether 1-1.3:1.0 eth2: kevent 12 may have been dropped
Tue Sep  4 02:19:47 2018 kern.err kernel: [  302.079125] cdc_ether 1-1.3:1.0 eth2: kevent 11 may have been dropped
Tue Sep  4 02:19:47 2018 kern.warn kernel: [  302.081053] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
Tue Sep  4 02:19:47 2018 kern.warn kernel: [  302.081053]
Tue Sep  4 02:19:47 2018 daemon.notice netifd: Network device 'eth2' link is down
Tue Sep  4 02:19:47 2018 daemon.notice netifd: Interface 'wan2' has link connectivity loss
Tue Sep  4 02:19:48 2018 kern.err kernel: [  303.591553] cdc_ether 1-1.3:1.0 eth2: kevent 12 may have been dropped
Tue Sep  4 02:19:48 2018 kern.err kernel: [  303.670855] raspberrypi-firmware soc:firmware: Get Throttled mailbox call failed
Tue Sep  4 02:19:52 2018 kern.debug kernel: [  307.192007] uas 1-1.3:1.2: no of_node; not parsing pinctrl DT
Tue Sep  4 02:19:52 2018 kern.debug kernel: [  307.192111] usb-storage 1-1.3:1.2: no of_node; not parsing pinctrl DT
Tue Sep  4 02:19:52 2018 kern.info kernel: [  307.192157] usb-storage 1-1.3:1.2: USB Mass Storage device detected
Tue Sep  4 02:19:52 2018 kern.info kernel: [  307.203482] scsi host1: usb-storage 1-1.3:1.2
Tue Sep  4 02:19:53 2018 kern.notice kernel: [  308.237177] scsi 1:0:0:0: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
Tue Sep  4 02:19:53 2018 kern.debug kernel: [  308.250125] sd 1:0:0:0: no of_node; not parsing pinctrl DT
Tue Sep  4 02:19:53 2018 kern.notice kernel: [  308.253152] sd 1:0:0:0: [sdb] Attached SCSI removable disk
Tue Sep  4 02:19:56 2018 daemon.notice netifd: Network device 'eth2' link is up
Tue Sep  4 02:19:56 2018 daemon.notice netifd: Interface 'wan2' has link connectivity
Tue Sep  4 02:19:56 2018 daemon.notice netifd: Interface 'wan2' is setting up now
Tue Sep  4 02:19:56 2018 daemon.notice netifd: wan2 (8699): udhcpc: started, v1.28.3
Tue Sep  4 02:19:56 2018 daemon.notice netifd: wan2 (8699): udhcpc: sending discover
Tue Sep  4 02:19:59 2018 daemon.notice netifd: wan2 (8699): udhcpc: sending discover
Tue Sep  4 02:20:02 2018 daemon.notice netifd: wan2 (8699): udhcpc: sending discover
Tue Sep  4 02:22:59 2018 kern.info kernel: [  493.025926] NETDEV WATCHDOG: eth2 (cdc_ether): transmit queue 0 timed out
Tue Sep  4 02:22:59 2018 kern.warn kernel: [  493.037298] ------------[ cut here ]------------
Tue Sep  4 02:22:59 2018 kern.warn kernel: [  493.046355] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:320 dev_watchdog+0x16c/0x258
[... cut ...]
Tue Sep  4 02:22:59 2018 kern.warn kernel: [  494.056021] [<ffffff80080c65d0>] cpu_startup_entry+0x20/0x28
Tue Sep  4 02:22:59 2018 kern.warn kernel: [  494.065133] [<ffffff800808cbcc>] secondary_start_kernel+0x17c/0x188
Tue Sep  4 02:22:59 2018 kern.warn kernel: [  494.074871] ---[ end trace dcf7fcf89d265702 ]---
Tue Sep  4 02:22:59 2018 kern.warn kernel: [  494.083781] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
Tue Sep  4 02:22:59 2018 kern.warn kernel: [  494.083781]
```

reedog117, 2018-09-04:
Have you tried two different Ethernet adapters? This almost seems like a driver issue, as I'm using two USB WiFi adapters (same make/model) and 2 USB 4G LTE hotspots (same make/model) to connect 4 different LTE connections with no issues, also on a RPi3B+.

afriza, 2018-09-04:
Hi @reedog117, I have just tried combining 1 EDIMAX USB 2.0 Fast Ethernet Adapter (asix) and 1 Huawei E3372-607 4G Modem in HiLink mode (cdc_ether), the result is the same.

```
[   14.041038] usb 1-1.1.3: new high-speed USB device number 6 using dwc_otg
[   14.181952] usb 1-1.1.3: New USB device found, idVendor=12d1, idProduct=14dc
[   14.192605] usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   14.203682] usb 1-1.1.3: Product: HUAWEI_MOBILE
[   14.211754] usb 1-1.1.3: Manufacturer: HUAWEI_MOBILE
[   14.219189] usb 1-1.1.3: no of_node; not parsing pinctrl DT
[   14.332247] cdc_ether 1-1.1.3:1.0: no of_node; not parsing pinctrl DT
[   14.377824] cdc_ether 1-1.1.3:1.0 eth1: register 'cdc_ether' at usb-3f980000.usb-1.1.3, CDC Ethernet Device, 0c:5b:8f:27:9a:64
[   14.400711] uas 1-1.1.3:1.2: no of_node; not parsing pinctrl DT
[   14.400774] usb-storage 1-1.1.3:1.2: no of_node; not parsing pinctrl DT
[   14.400798] usb-storage 1-1.1.3:1.2: USB Mass Storage device detected
[   14.409592] scsi host0: usb-storage 1-1.1.3:1.2
[   15.439880] scsi 0:0:0:0: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
[   15.450418] sd 0:0:0:0: no of_node; not parsing pinctrl DT
[   15.455182] sd 0:0:0:0: [sda] Attached SCSI removable disk
[   16.110292] 8021q: adding VLAN 0 to HW filter on device eth0
[   16.127024] cdc_ether 1-1.1.3:1.0 eth1: kevent 12 may have been dropped
[   16.136100] cdc_ether 1-1.1.3:1.0 eth1: kevent 11 may have been dropped
[  168.059026] usb 1-1.3: new high-speed USB device number 7 using dwc_otg
[  168.211972] usb 1-1.3: New USB device found, idVendor=0b95, idProduct=772b
[  168.223001] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  168.234506] usb 1-1.3: Product: AX88772B
[  168.242551] usb 1-1.3: Manufacturer: ASIX Elec. Corp.
[  168.251805] usb 1-1.3: SerialNumber: 15425E
[  168.260807] usb 1-1.3: no of_node; not parsing pinctrl DT
[  168.262306] asix 1-1.3:1.0: no of_node; not parsing pinctrl DT
[  168.714479] asix 1-1.3:1.0 eth2: register 'asix' at usb-3f980000.usb-1.3, ASIX AX88772B USB 2.0 Ethernet, 00:50:b6:15:42:5e
[  168.732794] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[  170.273029] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[  170.287510] asix 1-1.3:1.0 eth2: link up, 100Mbps, full-duplex, lpa 0x41E1
[  170.309619] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[  170.309619] 
[  170.519689] raspberrypi-firmware soc:firmware: Get Throttled mailbox call failed
```

reedog117, 2018-09-04:
I'm using ZTE Z915 hotspots - I just ensure they're powered on first and connected via USB before booting the RPi3.

rstanislav, 2018-09-04:
I have also same problem, patch that changes cpu freq to 1400000 is a cause i think , also with that patch on boot rpi says that it can't change frequency to 1400000 sometimes, only way to avoid it is to set force_turbo=1 in config.txt, but that only suppressed errors about freq change. 
with setting to defaults when connecting second usb it also sometimes spams:

> [  126.456519] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 600000 (-12)
> [  126.466987] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (600000)
> [  126.479651] cpufreq: __target_index: Failed to change cpu frequency: -22

also this error:

> ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case

After this errors eventually rpi usb subsystem crashes with

> Tue Sep  4 21:06:36 2018 kern.info kernel: [ 1083.993803] NETDEV WATCHDOG: eth2 (cdc_ether): transmit queue 0 timed out
> Tue Sep  4 21:06:36 2018 kern.warn kernel: [ 1084.001571] ------------[ cut here ]------------
> Tue Sep  4 21:06:36 2018 kern.warn kernel: [ 1084.007064] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:320 dev_watchdog+0x16c/0x258

etc...

and after that usb subsystem is completely stopped - no usb, no ethernet (its on usb on rpi) - but rpi itself works(i have connected it via uart connection for console)

I have tried to compile image with custom kernel with cpu governor completely disabled with force_turbo=1 and as result rpi usb is not crashing - its only sometimes spams

> ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case

and unable get dhcp address from second modem(for these bug tests i'm using 2 modems)

This is not power problem, i'm currently running my rpi 3b+ with 5A 5.2v power supply(testing with 4 LTE modems, increased voltage to avoid drop in wires to modems, at rpi terminals multimeter reads about 5.15v average(i have also shorted main power with usb pins on rpi to avoid rpi 700ma limitation on usb ports, so modems getting as much power as they need).

and.... at the end i have compiled custom 32bit image ... and all problems are gone, looks like 64 bit kernel/system for rpi is too buggy/unstable at the moment

Ysurac, 2018-09-05:
At least it's not a kernel 4.14 or OpenMPTCProuter specific bug.
And no problem with Raspbian Stretch 9.4 Desktop ? Can you give log on Raspbian when you plug the keys ? And give me the uname -a too.

Ysurac, 2018-09-05:
ok. I'm using raspberry kernel patch, so should be the same here. Maybe a kernel module that make problem... 
Can you give a lsmod on raspbian ?

afriza, 2018-09-05:
```
pi@raspberrypi:~$ lsmod
Module                  Size  Used by
sg                     28672  0
cdc_ether              16384  0
asix                   36864  0
fuse                  106496  3
rfcomm                 49152  4
bnep                   20480  2
hci_uart               36864  1
btbcm                  16384  1 hci_uart
serdev                 20480  1 hci_uart
bluetooth             368640  29 hci_uart,bnep,btbcm,rfcomm
ecdh_generic           28672  1 bluetooth
brcmfmac              307200  0
brcmutil               16384  1 brcmfmac
cfg80211              573440  1 brcmfmac
rfkill                 28672  6 bluetooth,cfg80211
snd_bcm2835            32768  1
snd_pcm                98304  1 snd_bcm2835
snd_timer              32768  1 snd_pcm
snd                    69632  5 snd_timer,snd_bcm2835,snd_pcm
uio_pdrv_genirq        16384  0
fixed                  16384  0
uio                    20480  1 uio_pdrv_genirq
i2c_dev                16384  0
ip_tables              24576  0
x_tables               32768  1 ip_tables
ipv6                  434176  26
```

reedog117, 2018-09-05:
I know I had to setup the cpu min and max scaling to the same number and change the CPU governor to performance in the advanced settings to prevent crashing. At first 2 USB Wifi and 1 USB hotspot was okay, but the 2nd hotspot would make things completely unstable. Changing these settings fixed this. I may try reducing the CPU speed to the min instead of using the max for power savings.

mykelinho, 2018-09-08:
I have test :
Lan = eth0 
WAN1 = eth1 (usb 3.0 adaptater to lan gigabit)
WAN2 = eth2 (usb 3.0 adaptater to lan gigabit)
Ok for me it's work

Ysurac, 2018-09-08:
At least it work for somebody :)
What is the manufacturer of your adapters ?
And why do you use adapters and not a switch (that is really better for a RPI that have really bad USB) ?

afriza, 2018-09-10:
@mykelinho can you share the [kernel log](http://192.168.100.1/cgi-bin/luci/admin/status/dmesg) when plugging in the USB 3.0 adapters?

afriza, 2018-09-10:
I don't know if I missed this before or if it was previously not working... But now on 32-bit image, I can connect 4 ethernet adaptors (2x asix and 2x cdc_ether) and all of them are getting IPs as DHCP clients..

Although I find the following in the kernel log (similar to [#131](https://github.com/Ysurac/openmptcprouter/issues/131#issuecomment-412972049)) 

```
[ 6449.184339] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
[ 6455.183129] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 6455.199493] brcmfmac: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
```

mykelinho, 2018-09-13:
log on plug USB  for me

Thu Sep 13 20:09:57 2018 kern.debug kernel: [    0.621354] pinctrl-bcm2835 3f200000.gpio: read 00200924 (43 => alt0)
Thu Sep 13 20:09:57 2018 kern.info kernel: [    0.621407] uart-pl011 3f201000.serial: cts_event_workaround enabled
Thu Sep 13 20:09:57 2018 kern.info kernel: [    0.622228] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 72, base_baud = 0) is a PL011 rev2
Thu Sep 13 20:09:57 2018 kern.debug kernel: [    0.624279] pinctrl core: add 0 pinctrl maps
Thu Sep 13 20:09:57 2018 kern.info kernel: [    0.624478] console [ttyS0] disabled
Thu Sep 13 20:09:57 2018 kern.info kernel: [    0.625241] 3f215040.serial: ttyS0 at MMIO 0x0 (irq = 151, base_baud = 31250000) is a 16550
Thu Sep 13 20:09:57 2018 kern.warn kernel: [    0.821084] Indeed it is in host mode hprt0 = 00021501
Thu Sep 13 20:09:57 2018 kern.notice kernel: [    1.016750] random: fast init done
Thu Sep 13 20:09:57 2018 kern.info kernel: [    1.031041] usb 1-1: new high-speed USB device number 2 using dwc_otg
Thu Sep 13 20:09:57 2018 kern.warn kernel: [    1.034023] Indeed it is in host mode hprt0 = 00001101
Thu Sep 13 20:09:57 2018 kern.info kernel: [    1.292136] usb 1-1: New USB device found, idVendor=0424, idProduct=2514
Thu Sep 13 20:09:57 2018 kern.info kernel: [    1.292140] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Thu Sep 13 20:09:57 2018 kern.debug kernel: [    1.297830] hub 1-1:1.0: no of_node; not parsing pinctrl DT
Thu Sep 13 20:09:57 2018 kern.info kernel: [    1.297845] hub 1-1:1.0: USB hub found
Thu Sep 13 20:09:57 2018 kern.info kernel: [    1.297897] hub 1-1:1.0: 4 ports detected

Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.161066] usb 1-1.1.2: new high-speed USB device number 4 using dwc_otg
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.291715] usb 1-1.1.2: New USB device found, idVendor=0bda, idProduct=8153
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.299674] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=6
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.308768] usb 1-1.1.2: Product: USB 10/100/1000 LAN
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.314654] usb 1-1.1.2: Manufacturer: Realtek
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.319920] usb 1-1.1.2: SerialNumber: 000001
Thu Sep 13 20:09:57 2018 kern.debug kernel: [    2.325291] usb 1-1.1.2: no of_node; not parsing pinctrl DT
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.421058] usb 1-1.1.3: new high-speed USB device number 5 using dwc_otg
Thu Sep 13 20:09:57 2018 user.info kernel: [    2.517982] kmodloader: loading kernel modules from /etc/modules-boot.d/*
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.535289] pps_core: LinuxPPS API ver. 1 registered
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.541120] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.552568] usb 1-1.1.3: New USB device found, idVendor=0bda, idProduct=8153
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.560533] usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=6
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.561140] PTP clock support registered
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.574466] usb 1-1.1.3: Product: USB 10/100/1000 LAN
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.580336] usb 1-1.1.3: Manufacturer: Realtek
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.585623] usb 1-1.1.3: SerialNumber: 000001
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.586573] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
Thu Sep 13 20:09:57 2018 kern.debug kernel: [    2.598564] usb 1-1.1.3: no of_node; not parsing pinctrl DT
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.598759] ehci-platform: EHCI generic platform driver
Thu Sep 13 20:09:57 2018 user.info kernel: [    2.612510] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Thu Sep 13 20:09:57 2018 user.info kernel: [    2.630872] init: - preinit -
Thu Sep 13 20:09:57 2018 kern.info kernel: [    2.901049] usb 1-1.1.1: new high-speed USB device number 6 using dwc_otg
Thu Sep 13 20:09:57 2018 kern.info kernel: [    3.031280] usb 1-1.1.1: New USB device found, idVendor=0424, idProduct=7800
Thu Sep 13 20:09:57 2018 kern.info kernel: [    3.039266] usb 1-1.1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Thu Sep 13 20:09:57 2018 kern.debug kernel: [    3.048847] lan78xx 1-1.1.1:1.0: no of_node; not parsing pinctrl DT
Thu Sep 13 20:09:57 2018 kern.notice kernel: [    3.052705] random: procd: uninitialized urandom read (4 bytes read)
Thu Sep 13 20:09:57 2018 kern.warn kernel: [    3.116398] NOHZ: local_softirq_pending 01
Thu Sep 13 20:09:57 2018 kern.warn kernel: [    3.121366] NOHZ: local_softirq_pending 01
Thu Sep 13 20:09:57 2018 kern.warn kernel: [    3.126283] NOHZ: local_softirq_pending 01
Thu Sep 13 20:09:57 2018 kern.warn kernel: [    3.131149] NOHZ: local_softirq_pending 01
Thu Sep 13 20:09:57 2018 kern.warn kernel: [    3.136000] NOHZ: local_softirq_pending 01
Thu Sep 13 20:09:57 2018 kern.warn kernel: [    3.140802] NOHZ: local_softirq_pending 01
Thu Sep 13 20:09:57 2018 kern.warn kernel: [    3.145578] NOHZ: local_softirq_pending 01
Thu Sep 13 20:09:57 2018 kern.warn kernel: [    3.150337] NOHZ: local_softirq_pending 01
Thu Sep 13 20:09:57 2018 kern.warn kernel: [    3.155080] NOHZ: local_softirq_pending 01
[... cut ...]
Thu Sep 13 20:09:57 2018 kern.debug kernel: [    7.797283] pinctrl-bcm2835 3f200000.gpio: request pin 41 (gpio41) for soc:audio
Thu Sep 13 20:09:57 2018 kern.debug kernel: [    7.797287] pinctrl-bcm2835 3f200000.gpio: read 24200924 (41 => alt0)
Thu Sep 13 20:09:57 2018 kern.info kernel: [    7.798327] bcm2835_alsa bcm2835_alsa: card created with 8 channels
Thu Sep 13 20:09:57 2018 kern.info kernel: [    7.815616] u32 classifier
Thu Sep 13 20:09:57 2018 kern.info kernel: [    7.819058]     input device check on

c1ngular, 2018-12-16:
@afriza did you get it working ? I have a raspberry pi 3b , before I buy USB 4g modems, I have several questions, could you help me out ? 
1. do you think I can use 4 USB 4g modems at same time ? what outcome did you get by these settings? which brand/model of modem are you using ?

2. how about Ethernet USB adapter ?  4 at same time possible ? which brand/model works for you? 

3. do I need external power supply if I want to use 4 USB ports same time ?  

4. how do I set it up , I have never used USB 4g modem/adapter before, according to OMR documentation, I have to set up gateways, turn off DHCP on these devices, how ? or it just works after plugging it in ?

5. what is the risk if I am using these USB ports instead of a switch ? I understand that using USB ports could effect Ethernet interface bandwidth, but my option is pretty limited since only mobile networks are available here，and I need my solution to be light and as portable as possible . 

thank you !

Ysurac, 2018-12-16:
I would say that it's always easier to buy ethernet modem :)

c1ngular, 2018-12-16:
@Ysurac,  so , raspberry Pi + 4G modems with Ethernet port connected to a switch would be my best option ? 

how about Banana Pi R2, is it fully functional/stable? if it is , I don't need a switch, am I correct ?

thank you, and wonderful project .

Ysurac, 2018-12-16:
For now I think BPI-r2 is not stable enough

afriza, 2018-12-19:
@c1ngular with 32 bit version OMR 4 USB modem are kind of working.. the use case is that we want to have mobile setup whereby we need to move around with Raspberry Pi helping to aggregate internet connection..

closed 2019-02-20 by Ysurac