Ysurac/openmptcprouter #143: VMWare image failed booting on esxi

TRSATOC, 2018-08-29:
When booting on esxi I have the following message:
```
SQUASHFS error: xz decompression failed, data probably corrupt
SQUASHFS error: squashfs_read_data failed to read block 0x6e

```
I've tried differents vmware image versions on the virtual machine configuration, conversions from vdi to vmdk, etc... always the same problem.
it will be nice to have a step by step process.

Ysurac, 2018-08-29:
You can try to follow #87 but I can't provide support on VMWare, I only use Virtualbox.

kamcio2603, 2018-08-29:
Vmware image works for me. Vmdk + IDE setup. Esxi 6.5 here.

TRSATOC, 2018-08-29:
OK, I've find the problem.
I've followed the #87 but you need also to remove the floppy drive \o/ ... That was not mentioned

TRSATOC, 2018-08-29:
@kamcio2603 I can't save anything on my VM, the change of the root password is not saved when I reboot.
And I always have some  SQUASHFS message like my first message on the thread:
[image]

TRSATOC, 2018-08-29:
I'm using this controller others don't boot, sorry it's in french ;)
[image]

Scaff31, 2018-09-04:
J'ai une nouvelle procédure qui fonctionne mieux est adaptable a toutes les versions d'esxi (6.x)

1- Recupere l'image workstation
2- Upload le vmdk sur ton esxi
3- Connecte toi en SSH sur l'esxi puis converti le disque de workstation à ESXI
vmkfstools -i [workstation.vmdk] -d zeroedthick [esxi.vmdk]

4- Crée ta VM avec le disque que tu viens de convertir et tu n'aura plus de souci de squashfs

TRSATOC, 2018-09-04:
Thank you so much!
That ok now!!!

closed 2018-09-04 by TRSATOC