Ysurac/openmptcprouter #145: Erreur connexion webpage 192.168.100.1

mykelinho, 2018-09-02:
Bonjour, je n'arrive pas à me connecter à l'interface web de openmptcprouter installer sur mon rpi3.
J'ai reussi une premiere connexion et suite à des mauvais reglages j'ai perdu la connexion.
J'ai donc reformatter ma carte SD pour réinstaller openmptcprouter mais je n'arrive toujours pas a ma reconnecter à l'interface web (adresse ip du pc en 192.168.100.11 et gateway en 192.168.100.1)
Je ne trouve pas de solution...

afriza, 2018-09-02:
Try to flash the sdcard with some other image, for example Raspbian Lite, then flashing with OpenMPTCProuter again. If you are like me, subsequent flash of OpenMPTCProuter doesn't clear out the settings and thus you need to flash the sdcard with some other image before flashing with OpenMPTCProuter image again.

Maudissure, 2018-09-02:
Essayes d'effectuer un formatage en mode complet avant de réinstaller l'image. C'est le seul moyen que j'ai trouvé pour initialiser correctement la carte SD après une configuration malheureuse.

mykelinho, 2018-09-02:
Thank you afriza it's OK i  have change image to raspbian lite and  flash new with openmptcptrouter.
Merci Maudissure, également.
J'avais tenté le formatage avant de remettre l'image sur la SD mais c'était pareil... en formatant puis en mettant l'image raspbian lite puis en reformatant avant de mettre l'image openmptcptrouter ça à fonctionné.
Maintenant j'ai d'autres problèmes mais liés au VPS OVH qui a bien mptcp d'implémenter mais je n'ai aucune clef généré...
Linux vps521722 4.9.60.mptcp #5 SMP Thu Nov 2 22:20:19 UTC 2017 x86_64 GNU/Linux

closed 2018-09-03 by Ysurac

afriza, 2018-09-04:
@Ysurac is it the expected behaviour that flashing the image doesn't reset the data?