Ysurac/openmptcprouter #147: Wireless on Raspberry 3B+

scith, 2018-09-03:
Hello and thanks for this great alternative to over-the-box!
Using ethernet works fine on my RPI3B+, but I cannot make wireless work

## Expected Behavior

- Working access points for other clients to connect to the RPI3B+

## Actual Behavior
- Wireless network radio0.network1 shows "Wireless is not associated"
- Wireless network radio0.network1 has a signal of 0% / 0 dB

## Steps to Reproduce the Problem

  1. Flash v0.34.4 on RPI3B+
  2. Setup OpenMPTCP and verify its functioning
  3. Add a new wireless access point under radio0
  4. Enable the access point

I tested this with:
- Access point wireless N 2.4Ghz
- Access point wireless AC 5Ghz
- Same but with an Edimax wireless usb key

## Specifications

  - OpenMPTCProuter version: 0.34.4 32bits
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: RPI3b+

Is there perhaps a specific setup needed for RPI3B+ AP to work?
Thanks in advance

Ysurac, 2018-09-03:
Seems to be working in Legacy mode 2.4Ghz. I will make more test to find what is working or not.

reedog117, 2018-09-04:
I had this problem but found that when running ```uci show wireless``` for some reason on my Pi the AP was set to radio2 (I have 2 USB WiFi adapters also installed). Once I figured that out, I set AP mode on radio2 and that worked correctly when radio0 wasn't working with AP mode.

mykelinho, 2018-09-06:
I have the bug on Flash v0.34.4 with RPI3B+

closed 2019-02-20 by Ysurac