Ysurac/openmptcprouter #148: Problems MPTCP and GLORYTUN

mykelinho, 2018-09-03:
Bonjour,
J'ai des problème pour configurer MPTCP et GLORYTUN sur mon vps OVH, j'ai poursuivi l'installation comme indiqué sur les différents tuto mais je ne vois pas comment récupérer ou paramétré ces softs.
Des idées. Merci.

[image]

État
Système
Nom d'hôte	OpenMPTCProuter
Model	Raspberry Pi 3 Model B Rev 1.2
Architecture	?
Version du micrologiciel	openmptcprouter v0.33.3 r7273-14580aaf81 / LuCI Master (git-18.239.22082-5a1f6d0)
Version du noyau	4.14.66

Ysurac, 2018-09-03:
Tout est dans la doc: https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS

mykelinho, 2018-09-03:
Merci, j'ai bien suivi cette doc à la lettre (j'ai même reinstallé à  zéro le VPS en debian 9).
Je n'arrive pas à obtenir ceci : After install, keys are saved in /root/openmptcprouter_config.txt

après avoir lancé l'installation de  MPTCP kernel and shadowsocks-libev,
lorsque je lance la commande wget -O - http://www.openmptcprouter.com/server/debian9-x86_64.sh | sh

j'ai le résultat suivant ;

root@vps521722:/tmp# sudo sh debian9-x86_64.sh
Atteint:1 http://security.debian.org stretch/updates InRelease
Ign:2 http://deb.debian.org/debian stretch InRelease
Atteint:3 http://deb.debian.org/debian stretch-updates InRelease
Atteint:4 http://deb.debian.org/debian stretch-backports InRelease
Atteint:5 http://deb.debian.org/debian stretch Release
Ign:6 https://dl.bintray.com/cpaasch/deb stretch InRelease
Atteint:8 https://dl.bintray.com/cpaasch/deb stretch Release
Lecture des listes de paquets... Fait
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances
Lecture des informations d'état... Fait
dirmngr is already the newest version (2.1.18-8~deb9u2).
patch is already the newest version (2.7.5-1+deb9u1).
Les paquets suivants ont été installés automatiquement et ne sont plus nécessaires :
  libsodium18 libudns0 linux-image-4.9.0-4-amd64 linux-image-4.9.0-7-amd64
Veuillez utiliser « sudo apt autoremove » pour les supprimer.
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.
Atteint:1 http://security.debian.org stretch/updates InRelease
Ign:2 http://deb.debian.org/debian stretch InRelease
Atteint:3 http://deb.debian.org/debian stretch-updates InRelease
Atteint:4 http://deb.debian.org/debian stretch-backports InRelease
Atteint:5 http://deb.debian.org/debian stretch Release
Ign:6 https://dl.bintray.com/cpaasch/deb stretch InRelease
Atteint:7 https://dl.bintray.com/cpaasch/deb stretch Release
Lecture des listes de paquets... Fait
--2018-09-03 22:27:02--  https://www.openmptcprouter.com/kernel/linux-image-4.14.64-mptcp-16e8846.amd64.deb
Résolution de www.openmptcprouter.com (www.openmptcprouter.com)… 94.23.252.192
Connexion à www.openmptcprouter.com (www.openmptcprouter.com)|94.23.252.192|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 37809302 (36M) [application/octet-stream]
Sauvegarde en : « /tmp/linux-image-4.14.64-mptcp-16e8846.amd64.deb »

/tmp/linux-image-4.14.64-mptcp-16e8846. 100%[============================================================================>]  36,06M  7,47MB/s    in 4,8s

2018-09-03 22:27:07 (7,50 MB/s) — « /tmp/linux-image-4.14.64-mptcp-16e8846.amd64.deb » sauvegardé [37809302/37809302]

--2018-09-03 22:27:07--  https://www.openmptcprouter.com/kernel/linux-headers-4.14.64-mptcp-16e8846.amd64.deb
Résolution de www.openmptcprouter.com (www.openmptcprouter.com)… 94.23.252.192
Connexion à www.openmptcprouter.com (www.openmptcprouter.com)|94.23.252.192|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 9082430 (8,7M) [application/octet-stream]
Sauvegarde en : « /tmp/linux-headers-4.14.64-mptcp-16e8846.amd64.deb »

/tmp/linux-headers-4.14.64-mptcp-16e884 100%[============================================================================>]   8,66M  7,72MB/s    in 1,1s

2018-09-03 22:27:08 (7,72 MB/s) — « /tmp/linux-headers-4.14.64-mptcp-16e8846.amd64.deb » sauvegardé [9082430/9082430]

Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances
Lecture des informations d'état... Fait
rename is already the newest version (0.20-4).
Les paquets suivants ont été installés automatiquement et ne sont plus nécessaires :
  libsodium18 libudns0 linux-image-4.9.0-4-amd64 linux-image-4.9.0-7-amd64
Veuillez utiliser « sudo apt autoremove » pour les supprimer.
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.
dpkg: version 4.14.64-mptcp-16e8846 de linux-image-4.14.64-mptcp-16e8846 déjà installée, paquet ignoré.
dpkg: version 4.14.64-mptcp-16e8846 de linux-headers-4.14.64-mptcp-16e8846 déjà installée, paquet ignoré.
Set MPTCP kernel as grub default...
--2018-09-03 22:27:09--  https://www.openmptcprouter.com/server/update-grub.sh
Résolution de www.openmptcprouter.com (www.openmptcprouter.com)… 94.23.252.192
Connexion à www.openmptcprouter.com (www.openmptcprouter.com)|94.23.252.192|:443… connecté.
[... 799 lines cut ...]
Construction de l'arbre des dépendances
Lecture des informations d'état... Fait
Correction des dépendances...Starting pkgProblemResolver with broken count: 1
Starting 2 pkgProblemResolver with broken count: 1
Investigating (0) shadowsocks-libev-build-deps:amd64 < 3.2.0-1 @iU mK Nb Ib >
Broken shadowsocks-libev-build-deps:amd64 Dépend on libc-ares-dev:amd64 < none | 1.12.0-1+deb9u1 @un uH >
  Considering libc-ares-dev:amd64 1 as a solution to shadowsocks-libev-build-deps:amd64 -2
  Removing shadowsocks-libev-build-deps:amd64 rather than change libc-ares-dev:amd64
Done
 Fait
Starting pkgProblemResolver with broken count: 0
Starting 2 pkgProblemResolver with broken count: 0
Done
Les paquets suivants ont été installés automatiquement et ne sont plus nécessaires :
  libsodium18 libudns0 linux-image-4.9.0-4-amd64 linux-image-4.9.0-7-amd64
Veuillez utiliser « apt autoremove » pour les supprimer.
Les paquets suivants seront ENLEVÉS :
  shadowsocks-libev-build-deps
0 mis à jour, 0 nouvellement installés, 1 à enlever et 0 non mis à jour.
1 partiellement installés ou enlevés.
Après cette opération, 9 216 o d'espace disque seront libérés.
(Lecture de la base de données... 89940 fichiers et répertoires déjà installés.)
Suppression de shadowsocks-libev-build-deps (3.2.0-1) ...
mk-build-deps: Unable to install shadowsocks-libev-build-deps at /usr/bin/mk-build-deps line 402.
mk-build-deps: Unable to install all build-dep packages

Ysurac, 2018-09-03:
Donc l'installation ne fonctionne pas. Il semble qu'il manque la dépence ```libc-ares-dev```.
Le script a été mis à jour en ajoutant cette dépendance.

mykelinho, 2018-09-04:
Merci.
Faut-il ajouter un dépot en particulier ?
En relançant le script j'obtient ->

Certains paquets ne peuvent être installés. Ceci peut signifier
que vous avez demandé l'impossible, ou bien, si vous utilisez
la distribution unstable, que certains paquets n'ont pas encore
été créés ou ne sont pas sortis d'Incoming.
L'information suivante devrait vous aider à résoudre la situation :

Les paquets suivants contiennent des dépendances non satisfaites :
 libc-ares-dev : Dépend: libc-ares2 (= 1.12.0-1+deb9u1) mais 1.14.0-1~bpo9+1 devra être installé
E: Impossible de corriger les problèmes, des paquets défectueux sont en mode « garder en l'état ».

Ysurac, 2018-09-04:
Je viens de lancer une installation avec la toute dernière version du script sur un VPS OVH sous debian 9 et aucun soucis.
Si elle ne fonctionne toujours pas, as-tu fais autre chose que lancé le script après l'installation par OVH ?

mykelinho, 2018-09-04:
Merci beaucoup le script à fonctionner !
effectivement j'avais fait des installations avant de lancer le script (j'avais mal lu/compris le tuto).
suite à ce message : You need to have a MPTCP kernel and shadowsocks-libev installed on the VPS.
j'avais réinstaller manuellement mptcp et shadowsocks-libev (en suivant les tutos issues des liens hypertextes).
Je viens de refaire une installation du vps puis lancer uniquement le script qui installe bien l'ensemble des composants nécessaire et j'ai abouti sur ce message ->

====================================================================================                                     
OpenMPTCProuter VPS is now configured !                                                                                  
SSH port: 65222 (instead of port 22)                                                                                     
Shadowsocks port: 65101                                                                                                  
Shadowsocks encryption: chacha20                                                                                         
Your shadowsocks key:                                                                                                    
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx                                                                             
Glorytun port: 65001                                                                                                     
Glorytun encryption: chacha20                                                                                            
Your glorytun key:                                                                                                       
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
====================================================================================                                     
Keys are also saved in /root/openmptcprouter_config.txt, you are free to remove them                                     
====================================================================================                                     
  /!\ You need to reboot to enable MPTCP, shadowsocks, glorytun and shorewall /!\                                        
------------------------------------------------------------------------------------                                     
 After reboot, check with uname -a that the kernel name contain mptcp.                                                   
 Else, you may have to modify GRUB_DEFAULT in /etc/defaut/grub                                                           
====================================================================================      

je testerais des que je suis à mon domicile sur la partie fonctionnelle avev le routeur raspberry ce soir

kamcio2603, 2018-09-04:
https://github.com/Ysurac/openmptcprouter/blob/a5198f3bfa1a4b1b3cb056e9109030216384c465/.github/ISSUE_TEMPLATE.md#expected-behavior 

***In English please.***

closed 2018-09-04 by Ysurac

mykelinho, 2018-09-04:
it's ok ! ça fonctionne.
thank you