Ysurac/openmptcprouter #152: Problem on agregation speed 4G+ADSL - débit trop lent très inférieur à la 4G

mykelinho, 2018-09-04:
Expected Behavior
the output kbps/s of the openmptcprouter router is not the sum of the 4G and ADSL speeds

_le débit en sortie du routeur openmptcprouter est très loins de la somme des débits 4G et ADSL en entrée_

Steps to Reproduce the Problem
Test on http://www.speedtest.net/
LAN (Wifi)
WAN1 (eth0)
WAN2 (eth1)

[image]

TEST 4G only (69 mega)
[image]

TEST ADSL ONLY (7 mega)
[image]

TEST AGREGATION by openmptcprouter (wifi rpi-3) (19 mega)
[image]

Specifications
OpenMPTCProuter version: openmptcprouter v0.33.3 r7273-14580aaf81 
OpenMPTCProuter VPS version: OVH Debian 9
OpenMPTCProuter platform: RPI-3

Ysurac, 2018-09-04:
You have a high latency on 4G connection.
Do you have ShadowSocks enabled or do you use only a VPN ?

To test speed you should try ```omr-test-speed``` on OpenMPTCProuter.

mykelinho, 2018-09-04:
Yes the latency is on a moment high but now ok (50ms).
ShadowsSocks/glorytun enabled.

result with command omr-test-speed 
[image]

Ysurac, 2018-09-04:
Seems better here. 7,761Mo*8=62Mb/s
You should be able to get a better speed, but 4G speed can be really unstable...

mykelinho, 2018-09-04:
ok, i buy new convert USB to RJ45 for the RPi3, for test on LAN this weekend
what solution? change value MTU?

mykelinho, 2018-09-04:
solution for better speed?

why it's not good idea USB to RJ45 on RPI ?

on the router RPI i have ->
 WAN1 (ADSL) = ethernet (eth0) 
 WAN 2 (4G) = ethernet by USB to RJ45 (eth0)
LAN = wifi on RPI 

what is the best solution in terms of hardware to have 3 ethernet ports on the RPI?

Ysurac, 2018-09-04:
Try to test speed at another time of the day, 4G speed is unstable, this can change a lot. 62Mb/s at real is not bad. You can also use ```omr-test-speed wan1``` and omr-test-speed wan2``` to test both interface without MPTCP.

RPI ethernet use already USB, so it's a lot of data for USB.
Use a switch, many modem already have one. OpenMPTCProuter will create virtual interfaces and can use eth0 for all.

Maudissure, 2018-09-06:
"what is the best solution in terms of hardware to have 3 ethernet ports on the RPI?"

Using a switch ?

Peppe2201, 2018-09-06:
I have similar problem on Raspberry Pi 3 and two 30/10 line, i can get only 17/20 on speedtest, is the RPI the problem?

Maudissure, 2018-09-06:
Test with a file download, not speedtest.

Ysurac, 2018-09-06:
Can i get the result of ```omr-test-speed``` and ```omr-test-speed wan1``` and```omr-test-speed wan2``` ?
Using USB ethernet adapter is a bad idea, using a switch is always far better.

Peppe2201, 2018-09-06:
Bonding speed:
[image]
WAN 1:
[image]
WAN 2:
[image]

I'm using one gigabit adapter and one 100mbit adapter, I think that is the raspberry the problem and for that I'm deploying a VM on VirtualBox, can you suggest me some good box for 4/8 WAN?

Ysurac, 2018-09-06:
Aggregation seems quite good here. 549+221 < 1816ko/s ;)
Why do you want to use adapters and not only a switch ?

Ysurac, 2018-09-06:
Follow this: https://github.com/Ysurac/openmptcprouter/wiki/OpenMPTCProuter-configuration
So static IP for both modems in different networks, modems and RPI connected to same switch, then complete OpenMPTCProuter wizard.

mykelinho, 2018-09-06:
ok thank you. the speed connection it's ok for me on full connexion eth0 with the switch (wan 1/wan 2/lan).
On wifi is very bad (test wifi N rpi3 and wifi N lan vonets -> 10 mega max)

Ysurac, 2018-09-07:
I use a Huawei router, but you need to get a router that is working in your country. If you don't need external antenna, any should work.

For RPI3 wifi, I will make some test but I will put all info in https://github.com/Ysurac/openmptcprouter/issues/147 issue.

closed 2018-09-07 by Ysurac