Ysurac/openmptcprouter #153: MAC addresses spam

scith, 2018-09-05:
## Expected Behavior

- Each interface gets a fixed MAC address

## Actual Behavior

- The interface MAC address changes after each edit/restart

## Steps to Reproduce the Problem

  1. Create an interface and write down the MAC address
  2. Restart the interface and see the MAC address change

## Specifications

  - OpenMPTCProuter version: 0.34.3
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: RPI3B+ 64bits

For instance, this is what my SFR box shows:
2 | ca:66:29:f5:7c:bf | - | 192.168.3.25 | LAN 3
3 | 52:4e:0e:1e:c3:b8 | - | 192.168.3.25 | LAN 3
4 | 7e:43:8a:c4:c9:5c | - | 192.168.3.22 | LAN 3

That's 3 different MAC addresses for OpenMPTCP

I also noticed that after many edits, the SFR box is not providing me an IP anymore. It is like the OpenMPTCP router gets blocked by the SFR box. I need to reinstall OpenMPTCP from scratch to solve this.
Could it be related to this? I don't know if the SFR box bans a client when many MAC addresses claim a same IP. I don't think so, I don't find anything in the SFR box admin interface. Perhaps it's a different issue. At the moment, anyway, I cannot fix the IP of OpenMPTCP on the SFR box because it requires the client to keep the same MAC

Thanks

Ysurac, 2018-09-05:
I will save generated mac address in config in next release.

Ysurac, 2018-09-08:
Fixed in latest release.

closed 2018-09-08 by Ysurac