Ysurac/openmptcprouter #154: Compilation failed with latest mptcp patch

rstanislav, 2018-09-06:
Hunk #1 FAILED at 240.
1 out of 1 hunk FAILED -- saving rejects to file net/mptcp/mptcp_output.c.rej
Patch failed!  Please fix /root/mfix/openmptcprouter/rpi3/source/target/linux/generic/hack-4.14/690-mptcp_v0.94.patch!
Makefile:104: recipe for target '/root/mfix/openmptcprouter/rpi3/source/build_dir/toolchain-aarch64_cortex-a53_gcc-7.3.0_musl/linux-4.14.67/.prepared' failed
make[3]: *** [/root/mfix/openmptcprouter/rpi3/source/build_dir/toolchain-aarch64_cortex-a53_gcc-7.3.0_musl/linux-4.14.67/.prepared] Error 1
make[3]: Leaving directory '/root/mfix/openmptcprouter/rpi3/source/toolchain/kernel-headers'
toolchain/Makefile:96: recipe for target 'toolchain/kernel-headers/compile' failed
make[2]: *** [toolchain/kernel-headers/compile] Error 2
make[2]: *** Waiting for unfinished jobs....

problem: 
cat net/mptcp/mptcp_output.c.rej
--- net/mptcp/mptcp_output.c
+++ net/mptcp/mptcp_output.c
@@ -240,6 +240,9 @@ static void __mptcp_reinject_data(struct sk_buff *orig_skb, struct sock *meta_sk
        if (!after(end_seq, TCP_SKB_CB(skb1)->seq))
            break;

+               if (before(end_seq, TCP_SKB_CB(skb1)->end_seq))
+                       break;
+
        __skb_unlink(skb1, &mpcb->reinject_queue);
        __kfree_skb(skb1);

and also there is always error about:

Checking if UEFI patch is set or not
Done
Remove gtime dependency
1 out of 1 hunk FAILED
Done

Ysurac, 2018-09-06:
I use CircleCI, so I know when there is a problem ;)
It's fixed in latest commit.
gtime message is not a problem.

Why do you compile it ?

closed 2018-09-07 by Ysurac

rstanislav, 2018-09-07:
I'm trying to resolve problem with kevent 4 :( guys at rpi thread posted some code to help debug this problem, i will try to do more tests today