Ysurac/openmptcprouter #156: VPN packet loss on 4G network after IP renew

SoGooDFR, 2018-09-06:
## Actual Behavior
On 4G network, we've got dynamic IP address. When IP change, the VPN is lost (but connected without logs errors) with ~50% packet loss. If i restart the vpn service, the connection is stable again.

For example, when i use SS for udp, when IP renew, we lose 10-15% packet during ~5 sec. After the connection is fine and stable.

## Steps to Reproduce the Problem

  1. Fresh install on both
  2. See the problem.

## Specifications

OpenMPTCProuter version: openmptcprouter v0.34.4 r7298-f7413579c0
OpenMPTCProuter VPS version: OpenMPCTProuter VPS 0.51
OpenMPTCProuter platform: x64 on Intel NUC

Ysurac, 2018-09-06:
What is the VPN used ?

SoGooDFR, 2018-09-06:
Glorytun(TCP/UDP)

Ysurac, 2018-09-06:
Same problem with both ? strange. What do you have in log ?

Ysurac, 2018-09-07:
Nothing is detected by omr-tracker about the 4G connection lost ?

SoGooDFR, 2018-09-07:
No... Zero logs

Ysurac, 2018-09-07:
ShadowSocks doesn't aggregate UDP. But you can enable it in ShadowSocks rules tab.

SoGooDFR, 2018-10-28:
Same problem with Glorytun TCP. Can we detect IP change and reload VPN with the minimal packet loss ?

closed 2019-02-20 by Ysurac