Ysurac/openmptcprouter #157: Error for cancel config in error on openmptcp 0.34.4

mykelinho, 2018-09-06:
Expected Behavior
if error on configuration router openmptcprouter, impossible to cancel
impossible d'annuler les paramètres erronné après décompte des 30 secondes si erreur de config sur le routeur

Steps to Reproduce the Problem
connexion wifi to lan on rpi 3+

[image]

Specifications
OpenMPTCProuter version: openmptcprouter v0.34.4 r7298-f7413579c0 / LuCI Master (git-18.248.76017-8951f5f)
OpenMPTCProuter VPS version: OVH Debian 9
OpenMPTCProuter platform: RPI-3+

mykelinho, 2018-09-06:
on 0.33 no problem, if problem configuration, it's possible to return on old config

Ysurac, 2018-09-07:
No change about that in 0.34.x since 0.33.
Not able to reproduce this issue.

closed 2018-09-07 by Ysurac