Ysurac/openmptcprouter #158: LED ethernet RJ45 don't work on RPI 3+ but LAN is OK

mykelinho, 2018-09-06:
Expected Behavior
led (orange and green) ethernet don't work with rpi 3+ and openmptcp 0.34.4

Steps to Reproduce the Problem
connexion lan on rpi 3+, it's work but led ethernet don't work

Specifications
OpenMPTCProuter version: openmptcprouter v0.34.4 r7298-f7413579c0 / LuCI Master (git-18.248.76017-8951f5f)
OpenMPTCProuter VPS version: OVH Debian 9
OpenMPTCProuter platform: RPI-3+

mykelinho, 2018-09-07:
it's good with raspbian img  but no on openmtcprouter on rpi 3+
with my rpi3, it's ok with openmptcrouter

afriza, 2018-09-08:
It's a problem in the [upstream OpenWRT](https://forum.openwrt.org/t/18-06-on-raspberry-pi-3-b/18670/13?u=afriza) and seems to [have been fixed](https://forum.openwrt.org/t/18-06-on-raspberry-pi-3-b/18670/16?u=afriza).

Ysurac, 2018-09-08:
Thanks @afriza 
This is the pull request https://github.com/openwrt/openwrt/pull/1353
So I will try to backport them to 4.14 (strange that this patch is not in raspberry git...)

Ysurac, 2018-09-17:
Fixed in latest release

closed 2018-09-17 by Ysurac