Ysurac/openmptcprouter #160: Iptv dont' work on OpenMPTCrouter with OMR-byPASS

mykelinho, 2018-09-07:
Expected Behavior
IPTV don't work on openmptcserver

Steps to Reproduce the Problem
connexion IPTV ADSL ok
connexion IPTV 4G ok
connexion IPTV with openmptcpserver HS

Specifications
OpenMPTCProuter version: openmptcprouter v0.34.4 r7298-f7413579c0 / LuCI Master (git-18.248.76017-8951f5f)
OpenMPTCProuter VPS version: OVH Debian 9
OpenMPTCProuter platform: RPI-3+

[image]

Ysurac, 2018-09-08:
You can't provide a port in domain list and OpenMPTCProuter must be used as DNS provider. Your computer can have some DNS cache so this can take some time before working.

mykelinho, 2018-09-08:
ok thank you.
i not understand... doesn't work on tv samsung and pc.
i have flushdns, renew etc... but i have the dhcp openmptcprouter activation and result is 

[image]

[image]

Ysurac, 2018-09-09:
8081 is not a protocole, you have to remove that.
It's using google DNS, so OMR-Bypass can't work here. DNS seems to be fixed somewhere, but I think it's windows and I don't have this OS.

mykelinho, 2018-09-09:
I not understand why i use dns google sorry...
Le routeur attribue les ips par dhcp donc fourni le dns du routeur mais je n'ai touché à aucun réglages dans cette partie... donc censé utilisé les dns openmprcprouter?

mykelinho, 2018-09-09:
[image]

Ysurac, 2018-09-09:
Pas l'adresse IP, le nom de domaine uniquement. Et rien dans protocole. Dans interface soit rien pour utiliser l'une ou l'autre interface, soit l'interface que tu souhaites utiliser.

Le transmissions DNS redirige les requêtes vers unbound qui interroge directement les serveurs DNS root.

mykelinho, 2018-09-09:
Ok. J'ai fais le test pas de changement... dans openmptcprouter il n y a pas de réglages pour l'ip des dns utilisés? Il inclus un serveur dns?
Si je met mes connexions en dhcp il me met automatiquement les dns a 8.8.8.8 on dirait

Ysurac, 2018-09-10:
Oui, il fait serveur DNS et relaye vers les serveurs DNS racines.
Le serveur 8.8.8.8 est inconnu de OpenMPTCProuter, donc ce serveur est mis par autre chose.

PxxBxx, 2018-09-12:
I don't know if this can help, but I had same problems with my Chromecast and Nexus Player.
I had to assign dedicated IPs to these 2 devices (.103 and .104) via the DHCP Static rules, and add a custom firewall rule to BLOCK Google DNSes for these devices.
Then, they fallback on the DNSes sent via the DHCP, the OMR ones.
Here are the custom FW Rules (Firewall > specific rules) :
```
# This file is interpreted as shell script.
# Put your custom iptables rules here, they will
# be executed with each firewall (re-)start.
# Chromecast/NexusTV had hardcoded Google DNSes… sad
iptables -I FORWARD -s 192.168.100.103 --destination 8.8.8.8 -j REJECT
iptables -I FORWARD -s 192.168.100.103 --destination 8.8.4.4 -j REJECT
iptables -I FORWARD -s 192.168.100.104 --destination 8.8.8.8 -j REJECT
iptables -I FORWARD -s 192.168.100.104 --destination 8.8.4.4 -j REJECT
```

Ysurac, 2018-09-12:
True, Chromecast and Nexus Player force the use of Google DNS. I may try to use DPI to change this but not sure this is a good idea... I think iptables rules are a better way.

mykelinho, 2018-09-13:
Thanks, i have test but no ok for me...
[image]

[image]

[image]

mykelinho, 2018-09-13:
exemple de flux HS présent dans un fichier M3U
(ok on connexion ADSL or 4G but no on OpenMPTCProuter)

#EXTINF:-1,----------- BOUQUET SPORT ET BEIN -----------
http://localhd.tv:8081/live/mikael.godinho/tFqisqBTqB/2452.ts
#EXTINF:-1,|FR| BEIN SPORTS 1 FHD
http://localhd.tv:8081/live/monUserID/monMDP/2431.ts
#EXTINF:-1,|FR| BEIN SPORTS 1 HD
http://localhd.tv:8081/live/monUserID/monMDP/85.ts

mykelinho, 2018-09-13:
exemple de flux HS présent dans un fichier M3U
(ok on connexion ADSL or 4G but no on OpenMPTCProuter)

#EXTINF:-1,----------- BOUQUET SPORT ET BEIN -----------
http://localhd.tv:8081/live/monUserID/monMDP/2452.ts
#EXTINF:-1,|FR| BEIN SPORTS 1 FHD
http://localhd.tv:8081/live/monUserID/monMDP/2431.ts
#EXTINF:-1,|FR| BEIN SPORTS 1 HD
http://localhd.tv:8081/live/monUserID/monMDP/85.ts

Ysurac, 2018-09-13:
No IPv6 on "TVSALON" ?

I will make some tests when I will be able to find some working (and not working) public localhd.tv feeds.

Ysurac, 2018-09-14:
Working for me (at least some tv channel) with omr-bypass, you can add the ip of localhd.tv (37.187.25.37), maybe there is some DNS cache...

mykelinho, 2018-09-16:
doesn't work for me. i think the dns for me is always 8.8.8.8

Ysurac, 2018-09-16:
By using the IP instead of the domain name, OMR don't care about DNS.

mykelinho, 2018-09-16:
for me work on vod but no in flux direct (vod 2 video ok and more than 100 videos no ok)

KoRnEr007, 2018-10-15:
I am using premium-ott.com and no issue.
Just add premium-ott.com on OMR-bypass with your interface and i am using the DNS 192.168.100.1
and working well for Live TV and VOD.

Jon

closed 2018-11-26 by Ysurac