Ysurac/openmptcprouter #161: Professional Computers lost Internet connection

Maudissure, 2018-09-09:
## Expected Behavior

Professional computer connected on internet and VPN's society.

## Actual Behavior

Computer get IPV4 address , connected on internet for few seconds, then disconnected.

## Steps to Reproduce the Problem

  1. Connect computer on wired or Wifi 
  2. Get IPV4 and internet connection
  3. After few seconds, lost internet connection

I get the same behavior with an other professional computer with VPN's society connection.

Some system log : 

Sun Sep  9 09:53:40 2018 daemon.info dnsmasq-dhcp[4257]: DHCPREQUEST(eth0) 192.168.1.235 68:f7:28:92:ea:2c
Sun Sep  9 09:53:40 2018 daemon.warn dnsmasq-dhcp[4257]: Ignoring domain siege.axa-fr.intraxa for DHCP host name PO469089
Sun Sep  9 09:53:40 2018 daemon.info dnsmasq-dhcp[4257]: DHCPACK(eth0) 192.168.1.235 68:f7:28:92:ea:2c PO469089
Sun Sep  9 09:53:46 2018 daemon.info dnsmasq-dhcp[4257]: DHCPDISCOVER(eth0) 34:02:86:07:b0:16
Sun Sep  9 09:53:46 2018 daemon.info dnsmasq-dhcp[4257]: DHCPOFFER(eth0) 192.168.1.153 34:02:86:07:b0:16
Sun Sep  9 09:53:46 2018 daemon.info dnsmasq-dhcp[4257]: DHCPINFORM(eth0) 192.168.1.235 68:f7:28:92:ea:2c
Sun Sep  9 09:53:46 2018 daemon.info dnsmasq-dhcp[4257]: DHCPACK(eth0) 192.168.1.235 68:f7:28:92:ea:2c PO469089
Sun Sep  9 09:55:16 2018 daemon.info dnsmasq-dhcp[4257]: DHCPINFORM(eth0) 192.168.1.235 68:f7:28:92:ea:2c
Sun Sep  9 09:55:16 2018 daemon.info dnsmasq-dhcp[4257]: DHCPACK(eth0) 192.168.1.235 68:f7:28:92:ea:2c PO469089

## Specifications

  - OpenMPTCProuter version: V0.35
  - OpenMPTCProuter VPS version: V0.51
  - OpenMPTCProuter platform: RPI3B+

Ysurac, 2018-09-09:
What is the VPN used ? OpenVPN ? IPsec ?
Strange DHCP here : some are in 192.168.1.x and some in 192.168.56.x with not the same MAC but with same name, it's expected ?

Maudissure, 2018-09-09:
The VPN is IPSEC/L2TP.

For the two subnet,I dunno. My computer try to connect to company's VPN automatically, perhaps it's the reason for the other subnet.

Ysurac, 2018-09-10:
I just tested an IPSEC VPN under Linux, and it's working without problem. Maybe a problem with the client used.

Can you try to use fixed IP on your computer, maybe there is a DHCP conflict with IPSEC VPN DHCP?

Maudissure, 2018-09-10:
I can't, company's computer block all systems configuration. 

It's DHCP only.

Althought, these computers works normaly with my load balancing router with same subnet/DHCP  parameters

Ysurac, 2018-09-10:
You may have to set MTU on the IPsec client.
What VPN do you use on OpenMPTCProuter ?

Maudissure, 2018-09-10:
As i said, i have no control on system parameters, the vpn client is from Cisco.

On OMR, i think it's Glorytun since i haven't change any settings from the original install except open port and fixed address on DHCP.