Ysurac/openmptcprouter #162: omr-tracker problem

rstanislav, 2018-09-09:
When connections lost, and for example only 1 available i see in log:
Sun Sep  9 23:00:43 2018 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 173.194.73.101)
Sun Sep  9 23:00:46 2018 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.164.1 dev eth2
Sun Sep  9 23:00:46 2018 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Sun Sep  9 23:00:46 2018 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP

But UDP for shadowsocks disabled as in default settings..

Also i have  a question - does omr tracker uses all configured remote ip addresses to test at same time, or 1 after another if first ip test failed..?

Ysurac, 2018-09-10:
* When VPN is down, ShadowSocks is used for UDP by default. At least if hi2 ss-redir client is configured in mode "tcp and udp". I don't understand what issue you have here.

* True. OMR server can't be used for tracker.

* True, all ips configured are used for test 1 after 1.

* Root DNS servers are used. MPTCP don't care about DNS, because many DNS queries are over UDP and not TCP, it's why a TCP VPN is used for UDP by default.
I will not use WAN defined DNS because each user would need to know them, and ISP often hijacks DNS.

Ysurac, 2018-09-10:
* It use ShadowSocks to get the same UDP IP output address as TCP.

* Can be a choice, but ISP can see all DNS traffic then...
I always accept good pull request with accepted CLA (Contributor License Agreement). But this will be not so easy to add, you have to play with route, iptables mark and ip rule.

closed 2018-10-24 by Ysurac