Ysurac/openmptcprouter #165: Question: Instructions on setting up IPv6?

reedog117, 2018-09-12:
Are there any setup instructions for setting up IPv6 both on the LAN and WAN sides? My VPS has a fully-working IPv6 address and connection, so I'm curious what additional steps are needed.

Ysurac, 2018-09-12:
Nothing. What do you have in OpenMPTCProuter status page ?

reedog117, 2018-09-12:
Originally I had "IPv6 Enabled" unchecked, as IPv6 wasn't originally setup on my VPS. I checked it, and now it's working, but I'm noticing this error on one of my interfaces (wan3) that I did not see before. I suspect it's because one of my WiFi-connected devices (an iOS device acting as a hotspot) is providing both IPv4 and IPv6 addresses via hotspot (T-Mobile is an IPv6-native mobile network). How do I handle "IPv6 route received" and change the connection from yellow to green?

[image]

Ysurac, 2018-09-12:
The only way is to disable IPv6 on your hotspot (or to send IPv6 route to wan3). But if it's working and IPv6 doesn't conflict on your network it's not a problem.

closed 2019-03-13 by Ysurac

Adorfer, 2020-05-12:
what are the minimum requirements for IPv6 on the VPS? (apart from working DNS6)
- for public ipv6 on the LAN: /64 on the VPS or more? (or less due to "intelligent" dhcp6 instead of SLAC?)
- of nat6 on the LAN: /128 on the VPS

Ysurac, 2020-05-12:
In current release only private ipv6 are distributed on the lan.
In next release a /48 is needed on the VPS if you want public IPv6 on LAN automatically. This would be a bit more difficult with a /64 but I will make some tests.

Adorfer, 2020-05-12:
Thanks for the quick reply. 
/128 on the VPN (ubuntu18.04 with netplan, definitly working ipv6 setup): NOK, route is not set up in current release. 
No idea how to debug, status-page complains "no ipv6 setup", despite beeing enabled. 

/48: No chance at resonables priced VPS services, /64 with some kind of "full cone" static maping possible? Would be nice especially for SIP phones.