Ysurac/openmptcprouter #167: Glorytun not running

kapook007, 2018-09-12:
hello
OpenMPTCProuter  with shadowsocks is working
but
OpenMPTCProuter  with glorytun (udp+tcp) is not working / not running
___________________
 in webconfig
glorytun Started = no
_____________________
 in console 
root@OpenMPTCProuter:~# glorytun
AES-256-GCM is not available on your platform
keyfile option must be set
__________________
myconfig

config glorytun 'server'
	option _description		"Simple TCP server configuration"
	option _role			"server"
	option port			"65001"
	option dev			"tun0"
	option key			"secretkey"
	option listener			"1"
	option localip			"192.168.99.1"
	option remoteip		"192.168.99.2"
	option proto			"tcp"
	option enable			"0"

config glorytun 'client'
	option _description 'Simple TCP client configuration'
	option _role 'client'
	option port '65001'
	option dev 'tun0'
	option key 'secretkey'
	option localip '192.168.99.2'
	option remoteip '192.168.99.1'
	option proto 'tcp'
	option host 'myserverip'
	option enable '1'

  - OpenMPTCProuter version: (v0.34)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2018-09-12:
You are not using default configuration, this will not work with OpenMPTCProuter VPS script.
Else, your issue is here "AES-256-GCM is not available on your platform", so you have to use chacha20.

Ysurac, 2018-09-12:
This is how the official image is build.
Glorytun give you the same error after chacha20 enabled ?

kapook007, 2018-09-12:
enable chacha20 and reboot
same error
root@OpenMPTCProuter:~# glorytun
AES-256-GCM is not available on your platform
keyfile option must be set

______________

in Glorytun webconfig 
add new sample config but  config is empty
i capture is config  empty
https://www.img.live/images/2018/09/12/11111fa4eb.jpg

Ysurac, 2018-09-12:
You must have a key defined.

True, there is a bug in Glorytun interface. I will check if bug is fixed in OpenWrt...

Why do you compile your own image and why do you change Glorytun default configuration ?

kapook007, 2018-09-12:
i try your build
https://www.openmptcprouter.com/download
x86-64 UEFI (64-bit) / VMware 3 or 4 image

same problem 
Glorytun not running 
please fix

Ysurac, 2018-09-12:
I'm running the image with Virtualbox right now, no problem.
Use wizard to configure it and check status page.

kapook007, 2018-09-12:
i try your build
https://www.openmptcprouter.com/download
openmptcprouter-v0.35-r7301
x86-64 UEFI (64-bit)  squashfs image  and write to usb
and use wizard setup

Glorytun not running
status page
https://www.img.live/images/2018/09/12/1111.jpg

my Glorytun  config
https://www.img.live/images/2018/09/12/222.jpg

Glorytun 
https://www.img.live/images/2018/09/12/333.jpg

only shadowsocks  is working
Glorytun   not working

Ysurac, 2018-09-12:
The glorytun config is not the default one, this config can't work with OpenMPTCProuter VPS part.

In the status page I see that MPTCP seems to be disabled on both wan, can you try MPTCP Support Check on both wan ? And the result of ```omr-mptcp-intf wan1``` and ```omr-mptcp-intf wan2```.
Glorytun can crash if MPTCP doesn't work.

kapook007, 2018-09-12:
my wan1 / wan2 is internet  usb 4G 
mptcp workink on shadowsocks 

this graph bw shadowsocks 
https://www.picz.in.th/images/2018/09/12/fPKEPS.jpg

i dont know status page show wan1 and wan2 show multipath seems to be block !!!
https://www.picz.in.th/images/2018/09/12/fPKqWV.jpg

this omr-mptcp-intf wan1 and omr-mptcp-intf wan2
https://www.picz.in.th/images/2018/09/12/fPK5KQ.jpg

kapook007, 2018-09-12:
please sent your working  glorytun config server and client for upd and tcp
thank you T T

Ysurac, 2018-09-12:
The default one, here: https://github.com/Ysurac/openmptcprouter-feeds/blob/master/glorytun/glorytun.config

kapook007, 2018-09-12:
this config 
https://github.com/Ysurac/openmptcprouter-feeds/blob/master/glorytun/glorytun.config

not working
Glorytun not running
T T

Ysurac, 2018-09-12:
There is no special log file. All is available via ```logread``` or in system log using the interface.

kapook007, 2018-09-12:
my glorytun log
Wed Sep 12 15:44:05 2018 daemon.err glorytun[3445]: unable to read secret key
Wed Sep 12 15:44:06 2018 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Sep 12 15:44:09 2018 daemon.err glorytun-udp[3972]: unable to read secret key

Ysurac, 2018-09-12:
Strange. The key should be in the file ```/tmp/glorytun-vpn.key```, could you check if this file exist ?

kapook007, 2018-09-12:
yes i found
glorytun-udp-vpn.key   and   glorytun-vpn.key

kapook007, 2018-09-12:
yes correct glorytun key

Ysurac, 2018-09-12:
Can you give me the ```/etc/config/glorytun``` with the key visible.

kapook007, 2018-09-12:
config glorytun 'vpn'
	option host '127.0.0.1'
	option port '65001'
	option dev 'tun0'
	option mptcp '1'
	option proto 'tcp'
	option chacha20 '1'
	option mtuauto '1'
	option key 'secretkey555'

Ysurac, 2018-09-12:
What is this key ? it's not in the good format. host is also wrong.

kapook007, 2018-09-12:
i dont understan
option host  this is 127.0.0.1  for server mptcp  Right?
option host  this is serverip  for client mptcp  Right?

option key 'secretkey555'  not ok ?

Ysurac, 2018-09-12:
*host* should be your VPS IP address.
*key* should be the key generated by OpenMPTCProuter VPS script that you run on your VPS (keys are then available in ```/root/openmptcprouter_config.txt``` on the VPS).

kapook007, 2018-09-12:
ok i got it 
VPS is only debian9-x86_64 witch script 
i can not use openwrt  (OpenMPTCProuter) make to VPS
openwrt  (OpenMPTCProuter) for mptcp client only

Ysurac, 2018-09-12:
There is no OpenMPTCProuter make for VPS, only the ```debian9-x86_64.sh``` script that add MPTCP kernel and install & generate key for shadowsocks and glorytun to a Debian 9 VPS. This part is needed to have a working OpenMPTCProuter.
So OpenMPTCProuter is always client of the VPS, and you must put the keys generated by ```debian9-x86_64.sh``` in the wizard.

closed 2018-09-13 by Ysurac