Ysurac/openmptcprouter #168: VLAN tagged port

HanXHX, 2018-09-15:
Hi,

Is it possible to add a tagged port with a RPI3?

I would like to add a tagged port (id = 100) on LAN.

Cheers!

Ysurac, 2018-09-15:
True, I do not create macvlan over vlan interface. This will be fixed in next release (this should work).

uppaljs, 2018-09-15:
Thanks a lot for the quick reply. A couple of questions:

1. When will the next release be?

2. Can a checkbox be added (VLAN_SUPPORT), that when checked asks for vlan number to tag packets of that certain interface through its master ( i.e. eth0 in case of rpi2B ). 

If # 2 is not possible, can you please provide a smalll walkthrough as to how vlan tagging can be achieved.

Ysurac, 2018-09-15:
1. Should be out Monday.
2. No

You can check OpenWrt documentation for vlan tagging. In fact it's using the interface eth0.100 for example.

Vinc32du44, 2018-11-25:
Hello,
How can i edit /etc/config/network and /etc/init.d/network reload on my Rpi ?
I connect to my RPi with SSH to root => it's Ok
Try edit with "sudo nano /etc/config/network" => not work ???
Can you help me ?
Thanks
P.S: Excuse my poor english :~|

Ysurac, 2018-11-25:
Why do you want to edit the config file and not use the LuCi interface ?

closed 2018-12-26 by Ysurac