Ysurac/openmptcprouter #169: firewall doesn't work for open port 8080

mykelinho, 2018-09-16:
Expected Behavior
firewall doesn't work for open port 8080

Steps to Reproduce the Problem
test firewall with adress 192.168.100.23
when i go to my ip public 54.xx.xx.xx:8080, doens't work

Specifications
OpenMPTCProuter version: openmptcprouter v0.34.4 r7298-f7413579c0 / LuCI Master (git-18.248.76017-8951f5f)
OpenMPTCProuter VPS version: OVH Debian 9
OpenMPTCProuter platform: RPI-3+

[image]

Ysurac, 2018-09-16:
Did you follow the documentation https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding ?

mykelinho, 2018-09-16:
sorry i have,'t read this documentation.
it's OK good!
thank you

closed 2018-09-16 by mykelinho