Ysurac/openmptcprouter #170: can't connect lan

Yanaem, 2018-09-17:
## Expected Behavior

Access to the wizard with my browser on 192.168.100.1

## Actual Behavior

I  can't connect the wizard with my browser on 192.168.100.1

## Steps to Reproduce the Problem

  1. Install img of openmptcprouter v.35
  2. connect to lan
  3. check if everything is ok with "ifconfig"
[image]

## Specifications

  - OpenMPTCProuter version: (v.035)
  - OpenMPTCProuter VPS version: ()
  - OpenMPTCProuter platform: (/RPI3 b+)

Ysurac, 2018-09-17:
Did you set an IP on 192.168.100.x network on your computer ?

Yanaem, 2018-09-17:
No here are my network : 
[image]

(je suis français aussi désolé pour mon anglais ;) )

Ysurac, 2018-09-17:
So you have to add, to the computer you want to use to access OpenMPTCProuter interface, an IP in 192.168.100.x network (or you can change netmask).

Yanaem, 2018-09-17:
maybe the problem comes here , my netgear router is 192.168.0.XX.
[image]

Ysurac, 2018-09-17:
It's not a problem. You only have to add a static IP in 192.168.100.x  on your computer network interface.