Ysurac/openmptcprouter #173: VPN over openmptcprouter (standard config)

dur3x, 2018-09-18:
I'm mandatory to use a certain VPN (PPTP vpn/tcp 1723) to communicate with my company.
When I do that without openmptcprouter it works fine but when I try that through openmptcprouter it doesn't work.

## Expected Behavior
VPN over VPN (PPTCP tcp1723) works

## Actual Behavior
VPN over VPN (PPTCP tcp1723) doesn't work

## Steps to Reproduce the Problem

  Simply click on my tablet (android) "connect to pptp vpn"

## Specifications

  - OpenMPTCProuter version: (Last is not a version) v0.35
  - OpenMPTCProuter VPS version: (Last is not a version) VPS 0.46
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) x86_64

MPTCP with shadowsocks (standard config) 
UDP,ping &rest through MLVPN but same results with None, glorytun tcp

dur3x, 2018-09-18:
> PPTP really ? I've not yet tested that... Don't know if I can't find a free VPN that using this old protocol to test :)
No comment. It's just used to access to a private NAS. It's clearly old fashioned.

> What is the result if you disable ShadowSocks and use Glorytun TCP ?
Unsuccessful

> Can you set MTU somewhere on PPTP client ?
Not on the client directly but when I'm connected on the server I can modify the MTU of the server.
Currently it's set to 1400.. I already tried 1500 but no update.

Perhaps the problem is between the VPS server and the vpn server.
I made a simple test (netcat to VPN server TCP 1723), this port looks open & accessible.

I don't know exactly how we can investigate this problem. 
At my side I can try to install a pptp client on the vps and try directly from there.
Another thing I can do is to give you a valid pptp server (and if needed valid credentials)

dur3x, 2018-09-18:
Another thing I can do is to take a pcap dump :) Not sure it will help

dur3x, 2018-09-18:
I tried that thing : https://docs.ovh.com/fr/overthebox/utiliser-un-serveur-vnp-pptp/
but without success

Ysurac, 2018-09-18:
I found a free PPTP VPN, I will make some test.

TRSATOC, 2018-09-19:
Hi, i've got the same problem here.
our PPTP Server is a Win Server 2008r2
I've tried with different clients (android phone, windows, etc...)
I've enable GRE forward on both openMPTCPServer et router without results

Ysurac, 2018-09-19:
Should be better in OpenMPTCProuter v0.36.3, even if I need to test more.

TRSATOC, 2018-09-19:
OK, i'll try tomorrow, thx

dur3x, 2018-09-19:
Thanks for the update :-)
I tested with the v0.36.3 but without success.
If it works at your side I will gladly accept your configuration/steps.

Ysurac, 2018-09-19:
It's working with only glorytun TCP enabled and ShadowSocks disabled.
I think it's because GRE and TCP connection doesn't use same path.

dur3x, 2018-09-19:
If I understand well you propose to bypass the mptcp part. 
It is a pity to not take advantage of the mptcp redundancy/aggregation.

Ysurac, 2018-09-19:
No. You bypass Shadowsocks and use glorytun TCP for PPTP. Glorytun TCP use MPTCP. So no real changes here, only Glorytun TCP can be slower for TCP than Shadowsocks, but it's not a real problem for PPTP that use GRE for data transfer.

dur3x, 2018-09-19:
Thanks @Ysurac I will try that asap and keep you informed !

dur3x, 2018-09-19:
@Ysurac Big thanks it works ! 👍

Ysurac, 2018-09-20:
This bypass all because I used the same list to bypass shadowsocks and to bypass glorytun...
This is fixed in OpenMPTCProuter v0.36.4 (that will be out in a few hours)

TRSATOC, 2018-09-20:
Perfect, with the 0.36.4 the PPTP Server Behind openMPTCPRouter works!

dur3x, 2018-09-20:
Thanks for the support, it's OK now 👍

closed 2018-09-20 by dur3x

cron410, 2019-06-01:
I'm still having this issue with a remote PPTP VPN server at work. Was able to fix it with Shadowsocks bypass @Ysurac mentioned above. I placed the IP of my office's VPN server in shadowsocks-libev > Redir Rules > Source Settings and also in Destination Settings, in the first field of both pages. Works great now. 

If there isn't a way to fix SS permanently, maybe we can add a rule that bypasses SS for any traffic to or from TCP 1723? Maybe the extra hop on a private network (10.255.255.1) making this look like double NAT to the PPTP client.