Ysurac/openmptcprouter #176: Gateway down

constant1010, 2018-09-19:
Hello
For my configuration I change parameters of WAN interfaces (type of interface: create macvlan.. to default) maybe because I use virtualbox, and it works. Because without this change I have gateway down.

Before V0.35 I change parameters of WAN interfaces (type of interface: create macvlan.. to default) for result, but since v0.35 if I change this parameter, system is blocking.

  - OpenMPTCProuter version: 0.36.3
  - OpenMPTCProuter VPS version: .50
  - OpenMPTCProuter platform: Virtualbox

constant1010, 2018-09-20:
I have one ethernet interface set in virtualbox for 2 wan and lan 
set as Intel PRO/1000 MT Desktop
bridged and promiscuous allow all

And an other WAN interface with LTE key (with message : "Multipath seems to be blocked on the connection"..) but it works.

Ysurac, 2018-09-20:
Set the interface as **Server Intel PRO/1000 MT (82545EM)**, the Desktop interface give strange results...

constant1010, 2018-09-20:
Perfect

closed 2018-09-21 by Ysurac