Ysurac/openmptcprouter #177: Multipath seems to be blocked on the connection

Yanaem, 2018-09-19:
## Expected Behavior

Agregate 4G (orange) + lan

## Actual Behavior

No agregation and the message : "Multipath seems to be blocked on the connection"

## Steps to Reproduce the Problem

  1.Connect the lan via a netgear router 
  2. Connect the 4g via netgar router 4G
  3. Both router connected on a switch.
[image]

## Specifications

  - OpenMPTCProuter version: (v0.36)
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: RPI3)

Ysurac, 2018-09-20:
Can you give me the result of *System*->*OpenMPTCProuter*->*MPTCP Support Check* on wan2 ?

Yanaem, 2018-09-20:
see below 

1: 192.168.2.1 
2: * 
3: 172.25.52.22 TCP::CheckSum IP::DiffServicesCP IP::TTL IP::CheckSum 
4: 192.168.120.5 TCP::CheckSum IP::DiffServicesCP IP::TTL IP::CheckSum +TCPOptionPad -TCPOptionMPTCPCapable 
5: 192.168.19.2 TCP::CheckSum IP::DiffServicesCP IP::TTL IP::CheckSum +TCPOptionPad -TCPOptionMPTCPCapable [Extra headers: ICMPExtension ICMPExtensionObject ICMPExtensionMPLS ICMPExtensionMPLS ] 
6: 192.168.19.69 TCP::CheckSum IP::DiffServicesCP IP::TTL IP::CheckSum +TCPOptionPad -TCPOptionMPTCPCapable [Extra headers: ICMPExtension ICMPExtensionObject ICMPExtensionMPLS ICMPExtensionMPLS ] 
7: 192.168.19.105 TCP::CheckSum IP::DiffServicesCP IP::TTL IP::CheckSum +TCPOptionPad -TCPOptionMPTCPCapable 
8: 193.252.227.197 TCP::CheckSum IP::DiffServicesCP IP::TTL IP::CheckSum +TCPOptionPad -TCPOptionMPTCPCapable 
9: 193.253.95.149  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
10: 81.253.131.2  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
11: 193.252.98.169  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
12: 193.252.98.234  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
13: 91.121.131.193 TCP::CheckSum IP::TTL IP::CheckSum +TCPOptionPad -TCPOptionMPTCPCapable 
14: 213.186.32.214 TCP::CheckSum IP::TTL IP::CheckSum +TCPOptionPad -TCPOptionMPTCPCapable 
15: * 
16: * 
17: * 
18: * 
19: * 
20: * 
21: * 
22: 51.38.232.252 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize -TCPOptionMPTCPCapable 

Result:
< Ethernet (14 bytes) :: DestinationMAC = ca:42:65:19:88:e6 , SourceMAC = 8c:3b:ad:fa:dd:05 , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 10 , ExpCongestionNot = 0 , TotalLength = 44 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 42 , Protocol = 0x6 , CheckSum = 0x71a7 , SourceIP = 51.38.232.252 , DestinationIP = 192.168.2.50 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 44091 , SeqNumber = 3770529300 , AckNumber = 1815758740 , DataOffset = 6 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 29200 , CheckSum = 0x522e , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1380 , >

Yanaem, 2018-09-20:
Ok i'll try with another FAI and then i will disable shadowsocks. I will let you know.

Yanaem, 2018-09-20:
Ok it works with a new FAI.

Just to know because i will prefere using Orange (the FAI which is not working). If i disable shadowsocks , it will be as efficient ? It's not so important to have the proxy on?

Evreything seems to be working but somethimes i have the error "No VPS IP adress, No WAN IP adress", and just after it's Ok and then it shows the same error again like a loop and then it's OK... (i can open a new issue if you want)
[image]

Ysurac, 2018-09-20:
If you disable ShadowSocks you must use Glorytun UDP or MLVPN to get aggregation without MPTCP. For me it's slower than ShadowSocks with MPTCP, but you should make some speed tests.

The error is because the connection make too much time to answer to get the IP from external website.
If you have nothing special in log, then you can ignore. This can be because the server used to get the IP is in France and latency is too high between, your location and the server.

Yanaem, 2018-09-20:
Ok thanks, when i disable wan 2 i don't have this loop.. i will make some test.

For now this is my configuration : 

1) To test with the FAI which is not working i will delete the shadowssocks key. And that's it right? Nothing else to do.

2) But you talk about MLVPN..i don't have password for that, maybe i made a mistake? 

Thanks

Ysurac, 2018-09-20:
The loop in status page is not a real problem if it's only in status page ;)

1. True delete shadowsocks key and set VPN as Glorytun UDP (not TCP)
2. No, you have to do ```MLVPN=yes sh debian9-x86_64.sh``` as you can see in doc: https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS But you should test Glorytun UDP first.

Yanaem, 2018-09-20:
Ok i tried with UDP and shadowsocks deleted and it's worst
[image]

So it's seems to not work with my orange mobile but ok with SFR..strange.

Yanaem, 2018-09-21:
Hello,

It a nushell : 

1) Ok good with a sim (SFR) , it seems to be stable. the ping as you said is high because the server is in France (ovh)

2) For the other sim (orange) nothing works, maybe they blocked something i don't know. It's unfortunately the sim i want.
I will try other test with free, etc..

3) I am using this system to just to add little more speed , i don't need all the power of 4G. Is there a way to say : user the 4G only when necessary to add more ? i think it might be a combine setting of : Vitesse de téléchargement (Kb/s) , and quota , right?

Ysurac, 2018-09-21:
1. ok
2. No idea what there are doing, many private networks before a public IP, they must have strange middle box...
3. By default it use the connection with the lowest RTT and if it need more bandwidth it use the next connection with the lowest RTT.

closed 2018-09-21 by Yanaem

dnwk, 2018-11-30:
One of my connections will show multipath block sometimes but not always.