Ysurac/openmptcprouter #178: VPS or Router Bottleneck

PxxBxx, 2018-09-23:
_Note : it's not an OMR issue, it may be a discussion regarding VPS / Router performances... I'm sorry if I'm opening an issue for a need of exchange with the OMR community. OMR is really great, It's now a stable product (My opinion), I'd love a forum for users :) :) :)_

## Expected Behavior

OMR Total Bandwidth = Wan1 + Wan2 - tunneling overload

## Actual Behavior

A lot less...
Wan1 is 40Mb down / 14Mb up
Wan2 is 25Mb down / 8Mb up
OMR is only 23Mb down / 6Mb up
(all tests on the same speedtest server, all in a row)

As CPU utilization seems incredibly low on the RPI (~7% including I/O, nice, wait, ...) I'm beginning to imagine my OVH VPS (VPS SSD 1 core) to be the bottleneck, not performance-wise, but maybe the network connectivity.

Does any other users with OVH VPS'es have the same symptoms ?
Or could it be a limitation of a RPI3B+ ? (I think not, I can OpenVPN at 88Mb/s with it)

## Specifications

  - OpenMPTCProuter version: 0.37 (also on 0.34 and before)
  - OpenMPTCProuter VPS version: 0.52 (also on 0.49 and before)
  - OpenMPTCProuter platform: RPI3B+

Ysurac, 2018-09-23:
Speedtest is often bad to test speed of OMR. You should try ```omr-test-speed``` on OpenMPTCProuter.
If some connections have a very high latency, this can also give a very bad speed with MPTCP.

What is the connection used as master and what are the connections ? (4G, ADSL,...)

I have a OVH VPS that I use for tests, and I can get full connections speed.

Ysurac, 2018-09-24:
What did you put as download and upload speed values for each wan ?
For 4G I also put 90% as peak speed, but not sure if it's really a good idea here... Maybe it's better to disable it for 4G connections.

PxxBxx, 2018-09-24:
I put 20Mb/2Mb when I first tested (so the ~1500kB omr-test-speed on each wans).
I tried 40Mb/2Mb, and had ~3200kB
I disabled QoS and had ~3500-6000kB (~9000kB without specifying any wan)
I'll keep it disabled and try when my first meetings will occurs (UDP Slack/Webex/Skype/...).
If it's not OK, I'll try to play with QoS traffic shaping limits.

FYI, there is a small bug on the System > OpenMPTCPRouter configuration page : 
 - you set bandwidth limits, then save -> QoS is enabled with the values
 - you empty the bandwidth limits, then save -> QoS is disabled
 - but the form on System>OMR still show values for bandwidth limiting, so If you save again, it re-enable the QoS.
Those fields should be empty when QoS is disabled.

Ysurac, 2018-09-24:
Bug with bandwidth in wizard should be fixed in OpenMPTCProuter v0.37.1

closed 2018-10-29 by Ysurac