Ysurac/openmptcprouter #179: Question: how to assign a different gateway/dns via OMR dhcp ?

PxxBxx, 2018-09-25:
## User story
As a Satellite Receiver leased by one of my ISPs
I want to receive via dhcp the IP of the ISP's modem as a gateway, and specific DNSes (eg. 8.8.8.8 is OK)
So that I can validate that I access the Internet network via a legit ISP IP

TL;DR : I did not found a way to tell the dhcp daemon to attribute a static lease with another gateway than the OMR one... And as the modem is on the OMR network with its dhcp disabled, I'm lost.. Thanks in advance !!

## Specifications

  - OpenMPTCProuter version: 0.37
  - OpenMPTCProuter VPS version: 0.52
  - OpenMPTCProuter platform: RPI3B+

Ysurac, 2018-09-25:
In **Network**->**DHCP**->**Static lease** ?

PxxBxx, 2018-09-26:
Unfortunately, It doesn't work.
My OMR setup has 192.168.100.0/24 network, and the gateway is 192.168.100.1

I've set (using MAC address for the Satellite tuner)
OrangeTV / <mac> / ip=192.168.3.219 / gw=192.168.3.1 / lease=24h

But the lease is a standard dhcp one, within OMR IP ranges (see below).
Maybe dhcp server cannot allocate IPs outside its range ?

> Wed Sep 26 18:54:38 2018 daemon.info dnsmasq-dhcp[14762]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Wed Sep 26 18:57:11 2018 daemon.info dnsmasq-dhcp[14762]: DHCPDISCOVER(eth0) 192.168.100.241 d4:7a:e2:c1:27:f3
Wed Sep 26 18:57:11 2018 daemon.info dnsmasq-dhcp[14762]: DHCPOFFER(eth0) 192.168.100.241 d4:7a:e2:c1:27:f3
Wed Sep 26 18:57:11 2018 daemon.info dnsmasq-dhcp[14762]: DHCPDISCOVER(eth0) 192.168.100.241 d4:7a:e2:c1:27:f3
Wed Sep 26 18:57:11 2018 daemon.info dnsmasq-dhcp[14762]: DHCPOFFER(eth0) 192.168.100.241 d4:7a:e2:c1:27:f3
Wed Sep 26 18:57:11 2018 daemon.info dnsmasq-dhcp[14762]: DHCPREQUEST(eth0) 192.168.100.241 d4:7a:e2:c1:27:f3
Wed Sep 26 18:57:11 2018 daemon.info dnsmasq-dhcp[14762]: DHCPACK(eth0) 192.168.100.241 d4:7a:e2:c1:27:f3

Ysurac, 2018-09-26:
True, it can't allocate IPs outside its subnet.

PxxBxx, 2018-09-26:
So, I think my only option is to have all wans in the same subnet.

eg:
OMR->192.168.100.1
wan1 gw -> 192.168.100.2
wan2 gw -> 192.168.100.3
...
dhcp leases in the 192.168.100.100-200 range

That way, it should work, I could be able to assign (via static leases), any wan gw to any device (ex: my satellite tuner, bound to work only with Orange network....)

Ysurac, 2018-09-26:
MPTCP will not work that way.
But you can add a new interface in the subnet you want and enable DHCP on it. Or, if possible, set a static IP/GW on your satellite tuner. You should also be able to use VLAN.

closed 2018-11-03 by Ysurac