Ysurac/openmptcprouter #180: Banana PI BPI-R2

Ysurac, 2018-09-27:
What is working:
* LAN & WAN
* HNAT

What is not working:
* Wifi
* Bluetooth

What is not tested:
* 4G using PCI-E

Ysurac, 2018-09-27:
Wifi was working, but not working now...
Can't load wlan_gen2 module:
```
[   14.028644] vmap allocation for size 425984 failed: use vmalloc=<size> to increase size
[   14.036630] kmodloader: vmalloc: allocation failure: 421888 bytes, mode:0x14000c0(GFP_KERNEL), nodemask=(null)
[   14.046586] CPU: 2 PID: 344 Comm: kmodloader Not tainted 4.14.71 #0
[   14.052797] Hardware name: Mediatek Cortex-A7 (Device Tree)
[   14.058340] [<c010f1d4>] (unwind_backtrace) from [<c010b2f8>] (show_stack+0x10/0x14)
[   14.066024] [<c010b2f8>] (show_stack) from [<c062c7bc>] (dump_stack+0x78/0x8c)
[   14.073192] [<c062c7bc>] (dump_stack) from [<c01a3698>] (warn_alloc+0xc0/0x16c)
[   14.080446] [<c01a3698>] (warn_alloc) from [<c01ce1cc>] (__vmalloc_node_range+0x218/0x224)
[   14.088645] [<c01ce1cc>] (__vmalloc_node_range) from [<c010c63c>] (module_alloc+0x4c/0x54)
[   14.096843] [<c010c63c>] (module_alloc) from [<c017f434>] (load_module+0x9e8/0x20f0)
[   14.104524] [<c017f434>] (load_module) from [<c0180c68>] (SyS_init_module+0x12c/0x160)
[   14.112378] [<c0180c68>] (SyS_init_module) from [<c01079a0>] (ret_fast_syscall+0x0/0x54)
[   14.120800] kmodloader: 1 module could not be probed
[   14.125904] kmodloader: - wlan_gen2 - 0
```
And it's not a problem with vmalloc size.

Ysurac, 2018-09-28:
Wifi is solved in OpenMPTCProuter 0.37.3

its27, 2018-09-30:
I am unable to get 0.37.3 working correctly.

1. I seem to loose my settings after a reboot, its like the settings are stored on a RAM disk. I am booting from SD not EMMC.
2. I set WAN1 to DHCP, an IP address was obtained and I could see the default route was set correctly (I had not enabled shadowsocks or the VPN yet) however internet connectivity was very intermittent, almost like HNAT isn't working.

Ysurac, 2018-09-30:
1. Changes are stored on JFFS2 partition that can take some time to be initialized at startup
2. There is a network problem when something is connected to "lan" ports... Need to fix this.

its27, 2018-09-30:
1. Can this be improved? When I boot it tells me I need to set a root password again.
2. Thanks, will it help if I make the LAN ports all bridges as LAN and use the single WAN port for now?

Ysurac, 2018-09-30:
1. Maybe
2. No. But you can use Macvlan instead.

its27, 2018-10-05:
Do you know when you may be able to release a fix for the LAN ports and config?

Ysurac, 2018-10-05:
When it will be fixed.

Ysurac, 2018-10-06:
When using a already used sd card, rootfs_data need to be erased. So for now a ```mtd -r erase rootfs_data``` on OpenMPTCProuter will erase this partition and reboot. At reboot the partition should be formatted to JFFS2 and is used to save all changes made.

kapook007, 2018-10-07:
working
emmc can save config
mtd -r erase rootfs_data

thanks

Ysurac, 2018-10-12:
LAN ports cycle up and down.

its27, 2018-10-12:
Ok does this need to be fixed by openwrt or the vendor?

Ysurac, 2018-10-12:
OpenWrt doesn't fully support Banana PI R2 and the vendor doesn't support kernel 4.14... So this will be fixed by me when I will be able to find exactly what is causing this.

Ysurac, 2018-10-17:
Last crash:
```
[171111.766436] WARNING: CPU: 3 PID: 0 at net/sched/sch_generic.c:320 dev_watchdog+0x158/0x224
[171111.774740] NETDEV WATCHDOG: eth1 (mtk_soc_eth): transmit queue 0 timed out
[171111.781737] Modules linked in: rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial option ipw cdc_mbim wlan_gen2 usb_wwan usb_serial_simple ti_usb_3410_5052 sr9700 smsc95xx sierra_net sierra rtlwifi rtl8xxxu rt2x00usb rt2x00lib rndis_host qmi_wwan pppoe ppp_async pl2303 oti6858 nf_nat_pptp nf_conntrack_pptp mt76x2e mt7603e mt7601u mt76 mos7720 mmc_spi mct_u232 mcs7830 mac80211 keyspan kalmia iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cypress_m8 cp210x ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem belkin_sa bas_gigaset ax88179_178a asix ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl
[171111.852312]  xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor via_velocity via_rhine usbserial usbnet usbhid usb_gigaset tulip ts_fsm ts_bm solos_pci sky2 skge sis900 sis190 ser_gigaset rtl8150 r8712u r8169 r8152 r6040 pppox ppp_generic pegasus pcnet32 of_mmc_spi nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre
[171111.922438]  nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda ne2k_pci macvlan kaweth iptable_raw iptable_mangle iptable_filter ipt_ECN ipheth ip6table_raw ip_tables hso hid_generic gigaset forcedeth ezusb ethoc et131x ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 e1000e e100 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_acm bnx2 atl2 atl1e atl1c atl1 arptable_filter arpt_mangle arp_tables 8390 8250_pci mtkhnat 8139too 8139cp sch_cake act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress configs hid evdev input_core i2c_gpio i2c_algo_bit spi_ks8995 xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport
[171111.993714]  ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 nf_nat nf_conntrack ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables dss1_divert isdn_bsdcomp isdn slhc 3c59x vmxnet3 e1000 ifb dm9000 sit sctp libcrc32c tunnel4 ip_tunnel hfcpci hfcmulti veth tun l1oip mISDN_dsp mISDN_core vfat fat br2684 atm nls_utf8 nls_iso8859_1 nls_cp437 natsemi broadcom bcm_phy_lib md5 usb_storage leds_gpio ahci libahci libata ehci_platform sd_mod scsi_mod ehci_hcd
[171112.064014]  gpio_button_hotplug tg3 b44 ssb ptp pps_core mii
[171112.069815] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.72 #0
[171112.075853] Hardware name: Mediatek Cortex-A7 (Device Tree)
[171112.081482] [<c010f1f4>] (unwind_backtrace) from [<c010b318>] (show_stack+0x10/0x14)
[171112.089254] [<c010b318>] (show_stack) from [<c063223c>] (dump_stack+0x78/0x8c)
[171112.096507] [<c063223c>] (dump_stack) from [<c0117210>] (__warn+0xe4/0x100)
[171112.103500] [<c0117210>] (__warn) from [<c0117264>] (warn_slowpath_fmt+0x38/0x48)
[171112.111012] [<c0117264>] (warn_slowpath_fmt) from [<c053cb48>] (dev_watchdog+0x158/0x224)
[171112.119213] [<c053cb48>] (dev_watchdog) from [<c0167c4c>] (call_timer_fn.constprop.3+0x28/0x94)
[171112.127928] [<c0167c4c>] (call_timer_fn.constprop.3) from [<c0167e14>] (run_timer_softirq+0x15c/0x1b4)
[171112.137247] [<c0167e14>] (run_timer_softirq) from [<c010155c>] (__do_softirq+0xe4/0x250)
[171112.145362] [<c010155c>] (__do_softirq) from [<c011bd9c>] (irq_exit+0xac/0xbc)
[171112.152614] [<c011bd9c>] (irq_exit) from [<c0155110>] (__handle_domain_irq+0xbc/0xe4)
[171112.160467] [<c0155110>] (__handle_domain_irq) from [<c0101440>] (gic_handle_irq+0x5c/0x90)
[171112.168836] [<c0101440>] (gic_handle_irq) from [<c010becc>] (__irq_svc+0x6c/0xa8)
[171112.176341] Exception stack(0xdfc75f88 to 0xdfc75fd0)
[171112.181435] 5f80:                   00000003 c07dae08 1f999000 c0114040 ffffe000 c0a03c34
[171112.189632] 5fa0: c0a03be8 c0a30e90 8000406a 410fc073 00000000 00000000 c0a03a6c dfc75fd8
[171112.197827] 5fc0: c01084e4 c01084e8 60000013 ffffffff
[171112.202925] [<c010becc>] (__irq_svc) from [<c01084e8>] (arch_cpu_idle+0x34/0x38)
[171112.210349] [<c01084e8>] (arch_cpu_idle) from [<c014a63c>] (do_idle+0xa4/0x114)
[171112.217685] [<c014a63c>] (do_idle) from [<c014a930>] (cpu_startup_entry+0x18/0x1c)
[171112.225279] [<c014a930>] (cpu_startup_entry) from [<8010176c>] (0x8010176c)
[171112.232284] ---[ end trace 355d4262dba74a94 ]---
[171112.236953] mtk_soc_eth 1b100000.ethernet eth1: transmit timed out
```
This put eth1 down, so all lan* ports are down too.

its27, 2018-10-24:
Is this still an issue in 0.39?

Ysurac, 2018-10-24:
No need to ask on each release. I will put a message here if I can fix something.

suyuan168, 2019-01-18:
hello PI BPI-R2 WIFI 
When can I solve this problem?

Ysurac, 2019-01-20:
Wifi work in latest version but all memory is not available.
Wifi can't work when HIGHMEM is enabled in Kernel config for now. I need to find the good settings to make full memory and wifi working...

suyuan168, 2019-01-20:
BPI-R2 shutdown problem. How to shut down. Power button is invalid。Thank you very much for the administrator to solve the problem of wifi

suyuan168, 2019-01-21:
> Wifi work in latest version but all memory is not available.
> Wifi can't work when HIGHMEM is enabled in Kernel config for now. I need to find the good settings to make full memory and wifi working...

Oh, I am sorry that there is always a problem. Is BPI-R2 working at V0.45.16 wifi now? I can see the options in openwrt.

its27, 2019-03-04:
Hi,

I just tested 0.46.3 with the BPI-R2.

1. The "LAN" ports seem to work OK now, I was able to use one of the ports to connect to my router via DHCP. The orange TX LED does not work.
2. I was unable to get the WiFi to work, I enabled both wireless interfaces but neither would find ant networks when I clicked 'Scan'.
3. Speed is good when using Shadowsocks however the speed is terrible when using Glorytun TCP. I get 38Mbps download with Shadowsocks and 2Mbpps with Glorytun TCP. Is hardware crypto used for both? All all encryption methods supported in hardware? Glorytun works fine with the same ADSL connection on my RPI3.
4. USB ports do not work, can this be fixed so we can add 4G modems / tether phones?

Thanks,

Ysurac, 2019-03-06:
1. OK
2. I will test again when I will have some times...
3. Both use chacha20 by default, so I don't think this use hardware crypto. You can change it.
4. I need to test that to

I will also make some tests with 4.19 kernel, this should perform better.

its27, 2019-03-06:
Thanks, do you know what encryption methods will use hardware crypto?

lygjxh, 2019-03-23:
why can't I get IP from LAN or WAN port? I'm running on v0.47.4.

Ysurac, 2019-03-23:
Really ? It's too much data. What do you mean by "I can't get IP" ?

lygjxh, 2019-03-23:
> Really ? It's too much data. What do you mean by "I can't get IP" ?

Trying to connect.....

Ysurac, 2019-03-23:
To what IP ? to 192.168.100.1 over lan interface ? Your computer on same network ?
The WAN interface on BPI-R2 is used as LAN and the 4 LAN interfaces are used as WAN.

bournehlm, 2019-09-10:
It seems wifi still not working on BPI R2 with v0.51.5 r0+10518-6d61fcfa9e version.

gitxmax, 2020-03-23:
Hello,
Thinking of getting the Banana R2 board, but seems the issues with it are not resolved since 2018.
Would you recommend to avoid it, or there's some work being done on it?
Thanks

Ysurac, 2020-03-23:
I would avoid it. For now latest release doesn't work on it and I still don't know why...
The manufacturer, sinovoip, sent me a BPI-R64 but I didn't receive it yet. This card should have a far better support.

OlegVekhov, 2020-12-03:
Hello! What's going for now on BPI-R2.

Current OpenWRT trunk work perfectly (i mean ethernet part, not interested in WiFi). But OMPTCP not working. In UART I can see it boots, but has problems in writing configuration. May be you can update to latest OpenWRT trunk?

I'm not familiar in programming, but can help in things that I understand...

Ysurac, 2020-12-03:
Snapshot image should work. So next release will work.

OlegVekhov, 2020-12-03:
Kindly waiting for next release :))

P.s. Also tried on RPI 3, but on RPI3 i can't use more than 3 modems, and it works very slow as I noticed. I think it RPI problem, I tried original OpenWRT and the problem persist. On x86 systems it works perfectly, but I need something compact...

Thank you again

Ysurac, 2020-12-03:
You can use more than 3 modems on RPI3. RPI4 is faster. There is some small x86_64 devices from China.

OlegVekhov, 2020-12-25:
According to download page, looks like bpi-r2 support is still broken in new build... Is there are any chance to get working snapshot?

Ysurac, 2020-12-25:
Not tested, but should work.

OlegVekhov, 2020-12-25:
Tested - the same issue. Can't do anything with overlayfs, network config is broken

Ysurac, 2020-12-28:
I made a new snapshot that should work: https://download.openmptcprouter.com/develop/5.4/bpi-r2/targets/mediatek/mt7623/

OlegVekhov, 2020-12-29:
Great! Thank you! Your build work perfectly, uptime 4 hours with three modems, tried to detach modems, all working as expected!

supagroova, 2021-01-06:
Hi there, I'm looking into to setting up a home router with a BPI-R2 to be able to bond my ADSL and 4G WAN connections. 

I see here in the notes it says that WiFi is not supported with the BPI-R2: Does that mean it can't be used as a WAN or a LAN connection or both?

Ysurac, 2021-01-06:
It's mean that BPI-R2 internal wifi is not recognized at all. Wifi driver is really dirty code that need some external program to work and it's not really stable, so for now it's not enabled.

supagroova, 2021-01-11:
Got my BPI-R2 today and would like to try the above snapshot. Is there any documentation around that I can read to learn how to install it?

Ysurac, 2021-01-12:
To install the router part: https://github.com/Ysurac/openmptcprouter/wiki/Router-install#for-banana-pi-bpi-r2

supagroova, 2021-01-12:
Many thanks, seems like that requires `fsync` when writing the image and OSX doesn't support that. Do I need a linux box to write the image?

Ysurac, 2021-01-12:
Use ```conv=sync``` or use etcher.

supagroova, 2021-01-13:
Thanks, tried those options before asking here. I think it must be a hardware problem because I've tried a dozen different images which are meant to work and nothing does.

Ysurac, 2021-01-13:
Or you make an error. What do you do exactly ?

OlegVekhov, 2021-01-13:
To flash firmware to EMMC I uset MTK Flash Tool and USB Cable. You need to download all files in bpi-r2 folder, extract firmware image from .gz to current directory, specify scatter file in MTK Flash Tool, choose format all + download, and flash firmware over USB cable

supagroova, 2021-01-13:
> Or you make an error. What do you do exactly ?

@Ysurac After entering the flashed SD (using Etcher the write the [image above](https://download.openmptcprouter.com/develop/5.4/bpi-r2/targets/mediatek/mt7623/)) I see a mustard coloured screen for about 10s then it powers off (The green LEDs turn off). 

However I have discovered that I can get [Ubuntu Mate](https://download.banana-pi.dev/d/3ebbfa04265d4dddb81b/?p=%2FImages%2FBPI-R2%2FUbuntu18.04&mode=list) and [Alpine Linux](http://forum.banana-pi.org/t/bpi-r2-new-image-basic-alpine-router-platform-with-wifi-accesspoint-on-pantavisor-005-2018-06-17/6032) to run, so at least I know there isn't a problem with the hardware. Would it be simpler for me to try and install OpenMPTCProuter on one of those?

supagroova, 2021-01-20:
Thanks @OlegVekhov, I've been able to try with a Windows laptop but in the end it was crazy complicated to be able to install the drivers without the INF checks and I didn't have the necessary bitlocker codes to get access to that on the Windows machine.

I did try putting the [above image](https://download.openmptcprouter.com/develop/5.4/bpi-r2/targets/mediatek/mt7623/) onto an SD using Etcher in Windows, but that gave the same (failed) results.

As I see it, my final try with OpenMPTCProuter would be to try and run on top of either Ubuntu or Alpine linux. Is there any possibility of that?

Ysurac, 2021-01-21:
No. OpenMPTCProuter is a full image OS.
For the image use https://download.openmptcprouter.com/release/v0.57.3/bpi-r2/targets/mediatek/mt7623/openmptcprouter-v0.57.3-r0+15225-bfc433efd4-mediatek-mt7623-bpi_bananapi-r2-squashfs-img.gz as it's the latest stable release.
Don't forget to decompress the image before writing it to SD card.

supagroova, 2021-01-23:
Well I ended up getting a Raspberry Pi 4B and returning the BPI-R2. I had the Raspberry Pi up and running in 5mins. 👍

GreatWriter, 2021-06-10:
BPI-R2 0.57.3 support wifi ?

volodeus, 2021-09-23:
Hello!

I bought Banana Pi BPI-R2 But I can't get it to boot.

I tried writing a squashfs image: openmptcprouter-v0.58.5-r0 + 16336-b36068d35d-mediatek-mt7623-bpi_bananapi-r2-squashfs-img.gz using etcher and with commands:
gunzip openmptcprouter - *. img.gz
dd bs = 4M if = openmptcprouter - *. img of = / dev / sdX conv = fsync

But all I get is a white screen on startup and nothing else happens.

If I write the Ubuntu image to the same SD card, then everything is fine.

Also I didn't have any problems with RPI4 but would like to use my Banana Pi BPI-R2.

Can someone tell me what else can be done for this?

volodeus, 2021-09-23:
> No. OpenMPTCProuter is a full image OS.
> For the image use [...]

When unpacking this and other archives for banana, gunzip and other archivers write that they have ignored garbage.

Kirball, 2022-06-09:
> When unpacking this and other archives for banana, gunzip and other archivers write that they have ignored garbage.

Is this still the best version to use as the version that's suggested on the downloads page is newer but gets stuck in a boot loop and will not start for me?  What version should I be using?

Ysurac, 2022-06-09:
@Kirball you have a BPI-R2 or a BPR-R64, as you post on both subject...

Kirball, 2022-06-09:
oh yeah you're right this is an R2 post I have an R64 🤦‍♂️