Ysurac/openmptcprouter #181: Company computers - No internet connection

Maudissure, 2018-09-27:
## Expected Behavior
Connecting professional computer and get internet connection

## Actual Behavior
Company computer get IPV4 address but no internet connexion.
VPN connexion said it cannot resolve DNS address.

Some logs : 
Thu Sep 27 19:00:31 2018 daemon.info dnsmasq-dhcp[2935]: DHCPINFORM(eth0) 192.168.1.188 c8:d9:d2:d9:53:82
Thu Sep 27 19:00:31 2018 daemon.info dnsmasq-dhcp[2935]: DHCPACK(eth0) 192.168.1.188 c8:d9:d2:d9:53:82 PO701786
Thu Sep 27 19:00:57 2018 daemon.info dnsmasq-dhcp[2935]: DHCPREQUEST(eth0) 192.168.1.188 c8:d9:d2:d9:53:82
Thu Sep 27 19:00:57 2018 daemon.warn dnsmasq-dhcp[2935]: Ignoring domain siege.axa-fr.intraxa for DHCP host name PO701786
Thu Sep 27 19:00:57 2018 daemon.info dnsmasq-dhcp[2935]: DHCPACK(eth0) 192.168.1.188 c8:d9:d2:d9:53:82 PO701786
Thu Sep 27 19:01:23 2018 daemon.info dnsmasq-dhcp[2935]: DHCPINFORM(eth0) 192.168.1.188 c8:d9:d2:d9:53:82
Thu Sep 27 19:01:23 2018 daemon.info dnsmasq-dhcp[2935]: DHCPACK(eth0) 192.168.1.188 c8:d9:d2:d9:53:82 PO701786
Thu Sep 27 19:02:46 2018 daemon.info dnsmasq-dhcp[2935]: DHCPREQUEST(eth0) 192.168.1.178 30:24:32:50:8b:ec
Thu Sep 27 19:02:46 2018 daemon.warn dnsmasq-dhcp[2935]: Ignoring domain siege.axa-fr.intraxa for DHCP host name PO701786
Thu Sep 27 19:02:46 2018 daemon.info dnsmasq-dhcp[2935]: DHCPACK(eth0) 192.168.1.178 30:24:32:50:8b:ec PO701786
Thu Sep 27 19:02:51 2018 daemon.info dnsmasq-dhcp[2935]: DHCPREQUEST(eth0) 192.168.1.178 30:24:32:50:8b:ec
Thu Sep 27 19:02:51 2018 daemon.warn dnsmasq-dhcp[2935]: Ignoring domain siege.axa-fr.intraxa for DHCP host name PO701786
Thu Sep 27 19:02:51 2018 daemon.info dnsmasq-dhcp[2935]: DHCPACK(eth0) 192.168.1.178 30:24:32:50:8b:ec PO701786
Thu Sep 27 19:03:27 2018 daemon.info dnsmasq-dhcp[2935]: DHCPDISCOVER(eth0) d0:d2:b0:83:83:16
Thu Sep 27 19:03:27 2018 daemon.info dnsmasq-dhcp[2935]: DHCPOFFER(eth0) 192.168.1.132 d0:d2:b0:83:83:16
Thu Sep 27 19:03:28 2018 daemon.info dnsmasq-dhcp[2935]: DHCPREQUEST(eth0) 192.168.1.132 d0:d2:b0:83:83:16
Thu Sep 27 19:03:28 2018 daemon.info dnsmasq-dhcp[2935]: DHCPACK(eth0) 192.168.1.132 d0:d2:b0:83:83:16 Apple-TV
Thu Sep 27 19:03:34 2018 daemon.info dnsmasq-dhcp[2935]: DHCPDECLINE(eth0) 192.168.1.132 d0:d2:b0:83:83:16

## Steps to Reproduce the Problem

  1. Connecting computer by RJ45 or WIFI
  2. Computer get IPV4 address
  3. No internet access
  4. Message from VPN utility (Cisco) : "Cannot resolve VPN DNS address"
  5. Same with another company computer

## Specifications

  - OpenMPTCProuter version: V0.37.1
  - OpenMPTCProuter VPS version: V0.52
  - OpenMPTCProuter platform: RPI3B+

Maudissure, 2018-09-27:
result of the tcpdump command : 
BusyBox v1.28.3 () built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|
       |_|
------------------------------------------------------------------------------
   (r7324-05606de1aa)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.37.1

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Mon Sep 24 15:10:46 UTC 2018
-----------------------------------------------------
root@OpenMPTCProuter:~# tcpdump -i eth0 -n port 67 and port 68
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
19:28:42.192610 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from c8:d9:d2:d9:53:82, length 328
19:28:42.194931 IP 192.168.1.2.67 > 192.168.1.188.68: BOOTP/DHCP, Reply, length 314
19:30:07.510097 IP 192.168.1.188.68 > 255.255.255.255.67: BOOTP/DHCP, Request from c8:d9:d2:d9:53:82, length 284
19:30:07.511732 IP 192.168.1.2.67 > 192.168.1.188.68: BOOTP/DHCP, Reply, length 300

But i think the problem is none of the company computer have internet access.

Ysurac, 2018-09-27:
They have internet before using the VPS ?

Maudissure, 2018-09-27:
Nope.

Maudissure, 2018-09-27:
None of the two.

Maudissure, 2018-09-27:
ok, and what should i put in local domain settings ?

Ysurac, 2018-09-27:
You can try empty, the domain used by your computer,... no idea.
Asus routers seems to have the same problem with locked computer, I can see the same question often but I didn't find an answer yet... They also use dnsmasq for DHCP.

Maudissure, 2018-09-27:
With no local server and no local domain, it works !

Thanks.

The only issues remaining for me are Apple devices connection.

Maudissure, 2018-09-28:
For the iPhone, static IP AND static DNS works..

For the AppleTV and the iPad static IP on OMR didn't work, no internet connection; I have to set ATV and iPad IP and DNS directly on one LTE modem/router to get normal internet connection.

Ysurac, 2018-09-28:
Do you use wired network ?
What do you have in log ? If it's full of DHCP request this can make the network really slow and unstable.

Maudissure, 2018-10-13:
After other tests : 

-Lan connexions for the two compagny computers
-Settings "Local server" and "Local Domain" with nothing : 
1 -Both computers get IPV4 address
2 - Both computers are connected to internet
3 - Several seconds later, both lost internet connexion
4 - Nothing in logs

Ysurac, 2018-10-13:
Without any log difficult to find where is the problem.
Computers have the same IP and gateway before and after connection is lost ?

Neustradamus, 2019-04-30:
Have you done other tests?

closed 2019-12-20 by Ysurac

Neustradamus, 2019-12-20:
@Ysurac: It is solved?

Maudissure, 2020-01-30:
No it's not.

Ysurac, 2020-01-30:
Sure, without any news, tests or answer.