Ysurac/openmptcprouter #182: Public IP - question

kamcio2603, 2018-10-01:
## Expected Behavior
Public IP from hosting company on mikrotik device
  
## Actual Behavior

## Steps to Reproduce the Problem

 Hello, how can I get public IP on my mikrotik device (eg: GRE interface / IPIP ) via mptcp router?

## Specifications

  - OpenMPTCProuter version: v0.37.2 r7329
  - OpenMPTCProuter VPS version: VPS 0.52
  - OpenMPTCProuter platform: x86_64

Ysurac, 2018-10-05:
What do have and what do you want to do exactly ?

kamcio2603, 2018-10-05:
What do have? Tunnel is working over two nets (fiber && docsis).
what do you want? tunnel/get Ip address from debian machine (another public IP from ISP) over mptcprouter and to my mikrotik device. I started test for running GRE tunnel between Debian-vps <- mptcprouter -> Mikrotik. At this moment, not successfully.

Ysurac, 2018-10-05:
You may have to play with MTU, because it will use Glorytun TCP tunnel.
I can't help with this configuration, it's out of the scope of OpenMPTCProuter project.

closed 2018-10-24 by Ysurac