Ysurac/openmptcprouter #184: nocrypto for shadowsocks

kapook007, 2018-10-06:
my router mtk7621  is very slow for shadowsocks  

how to setup nocrypto shadowsocks  for router and vps?

Ysurac, 2018-10-06:
Disabling crypto will not really change anything. What is the CPU usage ?

kapook007, 2018-10-07:
My cpu is MediaTek Ralink MIPS > MT7621 based boards 
I test differrent board band but same problem shadowsocks slow < 8mb/s
How to disabling crypto in  shadowsocks ?

Ysurac, 2018-10-07:
I was asking for CPU usage, with ```top``` command for example.

You have to set *none* method in Shadowsocks-libev interface in server tab and same method on the VPS in ```/etc/shadowsocks-libev/config.json``` and restart shadowsocks-libev on the VPS.

You should also try to disable SQM if you have speeds problems.

kapook007, 2018-10-08:
none method in Shadowsocks-libev  is working
very low cpu usage 
Can i nocrypto on Glorytun ?

Ysurac, 2018-10-08:
Crypto can't be disabled with Glorytun.
But MT7621 should support Hardware Crypto with cryptodev module and this should work quite well with ```aes-128-cbc```.

kapook007, 2018-10-08:
MT7621 have some problem about Shadowsocks
speed download is ok / i got over 30mbs on 4G 
but speed upload is very low  < 3-5mbs on 4G
i have 2 board band use mt7621 is same problem 
can i sent MT7621 board to you .

Ysurac, 2018-10-08:
Upload is always lower than download on 4G, I have 100/11. I'm not sure if I can really make better than that. I should be able to make some tests on upload with fiber soon and see if it's 4G problem or if I can optimize that.

You can send me MT7621 board, at least I can make an official support of this board, test all and generate images.

kapook007, 2018-10-09:
problem upload is only cpu MT7621 board
i change MT7621 board to banana pi bpi-r2 with same aircard  speed is up 30/30 mbs

please sent your address to my email  oui_sexy9(AT)hotmail.com
i sent MT7621 board to you
thanks.

kapook007, 2018-10-10:
ok i got your address i will sent to you
thank you.

closed 2018-10-24 by Ysurac

dnwk, 2018-11-30:
I can't seem to get nocrypt to work. Once I switch the encryption method to none, the local instance on router keep stop running. Did nocrypt patch applied on router?